gtkwave/gtkwave #3: Is this the new official repo for gtkwave?

eine, 2019-12-25:
The project at Sourceforge (https://sourceforge.net/projects/gtkwave/) was last updated on Nov 11. The first commit was pushed to this repo on Nov 16. These lead me to think that Gtkwave is being migrated from Sourceforge (SVN) to GitHub (git). However, I saw no *official* notice explaining it.

Furthermore, since Tony Bybell's user at Sourceforge is [gtkwave](https://sourceforge.net/u/gtkwave/profile/), I'd say that [gtkwave](https://github.com/gtkwave) here is him too. Actually, the email of user gtkwave is `bybell[at]rocketmail.com`.

Assuming that this is the new official repo:

- Is it expected to migrate/redirect the site from gtkwave.sourceforge.net to gtkwave.github.io?
- Are new releases to be published here only? Both on sourceforge and here?
- Should we stop using the SVN server on Sourceforge?
  - It seems that all the history is lost. Wouldn't it be sensible to export the SVN history to git instead of starting a *new* repo?
- Is there interest on having CI jobs (GitHub Actions) set up?
  - These might include multiple targets: GNU/Linux, macOS, Windows, docs, etc.
- Is it worth maintaining `gtkwave3` and `gtkwave3-gtk3` as subdirs of the same branch? Wouldn't it be better maintained by having two branches instead?
  - Note that, while branching/merging/rebasing is kind of broken in SVN, it is one of the main features of git.

gtkwave, 2019-12-25:
This is Tony here: yes, I set up the repo on here.  I didn't announce it as it's experimental for now.  Wilson Snyder asked if I could set up a repo as Verilator is also on github and it makes things easier for him automatically tracking changes to the FST writer code, he can submit patches directly to the repo, etc.  Currently, those changes only involve how his code would interface with gtkwave and no other users are affected.

Nothing else outside of those FST writer changes has been added since the 3.3.103 bugfix release.  I just merged them back into SVN as r1546.

That said, SVN for the time being still is the main repo.  I will likely migrate over to GitHub eventually and handle the SVN history properly then, but will probably wait until the GTK3 branch becomes the main one.  I still have one annoying GTK3 bug involving signal adds that I need to work around.  (The vertical scrollbar sometimes loses its position such that the new signal add is always at the bottom of the screen.)

As it stands right now, my day job is eating into what would otherwise be time to work on this.

-Tony

eine, 2019-12-26:
Hi! And thanks for such a quick reply!

> Verilator is also on github and it makes things easier for him automatically tracking changes to the FST writer code, he can submit patches directly to the repo, etc. 

I use gtkwave with [ghdl/ghdl](https://github.com/ghdl/ghdl) and [VUnit/vunit](https://github.com/VUnit/vunit). Since these projects are also on GitHub, I think that they can also benefit from being able to submit patches, cross-reference issues, etc.

Precisely, I maintain [ghdl/docker](https://github.com/ghdl/docker), where gtkwave is *packaged* along with those tools. There is [this](https://github.com/ghdl/docker/blob/master/broadway.sh) script that allows to use the GTK3 version of gtkwave in a web browser, through Broadway. I'd be glad to contribute either customized Dockerfiles, that script, or GitHub Workflows to this repo.

> That said, SVN for the time being still is the main repo. I will likely migrate over to GitHub eventually and handle the SVN history properly then, but will probably wait until the GTK3 branch becomes the main one.
> 
> As it stands right now, my day job is eating into what would otherwise be time to work on this.

I understand that it would be hard for you to maintain this *manual sync* workflow, should many different users contribute PRs here and others at sourceforge. Therefore, I'll refrain from opening GitHub specific PRs until you are ready to migrate *properly*.

If you are ok with it, we can leave this issue open for other users that might be curious as I was.

gtkwave, 2019-12-26:
Sure, please leave the issue open. 

Note that most users really don't contribute patches all that often, so syncing manually isn't too big a deal for me for the moment.  On SF, the project gtkwave has been in SVN since 2006 and it's only up to r1550 or so, with most of those updates being mine.  Outside of the update to GTK3 which required a lot of rewriting of code that interfaces to GTK* (two years ago), there haven't been a whole lot of updates.

In the past, most issues that have been opened were for feature adds.  I try to fix crash scenarios as soon as they've been identified to me.  Current bugs that I know of are mainly a handful of GTK3 vs GTK2 behavioral differences which I'll work around eventually.  Some of them might not be able to be worked around until supporting libraries are fixed.  Such cases are menu selections with touchscreens using X11, and DnD with Wayland.  Ironically, menu selections with touchscreens using X11 work perfectly fine in GTK2, but that's because the input events have been processed accordingly as GTK2 doesn't really support touchscreens.

(*) GTK major revision changes can be painful as the GTK developers keep redoing how things are done every single major revision.  At best I like to support one backlevel otherwise things become ifdef hell.  That's why there are two gtkwave3 directories: one supports GTK1/2 and the other one 2/3.  I'm sure GTK4 will be fairly involved to migrate to as well.  For the gtkwave3-gtk3 branch, I removed a lot of old dead code using cppp: https://www.muppetlabs.com/~breadbox/software/cppp.html

-Tony

Xiretza, 2021-05-02:
It's been one and a half years, is SVN still the main repo? If not, it's probably time to fix the git repo's history using a [graft](https://git.wiki.kernel.org/index.php/GraftPoint).

umarcor, 2021-05-02:
I didn't know about graft. Very interesting reference!

gtkwave, 2021-05-03:
Almost all development has been going on at the github side with commits cloned backward over to SVN. SF SVN pretty much is left around for legacy reasons. At some point I'm going to stop updating it completely and either consider it dead or delete it outright.

-Tony

wfjm, 2022-12-27:
We are now 19 month after Tony's last posting on this issue.  
And imho it's now time to consolidate the double repository situation.  

The [sourceforge repo](https://sourceforge.net/projects/gtkwave/) 
- has still releases, the last one 3.3.114 on 2022-12-18.
- is apparently heavily used, see [download statistics](https://sourceforge.net/projects/gtkwave/files/stats/timeline).
- and doesn't mention the existence of the github repo on the [home page](https://gtkwave.sourceforge.net/).

It seems prudent to me to organize the transition in steps
- first is to introduce releases in the github repository
- second is to add a 'github pages' equivalent of the sourceforge home page
- third is to mention the transition on the sourceforge home page
- and very last is to replace the sourceforge repo with a forwarding message

Unfortunately, the old revision history was not preserved when the GitHub repo was filled with content in commit https://github.com/gtkwave/gtkwave/commit/d4932822. But imho it is essential to introduce proper tags and releases in the GitHub repo as soon as possible.

gtkwave, 2022-12-29:
Walter,

SF's 3.3 version series is for the stable production-level version.  Its development intent/roadmap is officially bugfixes and dumpfile loader enhancements only.  I'm not dismantling the SF repo in the immediate future as I don't want another potential gtkwave 1.3 / 2.0 (abandonware) / 3.0 (from 1.3) version fragmentation situation similar to that which occurred in the early 2000s.  It caused far too much confusion.

GH is for the 3.4/4.x series which is where future software is headed.  Once GH gets up to a good level of QA, I'll pull the plug on SF and 3.3.  I use gtkwave on my day job.  When GH hosted source is good enough reliably to do debug of RTL for a full x86 CCX, it's certainly good enough for release.  That's the quality level I'm looking at as it's what users expect.  Yes, SF continues to see a reasonable number of downloads.

The GH repo initially was created only so GH projects such as Verilator could clone the FST loader from gtkwave automatically, and that's why both SF and GH were kept in sync for quite a while.  That's part of the reason it's not advertised: it was set up as a courtesy for other projects hosted on GH, however that obviously has been changing.  Moreover, I also don't know if it violates SF policy to refer to GH on an SF-owned resource and if it might have to be written in internal documentation in the distro, not directly on the SF site.  It's been forever since I've read SF's project hosting policies.

That's the long and short of it all with respect to repos and versions.

-Tony

 
 

   

   

Unfortunately, the old revision history was not preserved when the GitHub repo was filled with content in commit d4932822. But imho it is essential to introduce proper tags and releases in the GitHub repo as soon as possible.

AmeyaVS, 2023-03-22:
@gtkwave , I have recently migrated the GTKWave SVN History to Git.
Available here:[https://github.com/AmeyaVS/gtkwave-svn-archive](https://github.com/AmeyaVS/gtkwave-svn-archive).

If possible we can plan to change ownership to be moved under the gtkwave user/org.
Let me know if this is something you were looking out for.

Also, I have kept the `git-svn` conversion log in the branch [here](https://github.com/AmeyaVS/gtkwave-svn-archive/tree/git_conversion_log), just in-case any history is missing. From what I can tell the history should be mostly be complete.

AmeyaVS, 2023-03-22:
GitHub has an option to transfer ownership of  a repository as documented here:
https://docs.github.com/en/repositories/creating-and-managing-repositories/transferring-a-repository

Even I don't have necessary experience here.
But one way would be to just fork from my repository under the gtkwave user/org id for other people as a read-only repo.

Let me know your thoughts.

AmeyaVS, 2023-03-22:
Currently It is missing the tagged releases.
I can probably go trough the history to figure out release history till date, but that will take some time.
Let me know if that would also help.

rfuest, 2023-03-22:
I would suggest that to change `gtkwave` from a GitHub user into a GitHub organization, which will allow finer grained access control for multiple repositories: https://docs.github.com/en/account-and-profile/setting-up-and-managing-your-personal-account-on-github/managing-your-personal-account/converting-a-user-into-an-organization

rfuest, 2023-03-22:
Does the SVN repo also contain the CVS history? Otherwise it might be a good idea to also preserve that part of GTKWaves history in another repo.

gtkwave, 2023-03-22:
Sure, I can pass ownership one to whom or whatever as I don't want to be the long pole on holding anyone up.  I'll try to look at that in the next couple of days.  Thanks for the link.

-Tony

AmeyaVS, 2023-03-22:
> Does the SVN repo also contain the CVS history? Otherwise it might be a good idea to also preserve that part of GTKWaves history in another repo.

Sadly I don't think so, as the very first commit in the repo seems to be when the source was migrated away from CVS.

gtkwave, 2023-03-22:
I think that SF automatically shifted from CVS to SVN.  That was so long ago I can't remember, but I remember that I didn't have to do a whole lot for the change.

-Tony

gtkwave, 2023-03-22:
OK, thanks for the info.  I still have an xz or lzma "super tarball" laying around of all releases going back to 1.3.86 so I have some amount of history.

-Tony

Xiretza, 2023-03-22:
I was going to repeat [my suggestion of using a graft](https://github.com/gtkwave/gtkwave/issues/3#issuecomment-830862818), but it looks like the SVN and git histories have not been kept in sync, making this a much bigger mess than it was originally. It's probably too late to get a continuous history here, the best you can do is have a separate repo that contains all the SVN history.

rfuest, 2023-03-22:
I just tried to graft the GIT changes onto the SVN commits, but I couldn't get it to work fully: https://github.com/rfuest/gtkwave-graf
I've used `git replace --graft` to combine the GIT history with the last SVN commit before the GIT repository was created. This works locally, but GitHub doesn't show any of the SVN commits, because it does seem to ignore the replace ref. If you check out the repo you will still need to fetch the replace ref manually, as described here: https://stackoverflow.com/questions/42489421/how-to-push-a-git-replace-graft/42494208#42494208

If there isn't something I've missed, that would make the grafted repo easier to use, a separate repo is the best option IMO.

AmeyaVS, 2023-03-22:
I tried to extract the original CVS sources, and it seems to have been successful.
You can find the original CVS Archive here: https://github.com/AmeyaVS/gtkwave-cvs-archive

I think we can probably keep separate repositories for the history.

rfuest, 2023-03-22:
Would it be possible to merge the CVS and SVN history into one repository? The last commit in the CVS history is from 2011-01-25 and the first SVN one is from 2011-01-30.

Do you know why is there a CVSROOT directory inside the git repo?

AmeyaVS, 2023-03-22:
Hello @rfuest ,

You can try and merge the SVN and CVS repositories.
As for the CVSROOT directory I think it is an artifact from the cvs2git transform.

rfuest, 2023-03-22:
I've merged both repositories into one: https://github.com/rfuest/gtkwave-archive There are still a few things that need to be checked, but it's nice to have 16 years of GTKWave development in one repo:
[image]

Remaining issues:
- The summer of code branches/tags from CVS are missing.
- The repo needs a README to make clear that this repo is an archive.
- How should we handle the `gtkwave` directory? It was removed as part of the CVS to SVN transition, but is now visible in all revisions.

gtkwave, 2023-03-22:
That's impressive.
I wouldn't worry too much about summer of code.  That was where Google paid for a cooperative intern over the summer and he made the design more object based to support reload.  All the intermediate stuff as well as tools he wrote or modified to get there kind of are irrelevant as we're here where we are now and the past before that is mostly a historical curiosity.  He's probably the only guy who ever did branches on sourceforge for gtkwave.

Is the gtkwave directory just another extra level of subdirectory?  If so, I have no real preference on what's done with it.
-Tony

rfuest, 2023-03-22:
> I wouldn't worry too much about summer of code.  That was where Google paid for a cooperative intern over the summer and he made the design more object based to support reload.  All the intermediate [...]

OK. The branches were included in the original conversion by @AmeyaVS. I'll try to include them into the merged repository if it isn't too much effort.

> Is the gtkwave directory just another extra level of subdirectory?  If so, I have no real preference on what's done with it.

The `gtkwave` directory contains parts of an old version, which was marked as deprecated in May 2007. It hasn't been useable since the buildsystem was removed in this commit: https://github.com/rfuest/gtkwave-archive/commit/c650d0c815c0ebb66b0afb925a831f0ebfe77fba I guess it doesn't hurt to just keep it, even if anyone who might be interested in that code would need to look back into the history to get a buildable version of that directory.

AmeyaVS, 2023-03-23:
@rfuest , Please do share on how you got the history merged into one repo.

rfuest, 2023-03-24:
These are the commands I used to merge the repositories:
```sh
# Create a new empty repository:
git init gtkwave-archive
cd gtkwave-archive

# Add the CVS and SVN archives as remotes:
git remote add -f cvs-archive https://github.com/AmeyaVS/gtkwave-cvs-archive.git
git remote add -f svn-archive https://github.com/AmeyaVS/gtkwave-svn-archive.git

# Set HEAD to last SVN commit:
git reset --hard svn-archive/main

# Get reference for first SVN commit:
FIRST_SVN=$(git rev-list HEAD | tail -n1)
# FIRST_SVN: a60b973d7d1135f911c3293b3d74d7b53e496a0a

# Graft SVN commits onto CVS commits:
git replace --graft $FIRST_SVN cvs-archive/master

# Rewrite history to remove replace refs: 
git filter-repo --replace-refs delete-no-add --force
```

The last step requires `git-filter-repo` to be installed. I did also use `git filter-repo --mailmap` to fix Tony's email address, which was missing in the CVS conversion.

I'm not sure what I did wrong yesterday, but with these commands the SOC branches are kept and the `gtkwave` directory is removed in the first SVN commit.

AmeyaVS, 2023-03-24:
Hi @rfuest ,

I have updated(force pushed) the https://github.com/AmeyaVS/gtkwave-cvs-archive, along with author e-mails.
And, have used steps shared by you to create a combined gtkwave-archive: https://github.com/AmeyaVS/gtkwave-archive.

I think we don't need to execute the `git filter-repo --mailmap`, since the commit now have name and e-mails.

Also, created branches at the last `cvs-archive/master` and `cvs-archive/SummerOfCode2007`, along with the CVS Tags.

[image]

rfuest, 2023-03-24:
> I think we don't need to execute the git filter-repo --mailmap, since the commit now have name and e-mails.

Yes, all commits are now correctly linked to the `gtkwave` account, which wasn't the case with the missing e-mail address. There is still an inconsistency between the SVN and CVS commits: The SVN commits use some GitHub e-mail address, but as long as they are linked correctly to the user account in the GitHub UI this isn't a big issue IMO.

There seems to be a problem with the tags:
[image]

AmeyaVS, 2023-03-24:
Yes, when I migrated SVN repo to Git I used the GitHub generated e-mail for the migration.
But I think it should be consistent now if the e-mail associated with account have same e-mail it should be fine.

For the git tags I am not too sure why GitHub UI is complaining about them.
As, per my observations they are currently linked to unique commits branched off from the main branch by 
`cvs2git` commits.
 Maybe that is what is causing this for GitHub UI to complain that it is not on the main branch.

rfuest, 2023-03-24:
> For the git tags I am not too sure why GitHub UI is complaining about them.
> As, per my observations they are currently linked to unique commits branched off from the main branch by
> `cvs2git` commits.
> Maybe that is what is causing this for GitHub UI to complain that it is not on the main branch.

Maybe adding a branch for each tag would fix the issue.

AmeyaVS, 2023-03-24:
@rfuest , That did it. I have added branches along with the tags, and I currently do not see the warning from the GitHub UI.

rfuest, 2023-03-26:
I've creates tags for all versions in the CVS and SVN repositories: https://github.com/rfuest/gtkwave-archive/tags. They might not match the released source tarballs exactly, because their position is only based on the changelog entries and commit messages like "version bump", but I think they should be close enough to be useful to check out older versions.

AmeyaVS, 2023-09-08:
Is this repo going to become the official repo for gtkwave?
I see the history diverging from the SVN repository on the SF.
Should one merge the history with Git Development?, as it would make it easier to track changes using git-blame.

gtkwave, 2023-09-08:
SVN on SF currently exists *only* to track any bugfixes backported into the 3.3 codeline because SF posts binaries.  There is no ongoing 3.3 development.  Everything from here on out is on github.
-Tony
Message ID: ***@***.***>