gtkwave/gtkwave #5: Feature Request

jijingg, 2020-02-24:
I'm a Verid guy.  Recently, I am developing with SpinalHDL/Chisel,  so I have to consider some open-source waveform debugging tools.  I found gtkwave is good tool which exceeded my expectations. very smart and powerfull, especially cross-platform.

But I think there are still some things that can be improved

1. **ZoomIn/ZoomOut  Shortcut-Key** 

 those two actions with highest frequency, so i think should give a fast ShortCut-key 
- ZommIn   **"Z"** instead "Alt+Z", 
- ZommOut   **"z"** insteand "Alt+Shift+Z"

In addition, I think the mouse  left button is more convenient than the right button for zoomIn, 
Or left them to the users custom themeself  alos a better choice 

2. **add Groups** 

if there have large bundle of signals, creat **Group** is very necessary, so we can fold and unfold the group to focuse on the signal we track 
[image]

3. **Multi singnal move bug** 

when i multi select signal then drug by hold left-mouse-button only one signal be paster, Although copy and paste is also wrokable, but **ctrl + X** **ctrl + V** is a bit more cumbersome than drag

4. **Verilog Code track**

  add signal from Code , track the load and drive of signal by click on Verilog code  is a very productivity feature .   
It may not be possible in the short term. But in the long run, this feature is very attractive.

With the popularity of Chisel/SpinalHDL/verilator, i think more and more developer will use gtkwave.

gtkwave, 2020-02-24:
1) You're using an old version. Ctrl +/-/0 has been the default for over a year.
3.3.96 15nov18 Changed to standardized zoom in/out/full hotkeys.

2) Groups exist. Highlight a number of signals and press G. T will toggle it open/closed, or you can double-click on it.

3) Click on first signal then ctrl-click on various signals and/or shift click on last signal. Do not release the mouse click, move the pointer, and drop. It does more than one signal that way.

4) Someday I might have time for this, but I am extremely busy on my day job. Adding signals from code is possible if you drag/drop from rtlbrowse. See the man page for xml2stems and how it is used with Verilator.

-Tony

jijingg, 2020-02-25:
Thank's your reply,  thanks for your great work

closed 2020-05-05 by jijingg