gtkwave/gtkwave #11: GUI interface improvements

GlenNicholls, 2020-05-07:
1. Change groups from list toggled with `T` to a `+`, similar to commercial simulators. 
2. When creating group of signals, simply indenting signals one level will successfully remove the need to justify signals left or right. Especially with a `+`, it will be easy to tell if the signals are within a group.
3. Remove the "encapsulation" effect when creating a group. I.e. do not put the group name before and after the group of signals. This clutters the wave view and destroys efficiency when there are lots of groups/signals in the wave viewer.
4. Create a separate column for signal values that can be dragged to change the column width.
5. Create a different color scheme or some unique identifier for VHDL record types
6. Allow wave portion of horizontal GUI slider to extend right slightly beyond the final value of the waves.

I do not currently have access to simulator licenses, but I wanted to get these suggestions down before I forget about them. I will revisit this in the next few days to add screen captures to better explain each point. Once I get descriptions in, I can also break this out into multiple issues if you prefer.

just-now, 2020-05-31:
Is it possible to jump from signal to signal with some keyboard shortcut? 
For example, having signals:
```
                           V_____                     ___
--> clk                 /           \_________/       \
                                       
     data[7:0]
```
by pressing right arrow marker, V is moved to the next edge for clk:
```
                             ____V                     __
--> clk                 /           \_________/       \
                                       
     data[7:0]
```
by pressing down arrow, next signal has to be selected:
```
                             ____V                     __
     clk                 /           \_________/       \
                                       
--> data[7:0]
```

Can we have such feature?

GlenNicholls, 2020-08-18:
Another feature that is desperately required is to maintain original case for signal names. I assume the binary files that feed GTK from GHDL/Symbiyosys/etc. are all lower case, but this really needs to be changed. When I run sims with GHDL, all the VHDL signal names and entity names are lower case which makes reading signal names in the wave viewer extremely difficult and frustrating.

gtkwave, 2020-08-19:
That has nothing to do with gtkwave: it would be whatever is done by the simulator's dumper or prior to what feeds it. For example, if I do a hex dump of a GHW file, the signals can be seen as stored in all lower case. The best I could do in gtkwave is have an option to convert them to uppercase, or something like the 1st letter encountered is uppercase and all letters in a run after it are in lowercase with uppercase starting again after the next non-letter is encountered in the string.

I have no idea if internally GHDL normalizes all signals to one or the other case such that the original case as declared has been destroyed, but I would think it does that it might in order to aid in processing. Tristan would have to answer that question.

I've never used Symbiyosys. Is it feeding GHDL it? 

I just looked in a WLF file from Modelsim and it appears to preserve case.

-Tony

umarcor, 2020-08-19:
In fact, I thought that this thread was in ghdl/ghdl... I believe that GHDL removes all the comments and lowercases all the identifiers (because VHDL is case insensitive) during the parsing and building the AST. So, the original names are long gone when the wave dump uses them.

*ping* @tgingold

GlenNicholls, 2020-08-19:
@gtkwave that is what I thought, just wanted to get the request jotted down yesterday when I noticed it. I’ll ask Tristian if there’s a way to preserve case as it makes it difficult to read camel case names in the wave viewer because of this

ercanal, 2020-08-21:
It will be great to have to send the signals into new tab. Otherwise 
- Open new tab
- Select same Results file
- Find the signal
- plot

GlenNicholls, 2020-08-27:
Based on my promise to provide screen-caps to describe the above issues, here it goes. This will be periodically updated.

## Change groups from list toggled with T to a +, similar to commercial simulators.
As you can see below, I have created groups of signals denoted by the red triangle. To expand this group, there is a `+` that I click. This expansion is identical for signals/records as well. 

[image]

In GTKWave, however, grouped signals require pressing the `T` alpha-numeric key to expand and collapse. In addition to this, the closing bracket `}` on the group of signals makes it hard to quickly discern different groups when there are multiple.

[image]

As you can see, it is not as intuitive to navigate when groups become large. Instead, GTK should instead provide something similar to the first screen capture to make the interface easier to navigate when debugging. The other advantage of this format is it would be easier to nest groups together without making them confusing as the tool should indent their levels.

## When creating a group of signals, simply indenting signals one level will successfully remove the need to justify signals left or right. Especially with a +, it will be easy to tell if the signals are within a group.
From the first screen-cap above, you can see that there is no need for right or left justification of signals as they are indented like a file-view. In the GTK capture, I right justify so I can actually see signals more clearly, but it is ugly as with large data-widths, I have to move the bar over super far to see more of the values in the signal where the marker is (reference `o_snbwsq`)

## Remove the "encapsulation" effect when creating a group. I.e. do not put the group name before and after the group of signals. This clutters the wave view and destroys efficiency when there are lots of groups/signals in the wave viewer.
This should be obvious from the GTKWave capture above. This comes for free if a similar style to ModelSim/ActiveHDL is adopted.

## Create a separate column for signal values that can be dragged to change the column width.
In the following capture, you can see there is a separate column for the signal values during simulation (Noted with `Msgs` tag):

[image]

Thus, I can make the signal list column width specific to include the longest signal names and then adjust the value column to see the longest width value. Bonus points if there is a command to adjust both with TCL commands.

## Create some unique identifier for VHDL record types and vectors. As proposed above, the + would expand the record to show record elements or simarlarly with vectors/arrays, the different bits/vectors
When adding signals to the wave-viewer, records should be added similarly with the `+` suggested above. ModelSim looks like:

[image]

The most painful experience I have in GTK is the following situation for a 2-dimensional array or an array of arrays:

[image]

As you can see, it just blows up the waveform viewer and makes debugging nearly impossible when I can't collapse the `BpskCollapse` signal for example. I played around with making this a group, but because it isn't indented, it's really difficult to tell if I accidentally scrolled to a different expanded group by accident.

## Add TCL interpreter shell within the GUI. A tab at the bottom of the GUI that would open/close the TCL interpreter would make it really easy to develop scripts for the GUI. Development for automation is slow when I constantly have to close and re-run with changes in my script(s).
In ModelSim, there is a transcript window that runs TCL commands by the user:

[image]

It would be really nice if GTK had something similar to test easier and it would prevent the need to print information to the command window. I'm not sure if GHDL dumps the messages or GTK, but if it's GHDL then I understand the added complexity for defining more input files is not practical considering other tools use GTK. with that said, there should be a command window at the very least that allows the user to run TCL/GTK commands directly from the GUI.

I think that is everything, let me know if there are any questions about any of these clarifications. Overall I really like the tool, but these are my biggest complaints while using it that hinder productivity.

gtkwave, 2020-08-29:
I'll look at your suggestions. I have access to both ModelSim and Verdi so I can see how they deal with those ergonomic issues. 

Note that for an interactive Tcl shell, you should be able to add --wish to the command line. This won't work for all usage cases, but an interactive facility does exist.

-Tony

515:/home/bybell/gtkwave/code/gtkwave3-gtk3/examples> gtkwave des.gtkw --wish

GTKWave Analyzer v3.3.107 (w)1999-2020 BSI

FSTLOAD | Processing 1432 facs.
FSTLOAD | Built 1287 signals and 145 aliases.
FSTLOAD | Building facility hierarchy tree.
FSTLOAD | Sorting facility hierarchy tree.
Interpreter id is gtkwave_10731
% gtkwave::/File/Reload_Waveform
GTKWAVE | Reloading waveform...
FSTLOAD | Processing 1432 facs.
FSTLOAD | Built 1287 signals and 145 aliases.
FSTLOAD | Building facility hierarchy tree.
FSTLOAD | Sorting facility hierarchy tree.
GTKWAVE | ...waveform reloaded
% gtkwave::/File/Quit
Exiting.