gtkwave/gtkwave #12: Create Gitter channel

GlenNicholls, 2020-05-07:
If/when GTKWave moves to GitHub, it would be nice to have a Gitter channel. Instead of users reaching out to you directly with questions, it would facilitate the community fielding responses. This will also mitigate questions cluttering the issues.

On that same note, what is the reasoning for the continued support for the legacy version? I'm not familiar with GTK's architecture at all, but I have always wondered why backward compatibility is so important for this tool.

eine, 2020-05-07:
@GlenNicholls, see https://github.com/gtkwave/gtkwave/issues/3#issuecomment-569139993.

gtkwave, 2020-05-07:
I've read all your other notes. I'm too busy with the day job to reply to them at the moment.

As far as supporting the legacy versions, there are still some intermittent bugs in the GTK 3 toolkit behavior acting glitchy with respect to scrollbar function such as when adding signals on drag and drop from the SST to the signal window, and instead of the window scrolling where it should, it's wrong and the signals I added are at the very bottom of the screen. Touchscreen support is messed up in GTK3 as well. Some features such as accessing menus in X11 via touchscreen simply don't work and this is not a gtkwave issue but GTK3. DnD (at least the last I checked) from the touchscreen is broken on Wayland for GTK3. I have a file changes_from_gtk2.txt with a to do list I need to address. When/if I fix most of that, I can freeze the old GTK1/2 branch for good.

I don't know if I'll ever port to GTK4. GTK3 was necessary pain in order to get proper touchscreen support such as pinch zoom/enlarge, fling, etc., but the developers appear more interesting in working on shiny new features for GTK4 than in addressing what is wrong with the toolkit in GTK3. 

There is a video out there https://www.youtube.com/watch?v=ON0A1dsQOV0 that addresses a laundry list of issues that have plagued GTK due to how the project is managed and/or the personalities and priorities of its developers. I really liked GTK2, but from GTK3 on, my experiences with the toolkit largely mirrored those of Dirk Hohndel in his talk. At some point I might bite the bullet and abandon the toolkit completely as I have better things to do with my time than to fight a toolkit that by design or neglect proves to be hostile toward its user base. See also here: https://en.wikipedia.org/wiki/Audacious_(software)#Change_to_C++_and_Qt

-Tony