gtkwave/gtkwave #13: Change font size in  waveform view

fjullien, 2020-05-09:
Hi,

Is there an easy way to increase font size (and may be wave height at the same time) in the waveform view ? When working late at night it would be helpful :)

Thanks.

gtkwave, 2020-05-09:
You can put changes for fontname_signals/waves in your .gtkwaverc file or do it at the command line like this:

534:/home/bybell/gtkwave/code/gtkwave3-gtk3/src> gtkwave ../examples/des.gtkw --rcvar 'fontname_signals Monospace 12' --rcvar 'fontname_waves Monospace 12'

-Tony

fjullien, 2020-05-09:
`--rcvar` doesn't work for me, my version must be to old (v3.3.68).
Anyway, I created a .gtkwaverc file and changed fonts there. It works :+1: 
It would be great to be able to change font size dynamically.

gtkwave, 2020-05-09:
3.3.68 is almost 5 years old. Is that from a distribution or just a download you have on your computer?

I'll look at dynamic font size changes at some point. It's not really all that difficult to do. I actually wouldn't mind it for the reverse of what you need where I could occasionally put an extra couple of rows of signals on the screen.

Thanks,
-Tony

fjullien, 2020-05-09:
> It's not really all that difficult to do
Yes that's what I was thinking.

My gtkwave comes from Ubuntu 18.04 but there is something strange.
apt says gtkwave is up-to-date and is 3.3.86 but my program version is 3.3.68. Anyway...

Looking forward to see this change in gtkwave !

jotego, 2020-11-14:
I would use this new feature too.

gtkwave, 2020-11-14:
You can try something like this using rc vars:

gtkwave des.gtkw --rcvar 'fontname_signals Monospace 8' --rcvar 'fontname_waves Monospace 8'

-Tony

ppaulweber, 2021-06-30:
Hi @gtkwave, I've just tested the `--rcvar` option and it does work for me, but in order to read properly the signal values in the wave you should use one point bigger font for the signal font than the wave font, e.g.:
 
```
$ gtkwave -A --rcvar 'fontname_signals Monospace 13' --rcvar 'fontname_waves Monospace 12' trace.vcd trace.gtkw
$ gtkwave --version
GTKWave Analyzer v3.3.109 (w)1999-2020 BSI
```

gtkwave, 2021-06-30:
What system are you on? (OS, gtk version, X11 vs wayland, etc) The same size for both should work fine. It's been that way since 1998 so I can curious what is going on with your setup. Do you have a screen grab?

-Tony

ppaulweber, 2021-06-30:
Hi, I'm using an ArchLinux (`5.12.13-arch1-2`), GTK2 (`2.24.33`), GTK3 (`3.24.29`), and X11 (`1.20.11`) setup for `gtkwave` (`v3.3.109`).
Of course, here you go:

```
$ gtkwave -A --rcvar 'fontname_signals Monospace 12' --rcvar 'fontname_waves Monospace 12' \
trace.vcd trace.gtkw
```
[image]

```
$ gtkwave -A --rcvar 'fontname_signals Monospace 13' --rcvar 'fontname_waves Monospace 12' \
trace.vcd trace.gtkw
```
[image]

FYI: If I start `gtkwave` without the `--rcvar` option, the chosen default font is rendering the signals names, signal values, wave etc. as expected with proper spacing.

gtkwave, 2021-06-30:
That's weird. I wonder if Pango on your system has different attributes for the ascent/descent line for whatever font it maps to Monospace. See attached screen grabs from my system.

-Tony

void XXX_font_engine_draw_string
(cairo_t *cr,
struct font_engine_font_t *font,
wave_rgb_t *gc,
gint x,
gint y,
const gchar *string)
{
/* PangoRectangle ink,logical; */

pango_layout_set_text(GLOBALS->fonts_layout, string, -1);
pango_layout_set_font_description(GLOBALS->fonts_layout, font->desc);
/* pango_layout_get_extents(GLOBALS->fonts_layout,&ink,&logical); */

cairo_set_source_rgba (cr, gc->r, gc->g, gc->b, gc->a);
cairo_move_to (cr, x, y-font->ascent);
pango_cairo_show_layout (cr, GLOBALS->fonts_layout);
}

VasantJ3, 2022-09-08:
Could someone help with exactly how to scale the font size? I am using a 4K resolution laptop screen, and in that GTKwave is not being scaled. The fonts and signals are very very small.

gtkwave, 2022-09-08:
See if you can experiment with size numbers like this:
gtkwave whatever.vcd --rcvar 'fontname_signals Monospace 20' --rcvar 'fontname_waves Monospace 18'
-Tony

wotupfoo, 2024-04-06:
Please add this feature (especially as a .gtkw project setting.
When, on Windows, you're using the `apio` toolkit for Lattice ICE40 development it's all wrapped behind the apio executable, so you can't be adding --rcvar to the command line or add a .gtkwaverc file
Thx.

rfuest, 2024-04-07:
> ... especially as a .gtkw project setting. ...

I don't think that the font size should be a setting the the project file. Project files can be shared between different users who might prefer very different settings for the font size. But I'm planning to make the font size configurable inside the GUI in GTKWave 4.

> When, on Windows, you're using the `apio` toolkit for Lattice ICE40 development it's all wrapped behind the apio executable, so you can't be adding --rcvar to the command line or add a .gtkwaverc file

You should be able to set the font size in the global gtkwaverc file. I've never used this feature in Windows, but I did check the code and I think the gtkwaverc needs to be put in this path: `%USERPROFILE%/gtkwave.ini` (e.g. `C:\Users\username\gtkwave.ini`)