gtkwave/gtkwave #15: Fails to build with GCC 10

aelmahmoudy, 2020-05-25:
gtkwave fails to build with gcc-10/g++-10, but succeeds to build with gcc-9/g++-9.

The full build log can be found at:
http://people.debian.org/~doko/logs/gcc10-20200225/gtkwave_3.3.103-1_unstable_gcc10.log

{{{
g++ -I./.. -I./helpers  -pthread -I/usr/include/gtk-2.0 -I/usr/lib/x86_64-linux-gnu/gtk-2.0/include -I/usr/include/pango-1.0 -I/usr/include/atk-1.0 -I/usr/include/cairo -I/usr/include/pixman-1 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/pango-1.0 -I/usr/include/harfbuzz -I/usr/include/pango-1.0 -I/usr/include/fribidi -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/uuid -I/usr/include/freetype2 -I/usr/include/libpng16 -I./liblzma -D_WAVE_HAVE_XZ   -I./helpers/fst   -I/usr/include/tcl8.6 -DHAVE_LIBTCL -I/usr/include/tcl8.6   -D_WAVE_HAVE_JUDY   -DWAVE_HAVE_GSETTINGS  -I./cocoa -pthread -I/usr/include/gtk-unix-print-2.0 -I/usr/include/gtk-2.0 -I/usr/lib/x86_64-linux-gnu/gtk-2.0/include -I/usr/include/pango-1.0 -I/usr/include/atk-1.0 -I/usr/include/cairo -I/usr/include/pixman-1 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/pango-1.0 -I/usr/include/harfbuzz -I/usr/include/pango-1.0 -I/usr/include/fribidi -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/uuid -I/usr/include/freetype2 -I/usr/include/libpng16 -DWAVE_GTK_UNIX_PRINT -g -O2 -fdebug-prefix-map=/<<PKGBUILDDIR>>=. -fstack-protector-strong -Wformat -Werror=format-security  -Wl,-z,relro -Wl,--as-needed -o gtkwave fsdb_wrapper_api.o globals.o ae2.o analyzer.o baseconvert.o bitvec.o bsearch.o busy.o clipping.o color.o currenttime.o debug.o discardbuttons.o edgebuttons.o entry.o extload.o fetchbuttons.o fgetdynamic.o file.o fonts.o fst.o gconf.o getopt.o getopt1.o ghw.o ghwlib.o hierpack.o jrb.o help.o lxt2_read.o lxt_write.o vzt_read.o hiersearch.o interp.o logfile.o lx2.o lxt.o main.o markerbox.o menu.o mouseover.o mouseover_sigs.o pagebuttons.o pipeio.o pixmaps.o print.o ptranslate.o rc.o regex.o renderopt.o rgb.o savefile.o search.o shiftbuttons.o showchange.o signalwindow.o simplereq.o splash.o status.o strace.o symbol.o tcl_commands.o tcl_helper.o tcl_np.o tcl_support_commands.o timeentry.o translate.o tree.o treesearch.o tree_component.o ttranslate.o vcd.o vcd_keywords.o vcd_partial.o vcd_recoder.o vcd_saver.o vlist.o vzt.o wavewindow.o zoombuttons.o ./cocoa/libgtkwmacintegration.a  -lgtk-x11-2.0 -lgdk-x11-2.0 -lpangocairo-1.0 -latk-1.0 -lcairo -lgdk_pixbuf-2.0 -lgio-2.0 -lpangoft2-1.0 -lpango-1.0 -lgobject-2.0 -lglib-2.0 -lfontconfig -lfreetype ./liblzma/libgwlzma.a -llzma -lz -lbz2 ./helpers/fst/libfst.a  -L/usr/lib/x86_64-linux-gnu -ltcl8.6 -L/usr/lib/x86_64-linux-gnu -ltk8.6  -lJudy   -lgtk-x11-2.0 -lgdk-x11-2.0 -lpangocairo-1.0 -latk-1.0 -lcairo -lgdk_pixbuf-2.0 -lgio-2.0 -lpangoft2-1.0 -lpango-1.0 -lgobject-2.0 -lglib-2.0 -lfontconfig -lfreetype  -lnsl -lpthread -lm -ldl
/usr/bin/ld: ae2.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: analyzer.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: baseconvert.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: bitvec.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: bsearch.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: busy.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: clipping.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: color.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: currenttime.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: debug.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: discardbuttons.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: edgebuttons.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: entry.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: extload.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: fetchbuttons.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: fgetdynamic.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: file.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: fonts.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: fst.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: gconf.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: ghw.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: hierpack.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: help.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: hiersearch.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: logfile.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: lx2.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: lxt.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: main.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: markerbox.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: menu.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: mouseover.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: mouseover_sigs.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: pagebuttons.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: pipeio.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: pixmaps.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: print.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: ptranslate.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: rc.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: regex.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: renderopt.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: rgb.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: savefile.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: search.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: shiftbuttons.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
[... cut ...]
/usr/bin/ld: vlist.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: vzt.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: wavewindow.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: zoombuttons.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
collect2: error: ld returned 1 exit status
}}}

gtkwave, 2020-05-25:
See if the following fixes it.

Thanks,
-Tony

132:/home/bybell/git/gtkwave> git commit -a
[master b12fb91] Remove extern from wave_rpc_id in gconf.h.
 4 files changed, 6 insertions(+), 4 deletions(-)
133:/home/bybell/git/gtkwave> git push origin master
Username for 'https://github.com': gtkwave
Password for 'https://gtkwave@github.com': 
Counting objects: 17, done.
Delta compression using up to 4 threads.
Compressing objects: 100% (9/9), done.
Writing objects: 100% (9/9), 857 bytes | 0 bytes/s, done.
Total 9 (delta 8), reused 0 (delta 0)
remote: Resolving deltas: 100% (8/8), completed with 8 local objects.
To https://github.com/gtkwave/gtkwave.git
 abae31f..b12fb91 master -> master

 
 

http://people.debian.org/~doko/logs/gcc10-20200225/gtkwave_3.3.103-1_unstable_gcc10.log

{{{
g++ -I./.. -I./helpers -pthread -I/usr/include/gtk-2.0 -I/usr/lib/x86_64-linux-gnu/gtk-2.0/include -I/usr/include/pango-1.0 -I/usr/include/atk-1.0 -I/usr/include/cairo -I/usr/include/pixman-1 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/pango-1.0 -I/usr/include/harfbuzz -I/usr/include/pango-1.0 -I/usr/include/fribidi -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/uuid -I/usr/include/freetype2 -I/usr/include/libpng16 -I./liblzma -D_WAVE_HAVE_XZ -I./helpers/fst -I/usr/include/tcl8.6 -DHAVE_LIBTCL -I/usr/include/tcl8.6 -D_WAVE_HAVE_JUDY -DWAVE_HAVE_GSETTINGS -I./cocoa -pthread -I/usr/include/gtk-unix-print-2.0 -I/usr/include/gtk-2.0 -I/usr/lib/x86_64-linux-gnu/gtk-2.0/include -I/usr/include/pango-1.0 -I/usr/include/atk-1.0 -I/usr/include/cairo -I/usr/include/pixman-1 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/pango-1.0 -I/usr/include/harfbuzz -I/usr/include/pango-1.0 -I/usr/include/fribidi -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/uuid -I/usr/include/freetype2 -I/usr/include/libpng16 -DWAVE_GTK_UNIX_PRINT -g -O2 -fdebug-prefix-map=/<>=. -fstack-protector-strong -Wformat -Werror=format-security -Wl,-z,relro -Wl,--as-needed -o gtkwave fsdb_wrapper_api.o globals.o ae2.o analyzer.o baseconvert.o bitvec.o bsearch.o busy.o clipping.o color.o currenttime.o debug.o discardbuttons.o edgebuttons.o entry.o extload.o fetchbuttons.o fgetdynamic.o file.o fonts.o fst.o gconf.o getopt.o getopt1.o ghw.o ghwlib.o hierpack.o jrb.o help.o lxt2_read.o lxt_write.o vzt_read.o hiersearch.o interp.o logfile.o lx2.o lxt.o main.o markerbox.o menu.o mouseover.o mouseover_sigs.o pagebuttons.o pipeio.o pixmaps.o print.o ptranslate.o rc.o regex.o renderopt.o rgb.o savefile.o search.o shiftbuttons.o showchange.o signalwindow.o simplereq.o splash.o status.o strace.o symbol.o tcl_commands.o tcl_helper.o tcl_np.o tcl_support_commands.o timeentry.o translate.o tree.o treesearch.o tree_component.o ttranslate.o vcd.o vcd_keywords.o vcd_partial.o vcd_recoder.o vcd_saver.o vlist.o vzt.o wavewindow.o zoombuttons.o ./cocoa/libgtkwmacintegration.a -lgtk-x11-2.0 -lgdk-x11-2.0 -lpangocairo-1.0 -latk-1.0 -lcairo -lgdk_pixbuf-2.0 -lgio-2.0 -lpangoft2-1.0 -lpango-1.0 -lgobject-2.0 -lglib-2.0 -lfontconfig -lfreetype ./liblzma/libgwlzma.a -llzma -lz -lbz2 ./helpers/fst/libfst.a -L/usr/lib/x86_64-linux-gnu -ltcl8.6 -L/usr/lib/x86_64-linux-gnu -ltk8.6 -lJudy -lgtk-x11-2.0 -lgdk-x11-2.0 -lpangocairo-1.0 -latk-1.0 -lcairo -lgdk_pixbuf-2.0 -lgio-2.0 -lpangoft2-1.0 -lpango-1.0 -lgobject-2.0 -lglib-2.0 -lfontconfig -lfreetype -lnsl -lpthread -lm -ldl

aelmahmoudy, 2020-05-25:
Still fails with this error:

{{{
g++-10 -I./.. -I./helpers  -pthread -I/usr/include/gtk-2.0 -I/usr/lib/x86_64-linux-gnu/gtk-2.0/include -I/usr/include/pango-1.0 -I/usr/include/atk-1.0 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/pango-1.0 -I/usr/include/fribidi -I/usr/include/cairo -I/usr/include/pixman-1 -I/usr/include/harfbuzz -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/uuid -I/usr/include/freetype2 -I/usr/include/libpng16 -I./liblzma -D_WAVE_HAVE_XZ   -I./helpers/fst   -I/usr/include/tcl8.6 -DHAVE_LIBTCL -I/usr/include/tcl8.6   -D_WAVE_HAVE_JUDY   -DWAVE_HAVE_GSETTINGS  -I./cocoa -pthread -I/usr/include/gtk-unix-print-2.0 -I/usr/include/gtk-2.0 -I/usr/lib/x86_64-linux-gnu/gtk-2.0/include -I/usr/include/pango-1.0 -I/usr/include/atk-1.0 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/pango-1.0 -I/usr/include/fribidi -I/usr/include/cairo -I/usr/include/pixman-1 -I/usr/include/harfbuzz -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/uuid -I/usr/include/freetype2 -I/usr/include/libpng16 -DWAVE_GTK_UNIX_PRINT -g -O2 -fdebug-prefix-map=/builds/electronics-team/gtkwave/debian/output/gtkwave-3.3.104=. -fstack-protector-strong -Wformat -Werror=format-security  -Wl,-z,relro -Wl,--as-needed -o gtkwave fsdb_wrapper_api.o globals.o ae2.o analyzer.o baseconvert.o bitvec.o bsearch.o busy.o clipping.o color.o currenttime.o debug.o discardbuttons.o edgebuttons.o entry.o extload.o fetchbuttons.o fgetdynamic.o file.o fonts.o fst.o gconf.o getopt.o getopt1.o ghw.o ghwlib.o hierpack.o jrb.o help.o lxt2_read.o lxt_write.o vzt_read.o hiersearch.o interp.o logfile.o lx2.o lxt.o main.o markerbox.o menu.o mouseover.o mouseover_sigs.o pagebuttons.o pipeio.o pixmaps.o print.o ptranslate.o rc.o regex.o renderopt.o rgb.o savefile.o search.o shiftbuttons.o showchange.o signalwindow.o simplereq.o splash.o status.o strace.o symbol.o tcl_commands.o tcl_helper.o tcl_np.o tcl_support_commands.o timeentry.o translate.o tree.o treesearch.o tree_component.o ttranslate.o vcd.o vcd_keywords.o vcd_partial.o vcd_recoder.o vcd_saver.o vlist.o vzt.o wavewindow.o zoombuttons.o ./cocoa/libgtkwmacintegration.a  -lgtk-x11-2.0 -lgdk-x11-2.0 -lpangocairo-1.0 -latk-1.0 -lcairo -lgdk_pixbuf-2.0 -lgio-2.0 -lpangoft2-1.0 -lpango-1.0 -lgobject-2.0 -lglib-2.0 -lharfbuzz -lfontconfig -lfreetype ./liblzma/libgwlzma.a -llzma -lz -lbz2 ./helpers/fst/libfst.a  -L/usr/lib/x86_64-linux-gnu -ltcl8.6 -L/usr/lib/x86_64-linux-gnu -ltk8.6  -lJudy   -lgtk-x11-2.0 -lgdk-x11-2.0 -lpangocairo-1.0 -latk-1.0 -lcairo -lgdk_pixbuf-2.0 -lgio-2.0 -lpangoft2-1.0 -lpango-1.0 -lgobject-2.0 -lglib-2.0 -lharfbuzz -lfontconfig -lfreetype  -lnsl -lpthread -lm -ldl
/usr/bin/ld: ae2.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: analyzer.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: baseconvert.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: bitvec.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: bsearch.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: busy.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: clipping.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: color.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: currenttime.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: debug.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: discardbuttons.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: edgebuttons.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: entry.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: extload.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: fetchbuttons.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: fgetdynamic.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: file.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: fonts.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: fst.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: gconf.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: ghw.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
[... cut ...]
/usr/bin/ld: vlist.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: vzt.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: wavewindow.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
/usr/bin/ld: zoombuttons.o:./src/gconf.h:29: multiple definition of `wave_rpc_id'; globals.o:./src/gconf.h:29: first defined here
collect2: error: ld returned 1 exit status
}}}

aelmahmoudy, 2020-05-25:
See: https://salsa.debian.org/electronics-team/gtkwave/-/jobs/765097/raw

gtkwave, 2020-05-25:
I reverted back to what was originally in git as it looks correct, see here: https://gcc.gnu.org/gcc-10/porting_to.html

What is this?

| dpkg-source: info: applying gcc10-extern.patch

...is the compiler broken? I tried the -fno-common flag on my compile and see no problems with the original source.

-Tony