gtkwave/gtkwave #18: Making independent binary package for gtkwave3-gtk3

e-tabrez, 2020-06-07:
Hello,

Thanks for your efforts for this very useful tool.

gtkwave3-gtk3 has much better look and feel but binary distribution is not available for it.

So to use gtkwave3-gtk3, I have build  it MSYS2 on Win10. 
Its installed on MSYS2's in mingw64 directory and working fine.

Now I want to distribute gtkwave3-gtk3 binaries to my team members so they can use without MSYS2 (just like binary distribution available for gtkwave3 on SourceForge).

It will be helpful to add packaging instruction for gtkwave3-gtk3 (with list of dependencies dll etc)
so we can make independently usable gtkwave3-gtk3 binary distribution.

gtkwave, 2020-06-07:
For the dependencies, I don't have msys handy, but I think the ldd command might work on that as well. Other than that, you need to see what's on the sourceforge site for this and copy over the appropriate directories for things such as Tcl, etc. The lib/ and share/ directories are the tricky ones. See here:

https://sourceforge.net/projects/gtkwave/files/gtkwave-3.3.100-bin-win64/

I don't have an explicit build process for windows zip files. I got it working the one time then after that simply copy over the new executables for each release.

-Tony

e-tabrez, 2020-06-07:
Thanks for reply.

I just managed to make a win64 distribution for gtkwave3-gtk3 after many trial and errors.
I referred the directories and files available in gtkwave-3.3.100-bin-win64 (SourceForge package) and copied the required files from MSYS2 mingw64 to my distribution package.
Its working so far. 
But only thing is, since I am not sure which DLLs are needed, I had to copy all the DLLs from msys64\mingw64\bin\.
Also same for msys64\mingw64\share\icons\.

This makes my distribution zip package more than double in size (58 MB) compared to SourceForge package.
But not bad for now.

It would be nice to make official distribution packages for gtkwave3-gtk3 also on SourceForge.

closed 2020-06-07 by e-tabrez

gtkwave, 2020-06-07:
See if ldd works. I thought it did, but I might be wrong. For example on linux:

273:/home/bybell> ldd /usr/local/bin/gtkwave 
 linux-vdso.so.1 => (0x00007ffc6b1f2000)
 libgtk-3.so.0 => /lib64/libgtk-3.so.0 (0x00007fc0aaeaf000)
 libgdk-3.so.0 => /lib64/libgdk-3.so.0 (0x00007fc0aabc5000)
 libatk-1.0.so.0 => /lib64/libatk-1.0.so.0 (0x00007fc0aa99f000)
 libgio-2.0.so.0 => /lib64/libgio-2.0.so.0 (0x00007fc0aa600000)
<etc>

-Tony

e-tabrez, 2020-06-07:
Actually ldd didn't helped much.
I did ldd on gtkwave.exe. This is the output.

$ ntldd gtkwave.exe
        ADVAPI32.dll => C:\Windows\SYSTEM32\ADVAPI32.dll (0x0000000000720000)
        comdlg32.dll => C:\Windows\SYSTEM32\comdlg32.dll (0x0000000000720000)
        KERNEL32.dll => C:\Windows\SYSTEM32\KERNEL32.dll (0x00000000017d0000)
        msvcrt.dll => C:\Windows\SYSTEM32\msvcrt.dll (0x00000000017d0000)
        libbz2-1.dll => C:\msys64\mingw64\bin\libbz2-1.dll (0x0000000000190000)
        libcairo-2.dll => C:\msys64\mingw64\bin\libcairo-2.dll (0x00000000017d00
        libgdk-3-0.dll => C:\msys64\mingw64\bin\libgdk-3-0.dll (0x0000000001ae00
        libgdk_pixbuf-2.0-0.dll => C:\msys64\mingw64\bin\libgdk_pixbuf-2.0-0.dll
        libglib-2.0-0.dll => C:\msys64\mingw64\bin\libglib-2.0-0.dll (0x00000000
        libgobject-2.0-0.dll => C:\msys64\mingw64\bin\libgobject-2.0-0.dll (0x00
        libgtk-3-0.dll => C:\msys64\mingw64\bin\libgtk-3-0.dll (0x0000000001ae00
        libintl-8.dll => C:\msys64\mingw64\bin\libintl-8.dll (0x0000000001d10000
        liblzma-5.dll => C:\msys64\mingw64\bin\liblzma-5.dll (0x0000000000190000
        libpango-1.0-0.dll => C:\msys64\mingw64\bin\libpango-1.0-0.dll (0x000000
        libpangocairo-1.0-0.dll => C:\msys64\mingw64\bin\libpangocairo-1.0-0.dll
        tcl86.dll => C:\msys64\mingw64\bin\tcl86.dll (0x0000000001ae0000)
        tk86.dll => C:\msys64\mingw64\bin\tk86.dll (0x0000000001ae0000)
        zlib1.dll => C:\msys64\mingw64\bin\zlib1.dll (0x00000000001e0000)

I copied above listed DLLs from C:\msys64\mingw64\bin to my distribution pkg.
But after running gtkwave.exe from distribution pkg , many error keeps coming showing other DDLs are missing.

So I copied all the DLLs available in C:\msys64\mingw64\bin (total 110) to my distribution pkg.
GTKwave is working good with this.

There my be many unused and unnecessary DLLs. But for now its OK.

umarcor, 2020-06-07:
Hi @e-tabrez! The default package manager in MSYS2 is pacman. Hence, build/install recipes are normally provided as PKGBUILD files. In fact, there's already a PKGBUILD file for gtkwave since 2017 (msys2/MINGW-packages#2071). I can confirm that the package works because I have been using it in University courses for students to get gtkwave easily:

```sh
# pacman -Ss gtkwave
mingw32/mingw-w64-i686-gtkwave 3.3.104-1
    GTKWaveGTKWave is a fully featured GTK+ based wave viewer for Unix, Win32, and Mac OSX
mingw64/mingw-w64-x86_64-gtkwave 3.3.104-1 [installed]
    GTKWaveGTKWave is a fully featured GTK+ based wave viewer for Unix, Win32, and Mac OSX
```

Hence, I would strongly suggest to please (PLEASE) contribute the gtk3 variant that you have written, as a new official package of MSYS2. That is, to open a PR in [msys2/MINGW-packages](https://github.com/msys2/MINGW-packages). That will allow all MSYS2 users to easily install it.

On the other hand, I am very interested on some standard procedure to distribute "*binaries to my team members so they can use without MSYS2*". Precisely, in the context of [ghdl/ghdl](https://github.com/ghdl/ghdl), we are providing [nightly](https://github.com/ghdl/ghdl/releases/tag/nightly) releases built with MSYS2 in GitHub Actions. However, those packages need some additional dependencies in MSYS2: https://github.com/ghdl/setup-ghdl-ci/blob/master/main.js#L29-L33. I think that a general approach to achieve this should be discussed in MSYS, rather than doing it "manually" for each tool. Maybe MSYS2 does already have some script to get the deps of a package. /cc @lazka.

> I don't have msys handy

I can propose a PR to use @e-tabrez's build script in a GitHub Actions job (CI).

e-tabrez, 2020-06-07:
My first gtkwave-gtk3 distribution package is ready. Both light and Dark themes are working.
GUI is much better in look and feel compared to gtk2.
Package: gtkwave-3.3.105-gtk3-bin-win64.zip

I have created a SourceForge project and shared this package for other Windows users.
Please check at https://sourceforge.net/projects/gtkwave3-gtk3-win64-build/

Thanks Tony Bybell for this great wave analyzer tool.

e-tabrez, 2020-06-07:
> Hence, I would strongly suggest to please (PLEASE) contribute the gtk3 variant that you have written, as a new official package of MSYS2. That is, to open a PR in msys2/MINGW-packages. That will allow all MSYS2 users to easily install it.

I have zero experience in handling github and linux. This is my first time I am building a github software. So its difficult for me make any new build package.

However, I have uploaded gtk3 binary on SourceForge (https://sourceforge.net/projects/gtkwave3-gtk3-win64-build/).
Current version is gtkwave-3.3.105.

I will try to build and upload new versions as soon as updates are available in github project.

lazka, 2020-06-07:
The only "automatic" way atm is to install all deps with pacman to a different directory, install your app there and then delete everything you don't need and package it up. I'll plan to write some docs for this in the future..

umarcor, 2020-06-07:
> I have zero experience in handling github and linux. This is my first time I am building a github software. So its difficult for me make any new build package.

@e-tabrez, it's ok. Others (me) can help with that. Can you please just share the sequence of commands/steps that you use to generate the zipfiles?

> I will try to build and upload new versions as soon as updates are available in github project.
>
> If someone knows better place to upload this distribution, please guide me.

I'd like to make them available in this repository and have them updated automatically (as done with GHDL). See https://github.com/ghdl/ghdl/blob/master/.github/workflows/push.yml#L95-L97 and https://github.com/ghdl/ghdl/blob/master/.github/workflows/push.yml#L110-L114. That way, you won't need to build and upload new versions manually, you'll just need to download and test/use them.

@lazka, thanks a lot for clarifying. I guess we can use some ad-hoc script that copies "almost everything" for now (as @e-tabrez did), and we'll update the procedure when those docs are available.

e-tabrez, 2020-06-07:
> @e-tabrez, it's ok. Others (me) can help with that. Can you please just share the sequence of commands/steps that you use to generate the zipfiles?

Steps to build gtkwave3-gtk3 with MSYS2 on Windows10
(1) Uninstall any previous MSYS2 in computer and clean-up MSYS2 installation folder
(2) Download and install MSYS2 (https://www.msys2.org/). Don't installed at location where Admin right is needed. Best installation destination is C:\msys64
(3) Download gtkwave source from Github (https://github.com/gtkwave/gtkwave) using "Clone or Download" button
(4) Extract downloaded gtkwave-master.zip in MSYS2 desktop location (C:\msys64\home\<PC Name>\).
You can make new folder here for extraction (lets say gtkwave).
After extraction we should have below folders
C:\msys64\home\<PC Name>\gtkwave
C:\msys64\home\<PC Name>\gtkwave\gtkwave3
C:\msys64\home\<PC Name>\gtkwave\gtkwave3\gtkwave3-gtk3
(5) Now open C:\msys64\home\farah\tbz\gtkwave\gtkwave3-gtk3\README
Goto end of the file and check "MSYS2 notes for creating a working environment for compiling gtkwave:"
(6) Now open MSYS2 (Run C:\msys64\mingw64.exe)
(7) Now install the packahes as given in above README note (MSYS2 notes for creating a working environment for compiling gtkwave:)
3rd step "pacman -S base-devel mingw-w64-toolchain mingw-w64-i686-gtk2" may not work. Just skip it.
(8) After package installation steps are finished, change the current directory in MSYS2 to C:\msys64\home\<PC Name>\gtkwave\gtkwave3\gtkwave3-gtk3
(9) Now run command: ./configure --enable-gtk3
(10) Run command: make
(11) Run command: make install
(12) GTKwave exe files sould be created in C:\msys64\mingw64\bin
(13) Run gtkwave.exe to check if it is working correctly

Steps to create gtkwave3-gtk3 distribution zip
(1) Make a root package folder at any location (lets say gtkwave_gtk3_64)

(2) Create a new folder "bin" here
Copy the generated GTKWave exe file (atleast 17 file) from  C:\msys64\mingw64\bin to "gtkwave_gtk3_64\bin"
Copy all dll files from C:\msys64\mingw64\bin to "gtkwave_gtk3_64\bin"

(3) Create a new folder "lib"
Copy following folders from C:\msys64\mingw64\lib to "gtkwave_gtk3_64\lib"
	gdk-pixbuf-2.0
	tcl8
	tcl8.6
	tk8.6
	
(3) Create a new folder "share"
Copy following folders from C:\msys64\mingw64\share to "gtkwave_gtk3_64\share"
	applications
	gtkwave-gtk3
	icons
	man
	mime
	
(4) Run gtkwave_gtk3_64\bin\gtkwave.exe to check if its working correctly

(5) Packege gtkwave_gtk3_64 in a zip to distribute

tbzcode, 2020-06-17:
@gtkwave , @umarcor 
I am e-tabrez. Now username changed to tbzcode.
Just in continuation of our previous discussion, I have figured out which files are require in **bin** folder for gtk3 build.

I have release win64 and win32 binaries with these file in a forked project.
https://github.com/tbzcode/gtkwave/releases

## List of required file in **bin** for gtk3-build (win64):
### 17 EXE files:
evcd2vcd.exe
fst2vcd.exe
fstminer.exe
ghwdump.exe
gtkwave.exe
lxt2miner.exe
lxt2vcd.exe
rtlbrowse.exe
shmidcat.exe
twinwave.exe
vcd2fst.exe
vcd2lxt.exe
vcd2lxt2.exe
vcd2vzt.exe
vzt2vcd.exe
vztminer.exe
xml2stems.exe

### 43 DLL files:
libatk-1.0-0.dll
libbrotlicommon.dll
libbrotlidec.dll
libbz2-1.dll
libcairo-2.dll
libcairo-gobject-2.dll
libdatrie-1.dll
libepoxy-0.dll
libexpat-1.dll
libffi-7.dll
libfontconfig-1.dll
libfreetype-6.dll
libfribidi-0.dll
libgcc_s_seh-1.dll
libgdk-3-0.dll
libgdk-win32-2.0-0.dll
libgdk_pixbuf-2.0-0.dll
libgio-2.0-0.dll
libglib-2.0-0.dll
libgmodule-2.0-0.dll
libgobject-2.0-0.dll
libgraphite2.dll
libgthread-2.0-0.dll
libgtk-3-0.dll
libgtk-win32-2.0-0.dll
libharfbuzz-0.dll
libiconv-2.dll
libintl-8.dll
liblzma-5.dll
libpango-1.0-0.dll
libpangocairo-1.0-0.dll
libpangoft2-1.0-0.dll
libpangowin32-1.0-0.dll
libpcre-1.dll
libpixman-1-0.dll
libpng16-16.dll
libstdc++-6.dll
libthai-0.dll
libwinpthread-1.dll
libxml2-2.dll
tcl86.dll
tk86.dll
zlib1.dll

## List of required file in **bin** for gtk3-build (win32):
### 17 EXE files:
evcd2vcd.exe
fst2vcd.exe
fstminer.exe
ghwdump.exe
gtkwave.exe
lxt2miner.exe
lxt2vcd.exe
rtlbrowse.exe
shmidcat.exe
twinwave.exe
vcd2fst.exe
vcd2lxt.exe
vcd2lxt2.exe
vcd2vzt.exe
vzt2vcd.exe
vztminer.exe
xml2stems.exe

### 43 DLL files:
libatk-1.0-0.dll
libbrotlicommon.dll
libbrotlidec.dll
libbz2-1.dll
libcairo-2.dll
libcairo-gobject-2.dll
libdatrie-1.dll
libepoxy-0.dll
libexpat-1.dll
libffi-7.dll
libfontconfig-1.dll
libfreetype-6.dll
libfribidi-0.dll
libgcc_s_dw2-1.dll
libgdk-3-0.dll
libgdk-win32-2.0-0.dll
libgdk_pixbuf-2.0-0.dll
libgio-2.0-0.dll
libglib-2.0-0.dll
libgmodule-2.0-0.dll
libgobject-2.0-0.dll
libgraphite2.dll
libgthread-2.0-0.dll
libgtk-3-0.dll
libgtk-win32-2.0-0.dll
libharfbuzz-0.dll
libiconv-2.dll
libintl-8.dll
liblzma-5.dll
libpango-1.0-0.dll
libpangocairo-1.0-0.dll
libpangoft2-1.0-0.dll
libpangowin32-1.0-0.dll
libpcre-1.dll
libpixman-1-0.dll
libpng16-16.dll
libstdc++-6.dll
libthai-0.dll
libwinpthread-1.dll
libxml2-2.dll
tcl86.dll
tk86.dll
zlib1.dll