gtkwave/gtkwave #19: Horizontal menu-bar in gtkwave-gtk3

e-tabrez, 2020-06-07:
Is it possible to arrange the Menus in gtkwave-gtk3 in horizontal menu bar (Just like gtk2 version) ?
‐---------------------------------------------
File    Edit    Search    Time
----------------------------------------------

Since all menus are hidden in just one button, its very inconvinient to use.

e-tabrez, 2020-06-07:
I built gtk3 version with `#define WAVE_ALLOW_GTK3_HEADER_BAR` commented in file gtk23compat.h

```
#ifndef MAC_INTEGRATION
//#define WAVE_ALLOW_GTK3_HEADER_BAR
#endif
```
This worked.
I am not sure if this is correct solution. A command line option would be better.

gtkwave, 2020-06-07:
I'll look into it: I was under the impression the style guidelines were going away from menubars for a consistent look for gtk3. I found the "muscle memory" to get used to the vertical orientation took me a month or so to get used to it.

Note that to bring up the menus, F10 works just like it does in firefox, etc., F11 toggles full screen, F9 toggles the toolbar, etc.

-Tony