gtkwave/gtkwave #21: Show Wave Highlight is gated by Show Grid

alanmimms, 2020-06-16:
Excellent tool!

I just noticed Show Wave Highlight only works if you have the Show Grid checked. In fact I don't see much different with Show Grid checked and unchecked _except_ for the highlight of the current signal in the wave view, but I may just not be understanding what the "grid" is meant to be and when it's visible.

gtkwave, 2020-06-16:
Good catch. Thanks for reporting it. I had some bad nested ordering in the if() statements because Show Wave Highlight was a feature request that was implemented later.

The "grid" is the set of horizontal/vertical blue lines that separate traces and chunks of time that are marked off numerically in the topmost trace.

-Tony

98:/home/bybell/git/gtkwave> git commit -a
[master ab1f3be] Fix Show Wave Highlight so it is not dependent on Show Grid being active.
 4 files changed, 98 insertions(+), 92 deletions(-)
99:/home/bybell/git/gtkwave> git push origin master
Username for 'https://github.com': gtkwave 
Password for 'https://gtkwave@github.com': 
Counting objects: 18, done.
Delta compression using up to 4 threads.
Compressing objects: 100% (10/10), done.
Writing objects: 100% (10/10), 7.72 KiB | 0 bytes/s, done.
Total 10 (delta 9), reused 0 (delta 0)
remote: Resolving deltas: 100% (9/9), completed with 8 local objects.
To https://github.com/gtkwave/gtkwave.git
 705f789..ab1f3be master -> master

alanmimms, 2020-06-16:
Thanks. I thought that was the problem from a quick look at the code.

Do you use Cairo's layers to render things? The grid could be a separate (lower) layer that simply takes care of itself rather than having to have so much logic all around your interface code to know when it is damaged and needs redrawing, etc. At least I _think_ Cairo has this feature... If not you could probably achieve this with stacked transparent canvases. The same idea goes for various waveform overlays like marks and annotations of various sorts as well.

gtkwave, 2020-06-17:
I don't use Cairo layers. For the most part, it's keep it simple with the painter's algorithm being used rendering back to front with overdrawing. Grids, markers, etc., are all fairly straightforward.

If I use a touchscreen, at the usual zoom level I do debug in, I can get full framerate scrolling if I fling the waves with my finger so I don't really see too much optimization possible from stacked transparent canvases. The biggest time sink I've noticed are TLB misses if for example I am looking at waves that are ~1M cycles and we're zoomed out all the way. The rendering routines are very fast, but dealing with gigs of trace data spread all over memory is what can cause the biggest slowdown. I haven't messed with huge page optimizations yet. The viewer has its own memory allocator so it can be done, but I haven't done it yet as it functions well enough for my day job.

-Tony