gtkwave/gtkwave #22: Negative big-endian vector index causes symbols to be corrupted

alanmimms, 2020-06-18:
I am using VCD files produced by Verilator. My project has a lot of big-endian bit order bit vectors (I hate them too). Some of these are like

```
bit [-2:35] ADB;
```

I am pretty sure there is a correlation between symbols being corrupted or even lost and this pattern. I lost a symbol called ADB (possibly because another symbol called AD was present in the same scope). It simply didn't show up in the list for the scope. So I delved further. I renamed the symbol to `ADBxyzzy` and re-ran the test and then gtkwave. The result was that I now had a symbol in that scope showing up in the selector as `ADBxyzz` (the `y` is missing on the end). I found a few other examples.

I poked around in the code a while, but I can't find an easy explanation for why this would happen. I'm enclosing both the VCD file from this run and the resulting saved gtkw file showing the naming. The `ADB` and `AD` symbols I'm talking about are in `top.ebox0.edp0` scope, but there are other examples elsewhere as well.

[gtkwave3-gtk3.zip](https://github.com/gtkwave/gtkwave/files/4801397/gtkwave3-gtk3.zip)

gtkwave, 2020-06-18:
I used to use big endian a lot because I worked for 15 years at IBM, but not negative indices.

As a workaround for now try vcd2fst on the vcd file. The minus sign is causing unexpected results in the $var parsing routine in the vcd loaders and it's corrupting symbol names.

-Tony

gtkwave, 2020-06-19:
Try what I just checked into master. You've hit upon some code I wrote ~20 years ago that nobody's had an issue with until now. The funny thing is I was going to use negative LSBs on my last project (it failed the PD/synthesis flow so I just offset everything), but that still wouldn't have found this case.

The long and short is in the VCD loaders, MSB=-1 LSB=-1 for historical reasons indicates a single-bit signal with no subscripting. Yes, this should have been done with a "do not subscript" flag instead during the symbol build.

I think I have fixed things for vectors such as [-2:35] in the most recent check-in. I didn't fix it for single bit definitions such as x[-2:-2]. Mainly, any <0 value for a singla bit vector will be processed in vcd_build_symbols() as a non-subscripted signal.

The reason why some signal names chopped off a character on their names is because of an rc variable vcd_explicit_zero_subscripts and a useless old rc var atomic_vectors where signals added a level of hierarchy and the signal names themselves were simply numbers (with the signal itself one level up in the hierarchy). Sometime I need to clean that up by removing it. As I said, it's two decade old code.

-Tony

alanmimms, 2020-06-21:
Sorry for delay. Yes, that seems to have made the symbols I hacked to be of the form `ADBxyzzy` show their full names. I don't have any `x[-2:-2]` sorts of symbols. I simply don't use the brackets when declaring single bit vectors and I don't have any macros or similar that would do this as degenerate cases (I think).

Thanks for the fix. :smile: