gtkwave/gtkwave #23: "Hierarchy search" window can not show "Dot" in signal name.

tbzcode, 2020-06-20:
If a signal name has "Dot"character, it is shown correctly in SST and signal window.
But Hierarchy search window do not show it correctly.

[image]

gtkwave, 2020-06-20:
I need more information because I can't duplicate it and it looks ok for me. What is the dumpfile, what OS, etc? Are you using any rc files or rcvars such as alt_hier_delimeter?

Also, in src/vcd.h:

Chhange:
enum VCDName_ByteSubstitutions { VCDNAM_NULL=0,
#ifdef WAVE_HIERFIX
VCDNAM_HIERSORT,
#endif
VCDNAM_ESCAPE };

To:
enum VCDName_ByteSubstitutions { VCDNAM_NULL=0,
#ifdef WAVE_HIERFIX
VCDNAM_HIERSORT = 35,
#endif
VCDNAM_ESCAPE = 42 };

...and let me know if a # or * appears.

Thanks,
-Tony

tbzcode, 2020-06-21:
Hello Tony,
Thanks for reply.

I am using VCD file and **not using** any rc file. 
#### In  VCD file:
```
$scope module logic $end
$var wire 1 & Clock.rx_en $end
$var wire 1 ' Clock.tx_en $end
$var wire 1 ( empty $end
$var wire 1 ) underrun $end
$upscope $end
```
My OS is  Windows10 x64.
This problem is on both x86 and x64 builds (build using MSYS2) .
Also I tried your binaries from sourceforge (gtkwave-3.3.100-bin-win64.zip). Same problem here.

I build with the above change you suggested. A * appears
[image]

gtkwave, 2020-06-21:
OK, thanks, I'll look into it on my Windows build laptop that has sitting dormant in my bedroom closet for months. :-)

There seems to be no logical reason why it's doing that.

In vcd.c, there are two sequences like this at the bottom of vcd_sortfacs():

if(GLOBALS->escaped_names_found_vcd_c_1)
{
... stuff ...
}

I want you to replace the 2nd if(GLOBALS...) compare with if(1) and rerun it.
If there is no change, also replace the 1st with if(1) and rerun it. You can also experiment with printfs later to see if it is getting into both of those code blocks.
printf("REACHED LINE: %d\n", __LINE__);

If I comment out the line for the call to treenamefix(), I get a square block that has 0001 in it, which is what you are seeing.

I suspect on windows there might be a buffer overflow or stray pointer write somewhere OR it's in all OSs but invisible on linux due to how memory is allocated.

Something you can also try is run vcd2fst on it.
vcd2fst whatever.vcd whatever.fst

...then tell me if you see the same thing with the FST file.

OK...I booted up the laptop and tried compiling the latest. Things seem to work for me. See attached. You might have to send me your original VCD file in case the memory layout is important. Note also that the possibility exists that because the versions of gcc/msys2/gtk I'm using is different from yours, if the problem is memory corruption I will never see it in my builds. 

-Tony

tbzcode, 2020-06-24:
Hi Tony,
Thanks for looking.
I see this issue in your binary distribution available on Sourceforge (gtkwave-3.3.100-bin-win64.zip). So this doesn't seems like build issue due to msys or gcc.

I tried on 2 Windows 10 computers (home and office) and both show it.

I will try your suggestions and try to debug this weekend.

gtkwave, 2020-06-24:
I don't see it. See attached. Just for a test, try wine to see if it still fails under there.

If it does not, it looks to be a memory corruption issue so I will need your VCD.
If it does, I can duplicate it on my end so I need the VCD as well.

-Tony

tbzcode, 2020-06-27:
Hi Tony,

I have attached my VCD file here.

Thanks

Regards,
Tabrez Alam
Mobile: +82-10-2937-3924

gtkwave, 2020-06-27:
I don't see the file.  Did it not attach properly?

Thanks,
-Tony

tbzcode, 2020-06-28:
Sorry..
Please find here

[test_VCD.zip](https://github.com/gtkwave/gtkwave/files/4842414/test_VCD.zip)

gtkwave, 2020-06-28:
Thanks. That triggers a problem on linux. I'll dig into it as I can reproduce the issue.

-Tony

gtkwave, 2020-06-28:
Thanks, fixed. It's much easier when I have a good testcase to duplicate the fail. :-)

-Tony

572:/home/bybell/git/gtkwave> git commit -a
[master 44d3088] add missing recursion case to treenamefix()
 4 files changed, 6 insertions(+), 2 deletions(-)
573:/home/bybell/git/gtkwave> git push origin master
Username for 'https://github.com': gtkwave
Password for 'https://gtkwave@github.com': 
Counting objects: 19, done.
Delta compression using up to 4 threads.
Compressing objects: 100% (10/10), done.
Writing objects: 100% (10/10), 955 bytes | 0 bytes/s, done.
Total 10 (delta 9), reused 0 (delta 0)
remote: Resolving deltas: 100% (9/9), completed with 9 local objects.
To https://github.com/gtkwave/gtkwave.git
 4a40319..44d3088 master -> master

tbzcode, 2020-06-29:
Great... 
Thank you.

closed 2020-07-01 by tbzcode