gtkwave/gtkwave #24: Thread safety problem on FreeBSD

gezalore, 2020-06-28:
While trying to get the Travis CI build of Verilator passing on FreeBSD, I discovered the that Clang reports a thread safety issue. To reproduce:

```
$ uname -a
FreeBSD travis-job-ecfacb11-f904-4349-abbb-d6c0f674e06b 12.1-RELEASE-p3 FreeBSD 12.1-RELEASE-p3 GENERIC  amd64
```

```
$ git clone https://github.com/gtkwave/gtkwave.git
$ cd gtkwave/gtkwave3-gtk3/src/helpers/fst
$ touch config.h
$ clang -I. -Wthread-safety -Werror -DHAVE_LIBPTHREAD -DFST_WRITER_PARALLEL -c -o fstapi.o fstapi.c
fstapi.c:1811:1: error: releasing mutex 'xc->xc_parent->mutex' that was not held [-Werror,-Wthread-safety-analysis]
pthread_mutex_unlock(&(xc->xc_parent->mutex));
^
fstapi.c:1883:1: error: mutex 'xc->mutex' is not held on every path through here [-Werror,-Wthread-safety-analysis]
}
^
fstapi.c:1868:9: note: mutex acquired here
        pthread_mutex_lock(&xc->mutex);
        ^
2 errors generated.
```

Verilator's threaded FST tracing tests are all faling in the FreeBSD build (after disabling this warning), so it seems like there is a genuine issue there. I appreciate if you do not fancy lookin too much into this given the platform, so In the meantiem I will just disable those theses in Verilator on FreeBSD. If you would like, I can provide instructions on how to get SSH access into a Travis FreeBSD instance for debugging.

gtkwave, 2020-06-28:
I'll look at it, but their warning might be wrong. I'm acquiring the lock in a thread and then releasing it in the spawned/child thread which should be legal since it's the same address space. I don't know how well clang traces locks through threads.

I'd ask you to get the lock acquisition in line 1868 and move it to line 1808 in front of the fstWriterFlushContextPrivate2() call to see if the problem persists or not.

I'll try compiling with clang as well when I have the time.

-Tony

gezalore, 2020-06-28:
Actually I think that is undefined behaviour: 
https://stackoverflow.com/questions/4608843/locking-mutex-in-one-thread-and-unlocking-it-in-the-other
The threaded FST writer core dumps on FreeBSD while everything else seems to work, so I think the warning is genuine.

gtkwave, 2020-06-28:
OK, thanks for reporting this. I'll mess with it. Interesting that it works fine under Linux.

-Tony

 
 

https://stackoverflow.com/questions/4608843/locking-mutex-in-one-thread-and-unlocking-it-in-the-other

gtkwave, 2020-06-29:
OK, see if what I pushed to master fixes things.

Thanks,
-Tony

593:/home/bybell/git/gtkwave> git commit -a
[master e1f2884] fix misuse of cross-thread pthread_mutex lock/unlock.
 4 files changed, 40 insertions(+), 4 deletions(-)
594:/home/bybell/git/gtkwave> git push origin master
Username for 'https://github.com': gtkwave
Password for 'https://gtkwave@github.com': 
Counting objects: 24, done.
Delta compression using up to 4 threads.
Compressing objects: 100% (13/13), done.
Writing objects: 100% (13/13), 1.27 KiB | 0 bytes/s, done.
Total 13 (delta 12), reused 0 (delta 0)
remote: Resolving deltas: 100% (12/12), completed with 11 local objects.
To https://github.com/gtkwave/gtkwave.git
 44d3088..e1f2884 master -> master

 
 

https://stackoverflow.com/questions/4608843/locking-mutex-in-one-thread-and-unlocking-it-in-the-other

gezalore, 2020-06-29:
Thank you for the premium service. Sadly Travis seems to be broken at the moment. I will try to confirm this as soos as I can.

gezalore, 2020-06-30:
Confirmed this fixes the warning and the crashes as well. Thank you again for the swift action.

closed 2020-06-30 by gezalore