gtkwave/gtkwave #25: [Suggestion]Removing blur from fat lines (sharp fat lines)

tbzcode, 2020-07-01:
When using fat lines (use_fat_lines), GTKWave renders blurry lines.
Its not comfortable for eyes while doing analysis.
Is it possible to remove the blur while using fat liens?

Here are some discussion regarding blurry line rendering by cairo. 
[cairo : blurred shapes and text](https://stackoverflow.com/questions/23725883/cairo-blurred-shapes-and-text)
[How do I draw a sharp, single-pixel-wide line?](https://www.cairographics.org/FAQ/#sharp_lines)

Is it possible to remove blur by rounding the co-ordinates?
```
cairo_user_to_device
round
cairo_device_to_user
```

gtkwave, 2020-07-01:
The problem is that for single pixel lines, all coordinates need to be offset by 0.5, otherwise the single pixel line is blurry.
For double-wide lines, if you add 0.5, the line is blurry. If you don't, it's sharp.

You can try the following patch to see if it works on your system. This isn't how I'd do it in the source but is only a quick test special case so you can see the above behavior in both modes and if it fixes your problem.

-Tony

tbzcode, 2020-07-02:
@gtkwave , Sure...
Thanks for pointing out.
I will try it.

tbzcode, 2020-07-02:
@gtkwave , Thanks for pointing out.
This was the quick fix.
I made a global variable for pixel coordinate offset (cr_px_ofs) and replaced 0.5 offset with this variable in [`wavewindow.c`](https://github.com/tbzcode/gtkwave/blob/master/gtkwave3-gtk3/src/wavewindow.c) file
```
If cr_line_width > 1.0 then cr_px_ofs = 0.0
else cr_px_ofs = 0.5
```
Also I have made changes to configure the line width using rc variable (use_fat_lines) in my [fork](https://github.com/tbzcode/gtkwave) 
For example `use_fat_lines 1.5`

#### Fat line (width=2.0) before fix
[image]

#### Fat line (width=2.0) after fix
[image]

#### Normal line after fix
[image]

Thanks again.

gtkwave, 2020-07-02:
I checked something similar into SVN and git yesterday afternoon, so you should be good with what is in master.

-Tony

tbzcode, 2020-07-04:
Ok.. I merged your changes in my Fork. Better to keep in synch with main master project.

closed 2020-07-04 by tbzcode