gtkwave/gtkwave #29: Fixed point shift: displayed values are truncated to integer values if fpshift_count exceeds 30

dsvf, 2020-08-24:
Version: gtkwave v3.3.100 64 bit running on Windows 10 (the latest prebuilt binary for Windows from Sourceforge)

Test module:
```verilog
module tb_verilator(input clk);
initial
begin
    $dumpfile("dump.fst");
    $dumpvars();
end
logic [31:0] sig_32 = 32'd3;
always @(posedge clk) sig_32 <= sig_32 << 1;
logic [47:0] sig_48 = 48'd3;
always @(posedge clk) sig_48 <= sig_48 << 1;
logic [63:0] sig_64 = 64'd3;
always @(posedge clk) sig_64 <= sig_64 << 1;
endmodule
```
Simulate for e.g. 64 clock cycles and load the FST into gtkwave.
The image below shows the displayed values for fpshift between 29 and 32.
Setting fpshift up to 30 results in a correct display of the fractional bits.
Setting fpshift higher than 30 results in a display that discards the fractional bits.

[image]

gtkwave, 2020-08-24:
Try it on Linux if you can to see if the problem still exists. I'm thinking this is a case where 32-bit values are sneaking in on TimeType which in this case the code is using to store 64-bit values. In the past--like 15-20 years ago--there used to be a 32-bit limitation on Windows (back when it was compiled with Visual C) and I don't know if some of that is sneaking through ifdefs.
 
-Tony

dsvf, 2020-08-24:
Unfortunately something similar happens on Linux.

Version: gtkwave v3.3.107, 64 bit on Linux Mint, built from github.

Things do work when I set the radix to **signed** decimal:
[image]

Weirdly enough, when I set the radix to (_unsigned_) decimal, it fails again for fpshift > 30:
[image]

I will check tomorrow if the same occurs on Windows.

gtkwave, 2020-08-25:
I think it's this. "1" overflows off the end as it's a 32-bit constant. It needs to be cast to a 64-bit value.

-Tony

588:/home/bybell/gtkwave/code/gtkwave3-gtk3/src> diff -cNb -C 5 baseconvert.c.old baseconvert.c
*** baseconvert.c.old 2020-08-24 21:51:46.191328578 -0400
--- baseconvert.c 2020-08-24 21:51:58.135369786 -0400
***************
*** 385,395 ****
 
 static void cvt_fpsudec(Trptr t, TimeType val, char *os, int len)
 {
 int shamt = t->t_fpdecshift;
 UTimeType lpart = ((UTimeType)val) >> shamt;
! TimeType rmsk = (1 << shamt);
 TimeType rbit = (val & (rmsk-1));
 double rfrac;
 char dbuf[32];
 char bigbuf[64];
 
--- 385,395 ----
 
 static void cvt_fpsudec(Trptr t, TimeType val, char *os, int len)
 {
 int shamt = t->t_fpdecshift;
 UTimeType lpart = ((UTimeType)val) >> shamt;
! TimeType rmsk = (ULLDescriptor(1) << shamt);
 TimeType rbit = (val & (rmsk-1));
 double rfrac;
 char dbuf[32];
 char bigbuf[64];

dsvf, 2020-08-25:
5574d0b seems to have fixed this. I checked my original testbench where I initially noticed the faulty behavior, and the displayed values are now correct for both signed and unsigned integers (at least under Linux; I don't have a usable build environment under Windows, so I can't check there). I'll defer the closing of this issue to you.

Thank you for the swift response and bugfix!

bpringlemeir, 2024-10-21:
I upgraded, but still have issues with a [32:-34] values.  **I use an assign real in test bench.**  It also circumvents the need to recompile gtkwave for those on older distros.  I think this is a new issue with even larger lengths?  I compiled tag v3.3.107 which contains noted commit on Ubuntu 20.04 (head was using meson, which was too difficult to update to).  I may retry with 24.04 at home.

tbybell, 2024-10-21:
Please provide a direct example: for example, by attaching a vcd file and a save file and describe what should occur versus what is occurring.  I know it might be a bit of a pain, but exact examples help a lot.

Fixed point was added back in 2015 for version 3.3.68.  3.3.107 back in 2020 as shown above was the last update to that chunk of code.  Any of the recent tarballs found on sourceforge should compile fine for you as the 3.3 series is considered LTS and they don't use meson as I don't want to break builds on older machines.

svn checkout svn://svn.code.sf.net/p/gtkwave/code/ gtkwave-code

should work also, but you'll have to descend into the regular or -gtk3 directory depending on which version you're accustomed to.