gtkwave/gtkwave #31: unknown type 37 when using unconstrained vectors in ports

FranzForstmayr, 2020-09-25:
I get the following error when opening this file:

```
GTKWave Analyzer v3.3.103 (w)1999-2019 BSI

ghw_read_type: unknown type 37
ghw_read_base: error in section TYP
Error in ghw file 'out.ghw'.
```

I have this type in a port connection in a vhdl entity:

```
type vector_array_t is array(natural range <>) of std_logic_vector;
```

The entity looks like this:
```

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library work;
use work.pkg.all;

entity dut is
    generic (
        REGISTERS           : natural  := 1
        );
    port (
        read_reg      : in    vector_array_t(REGISTERS downto 0)(31 downto 0) := (others => (others => '0'))
    );
end entity;

architecture arch of dut is
begin
end architecture;
```

[out.zip](https://github.com/gtkwave/gtkwave/files/5283974/out.zip)

gtkwave, 2020-09-25:
Tristan would have to comment on this one as he's written both the writer/reader code for GHW.

-Tony

umarcor, 2020-09-25:
Indeed, it might be good to have an issue created in ghdl/ghdl and labeled [Output: GHW](https://github.com/ghdl/ghdl/labels/Output%3A%20GHW).

FranzForstmayr, 2020-09-29:
Works in latest 3.3.106 release! 
Sorry for inconvenience!

closed 2020-09-29 by FranzForstmayr

umarcor, 2020-09-29:
I submitted a PR for updating it in MSYS: msys2/MINGW-packages#7055.