gtkwave/gtkwave #32: How to remove leading zeros on the waveform?

phillbush, 2020-10-29:
I have the following waveform.  
pc is a 64-bit bus, but only the least significant bits of it are set, the leading bits are zero.

How can I view the value on the waveform without the leading zeros, so they can appear on the screen with the current zoom?  
Or, otherwise, can I make the value align to the right, so that the `+` appears on the left hiding some of the leading zeros?

[image]

gtkwave, 2020-10-29:
Currently there's no way to do it directly. For now, you can compile this and attach it as a process filter.

#include <stdio.h>
int main(int argc, char **argv)
{
while(!feof(stdin))
{
char buf[1025];
buf[0] = 0;
fscanf(stdin, "%s", buf);
if(buf[0])
{
char *buf2 = buf;
while((*buf2=='0') && *(buf2+1)) buf2++;
printf("%s\n", buf2);
fflush(stdout);
}
}
return(0);
}

See "Alias Files and Attaching External Disassemblers" in the documentation gtkwave.pdf (or the .odt file in the tarball).

-Tony

closed 2020-10-29 by phillbush

MrJake222, 2024-06-03:
As it is highest result of google search `gtkwave leading zeros`, I feel obliged to post this:

[image]

KireinaHoro, 2024-08-22:
@MrJake222 which version of GTKWave are you using?  I don't see such an option in v3.3.104...

tbybell, 2024-08-22:
3.3.108 or greater.