gtkwave/gtkwave #33: Closing SST panel does not increase waves room

jotego, 2020-11-14:
When I click on the SST arrow (at the top of the design hierarchy) the left panel goes blank but its window area is still shown. I would have expected to see the signals and waves windows move left and take more room. That is the behaviour in SimVision, for instance. So clicking on SST toggles between showing the design or giving more room to waveforms.

My version is 3.3.103, just in case this has already been addressed.

Thank you

gtkwave, 2020-11-14:
It's a toolkit limitation where the blank SST doesn't collapse. You still should be able to drag the pane over to the left and reclaim most of the room.

-Tony

jotego, 2020-11-14:
I understand. I can drag it manually, yes. Thank you.

closed 2020-11-14 by jotego