gtkwave/gtkwave #35: Can't refresh with gconftool-2 or gsettings

ItsaMeTuni, 2020-11-16:
I've been trying to refresh gtkwave from the terminal using
```
gconftool-2 --type string --set /com.geda.gtkwave/0/reload 0
```
and
```
gsettings set com.geda.gtkwave reload 0
```
as described at the end of [gconf.c](https://github.com/gtkwave/gtkwave/blob/master/gtkwave3-gtk3/src/gconf.c)

I have gtk v3.3.107

I have tried manually installing the [gschema.xml](https://github.com/gtkwave/gtkwave/blob/d4932822c7c0a3b7c6e40f4de26c178365d64471/gtkwave3-gtk3/src/com.geda.gtkwave.gschema.xml) by copying it into `/usr/share/glib-2.0/schemas/` and running `sudo glib-compile-schemas /usr/share/glib-2.0/schemas/`, but it still doesn't work.

I have never used `gconftool-2` or `gsettings` before, so I might be missing something. Any ideas?

gtkwave, 2020-11-18:
I can't tell anything because I have no idea if your executable was built against either of those.

 --with-gsettings Use GSettings to store preferences (precludes GConf)
 --with-gconf Use GConf to store preferences

If this wasn't specified in the ./configure command line, you wouldn't have the capability to do it.

-Tony

539:/home/bybell/gtkwave/code/gtkwave3-gtk3/examples> ldd /usr/local/bin/gtkwave | grep gco
 libgconf-2.so.4 => /lib64/libgconf-2.so.4 (0x00007fb5a8e09000)
540:/home/bybell/gtkwave/code/gtkwave3-gtk3/examples> gconftool-2 --type string --set /com.geda.gtkwave/0/reload 0
GTKWAVE | Reloading waveform...
541:/home/bybell/gtkwave/code/gtkwave3-gtk3/examples> FSTLOAD | Processing 1432 facs.
FSTLOAD | Built 1287 signals and 145 aliases.
FSTLOAD | Building facility hierarchy tree.
FSTLOAD | Sorting facility hierarchy tree.
GTKWAVE | ...waveform reloaded

ItsaMeTuni, 2020-11-19:
> I can't tell anything because I have no idea if your executable was built against either of those.
> […](#)
> --with-gsettings Use GSettings to store preferences (precludes GConf) --with-gconf Use GConf [...]

Thank you very much! Running `./configure --with-gsettings` did the trick. This doesn't seem to be documented anywhere, or am I missing something?

gtkwave, 2020-11-20:
The documentation for it is in "./configure --help" but a lot of people gloss over the output of that script as they're used to using maybe the --prefix option and that's it.

-Tony

closed 2020-11-20 by ItsaMeTuni