gtkwave/gtkwave #37: Unable to compile the gtk3 version: unknown type name Window

DDoSolitary, 2020-11-28:
I'm trying compile gtkwave-gtk3 on Arch Linux but got a few errors:

```
In file included from analyzer.c:15:
globals.h:603:1: error: unknown type name ‘Window’
  603 | Window socket_xid; /* from main.c 241 */
      | ^~~~~~
In file included from bitvec.c:15:
globals.h:603:1: error: unknown type name ‘Window’
  603 | Window socket_xid; /* from main.c 241 */
      | ^~~~~~
In file included from busy.c:10:
globals.h:603:1: error: unknown type name ‘Window’
  603 | Window socket_xid; /* from main.c 241 */
      | ^~~~~~
In file included from color.c:10:
globals.h:603:1: error: unknown type name ‘Window’
  603 | Window socket_xid; /* from main.c 241 */
      | ^~~~~~
In file included from ae2.c:10:
globals.h:603:1: error: unknown type name ‘Window’
  603 | Window socket_xid; /* from main.c 241 */
      | ^~~~~~
In file included from globals.c:10:
globals.h:603:1: error: unknown type name ‘Window’
  603 | Window socket_xid; /* from main.c 241 */
      | ^~~~~~
In file included from bsearch.c:10:
globals.h:603:1: error: unknown type name ‘Window’
  603 | Window socket_xid; /* from main.c 241 */
      | ^~~~~~
In file included from debug.c:20:
globals.h:603:1: error: unknown type name ‘Window’
  603 | Window socket_xid; /* from main.c 241 */
      | ^~~~~~
make[4]: *** [Makefile:629: busy.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[4]: *** [Makefile:629: color.o] Error 1
make[4]: *** [Makefile:629: ae2.o] Error 1
In file included from clipping.c:10:
globals.h:603:1: error: unknown type name ‘Window’
  603 | Window socket_xid; /* from main.c 241 */
      | ^~~~~~
make[4]: *** [Makefile:629: bsearch.o] Error 1
In file included from currenttime.c:10:
globals.h:603:1: error: unknown type name ‘Window’
  603 | Window socket_xid; /* from main.c 241 */
      | ^~~~~~
make[4]: *** [Makefile:629: analyzer.o] Error 1
make[4]: *** [Makefile:629: debug.o] Error 1
make[4]: *** [Makefile:629: bitvec.o] Error 1
make[4]: *** [Makefile:629: clipping.o] Error 1
make[4]: *** [Makefile:629: globals.o] Error 1
In file included from baseconvert.c:10:
globals.h:603:1: error: unknown type name ‘Window’
  603 | Window socket_xid; /* from main.c 241 */
      | ^~~~~~
make[4]: *** [Makefile:629: currenttime.o] Error 1
make[4]: *** [Makefile:629: baseconvert.o] Error 1
In file included from discardbuttons.c:10:
globals.h:603:1: error: unknown type name ‘Window’
  603 | Window socket_xid; /* from main.c 241 */
      | ^~~~~~
make[4]: *** [Makefile:629: discardbuttons.o] Error 1
make[4]: Leaving directory '/build/gtkwave-gtk3/src/gtkwave-gtk3-3.3.107/src'
make[3]: *** [Makefile:705: all-recursive] Error 1
make[3]: Leaving directory '/build/gtkwave-gtk3/src/gtkwave-gtk3-3.3.107/src'
make[2]: *** [Makefile:454: all] Error 2
make[2]: Leaving directory '/build/gtkwave-gtk3/src/gtkwave-gtk3-3.3.107/src'
make[1]: *** [Makefile:413: all-recursive] Error 1
make[1]: Leaving directory '/build/gtkwave-gtk3/src/gtkwave-gtk3-3.3.107'
make: *** [Makefile:353: all] Error 2
```

It seems that the source code uses the X11 Window type but doesn't include X11 headers. I can build it successfully after adding `#include <X11/X.h>` to the beginning of `globals.h`.

gtkwave, 2020-11-28:
#ifdef GDK_WINDOWING_X11
Window socket_xid; /* from main.c 241 */
#else
unsigned long socket_xid; /* for windows compiles */
#endif
#else
GdkNativeWindow socket_xid; /* from main.c 241 */
#endif

...at the beginning of globals.h, please see if it compiles if you add that header like this, which would be safer:

#ifdef GDK_WINDOWING_X11
#include <X11/X.h>
#endif

-Tony

DDoSolitary, 2020-11-29:
@gtkwave Yes, it compiles with the `#ifdef` guard added, but in this case I need to add it after the long list of other `#include`s, where the definition of `GDK_WINDOWING_X11` has been included.

gtkwave, 2020-11-29:
Thanks. I checked it in to master.

-Tony

closed 2020-11-30 by DDoSolitary