gtkwave/gtkwave #38: Icons for the menu bar are not showing up in Ubuntu.

wel97459, 2020-12-04:
The icons on the menu bar are not showing up after loading a waveform file.
  
When I first launch GTKWave the disabled icons are shown:
[image]
 
But after loading a file the icons are no longer shown:
[image]

I first installed GTKWave with apt-get, and just pulled down this repo and compiled it, with no luck.

gtkwave, 2020-12-04:
Those are stock icons so I don't know what is going on with that.
Do you see missing stock icons in gtk-demo or gtk3-demo (whichever version gtkwave is using)?

If you don't know what version gtkwave is using either run the ldd command against the gtkwave binary (wherever it is located) to see what libraries are used or simply check both of the gtk demos.

-Tony

wel97459, 2020-12-04:
Seems that gtkwave was picking up on the dark theme setting and making the icons lighter and not drak, and since the color of the window stays white/gray I couldn't see them.

gtkwave, 2020-12-04:
At least for the gtk3 version you should be able to try this command line flag:

 -6, --dark set gtk-application-prefer-dark-theme = TRUE

...for some reason it doesn't always communicate back to the application. I don't know why.

-Tony

wel97459, 2020-12-04:
Thanks for your help I'll give it a try.

closed 2020-12-04 by wel97459