gtkwave/gtkwave #42: Conflicts between gtkwave-gtk2 and gtkwave-gtk3

umarcor, 2021-01-15:
When building gtk2 and gtk3 versions of gtkwave, and trying to install them on the same system, most of the files conflict: https://github.com/umarcor/MINGW-packages/runs/1706772202?check_suite_focus=true#step:5:12438. Are all those binaries actually different or is `gtkwave.exe` the only one that should be duplicated (`gtkave-gtk2.exe` and `gtkwave-gtk3.exe`)?

gtkwave, 2021-01-15:
The gtkwave, rtlbrowse, and twinwave executables are the only ones that directly use gtk. For all practical purposes, gtkwave is the only one that should be duplicated.

-Tony

umarcor, 2021-01-15:
@gtkwave, thanks! We are deciding whether to drop the GTK2 build in msys2/MINGW-packages#7671. However, should both be preserved in the end, I will allow installing both of them side-by-side, while avoiding duplication of other executables.

I'm closing this, because the question is solved.

closed 2021-01-15 by umarcor

ripopov, 2021-03-19:
@umarcor  For some reason Gtk3 version is much slower for me, I hope GTK2 build won't be dropped.

umarcor, 2021-03-19:
@ripopov it's not something I can decide. While GTK2 is available in MSYS2, I will keep building and providing both of them. Currently, if you install GTKWave through MSYS2, you get both binaries (the default is GTK3, but you can execute gtkwave-gtk2 too). However, when they drop it, I cannot keep maintaining it for GTKWave only.

gtkwave, 2021-03-20:
I have a Core2Duo Thinkpad as a build machine that has both Snow Leopard and MSYS with GTK2 and 3 installed. If need be I can produce binaries indefinitely and post them on Sourceforge as I have been doing for the past number of years.

I think that GTK3 is bloated out in some respects and there are obvious speed differences you can notice such as if you click down the primary marker and drag to the right to see how fast scrolling is. The maintainers argue that it is more feature rich, etc., but I want an application toolkit, not something suitable for rendering videogames or anything that needs to be frame accurate. I'd use SDL if I wanted that.

Note that if you really want to see fast, try compiling against gtk1.2 or 1.3! :-) The widgets and interface are super ugly in this day and age, but the GDK rendering flies.

I think someone posted on here that the GTK4 maintainers said 4 is faster than 3, but I don't want to spend the next three months porting to GTK4. The amount of breakage across major revisions of GTK is incredible. GTK 1->2 was ok, but 2->3 was horrible as I had to work around toolkit issues or code up a bunch of compatibility functions in gtk23compat.c/.h. (That's why you see the weird XXX_function() naming in the source in places.)

I'd expect GTK3->4 to be more of the same pain I saw from GTK2->3 but I'll try to keep an open mind.

https://blog.gtk.org/2020/12/16/gtk-4-0/

...at least it's finally out so the API shouldn't be a moving target.

-Tony

umarcor, 2021-03-20:
@gtkwave, what about Gtk vs Qt? You mentioned some months ago that you had considered it because of the pain with migrating between Gtk versions. Is that something that you'd consider seriously? Or is Gtk-4 less painful, even if not easy?

gtkwave, 2021-03-20:
Qt in the long run is probably the better way to go as its interface appears far more stable so it's not like the case that when GTK5 comes out, we're back where we started again in having to do a port of GUI code. The only real negative on it that I can think of for Qt would be the learning curve for implementing more involved GUI elements such as the SST, but it might not be bad since I had to adopt a model/view/controller implementation already for GTK3.

The only real main advantage in sticking with GTK is that the overall event model, etc., would be the same so I don't think I'd have to deal with any difficult problem solving is getting the Tcl interface working with GTK as that problem has already been solved. The big unknown is that I don't know what has been deprecated or changed until I sift through all the compiler errors and warnings. I haven't done that since when 3.89 came out as I was curious how much of GTK changed from versions 3 to 4 and it looked like a fair amount.

I'll have to try compiling against GTK4 as an experiment to see how much work is involved. Looking back at my notes, it was pretty much a solid month of work during holiday vacation for 2017.

The funny thing is the gtkwave GTK3 code is over 3 years old and I still don't know why DND of signals from the lower part of the SST works fine for GTK2 but causes scrolling issues in GTK3 (where the bottom of the screen is where the signals wind up). It's the same exact code, but scrollbars seem to be triggered or set slightly differently.

-Tony

m-kru, 2022-05-23:
There is yet another option, namely [Dear ImGui](https://github.com/ocornut/imgui). It seems to be more stable and less bloated than both Gtk and Qt.