gtkwave/gtkwave #43: High-DPI display issues

markusdd, 2021-02-25:
Hi,

I am using gtkwave on Mac and on High DPI displays it appears super blurry:

[image]

is there anything to fix this?

gtkwave, 2021-02-26:
There was a fix a user provided for this a while ago but I think it was for GTK3 only. I only provide the GTK2 binaries on SourceForge, mainly because I never got around to installing GTK3 in my build environment for Mac. For a non-blurry screen, I think you'll have to compile from source. Someone on this mailing list would have the step-by-step info for it. I'm sorry I can't be of more direct help.

I don't have a Mac, but use an old Snow Leopard VM I installed from CDs very long ago.

-Tony

markusdd, 2021-02-27:
Thanks for the response.

I am vurrently attempting to build from source, but I noticed an issue:

The configure flow has a problem when --enable-gtk3 is given, it does not pick up gtk-mac-integration in this case.
When left out it stays in gtk2 mode, which is non-sensical for the gtk3 tarball.

Also gtk-mac-integration now delivers gtk-mac-integration-gtk2.pc and -gtk3.pc as package config files.
The generic one without suffix is a fallback onto gtk2.

Can you possibly give guidancy on how this could be fixed, so mac-integration is used with gtk3?

I have managed to build the application, but something is wrong still, because when launched the application complains gtk2 and gtk3 symbols are loaded at the same time, which is forbidden.

If we can iron out the configure flow it should work.

markusdd, 2021-02-27:
ok I got some way further now: Major problem was that in pkgconfig file for gtk3 it says 'targets' as a variable, not 'target', which is queried in configure.
I changed it to target and now GTK_MAC is picked up for gtk3.

Issues remain for building code that uses gtk socket code, as the header files somehow have an explusion to only pull in stuff when x11 is there.

markusdd, 2021-02-27:
[image]

Ok, so I managed to build against gtk3 with quartz backend with the following hacks (3.3.108 tarball):

in configure.ac I need to change line 613 to:
_gdk_tgt=`$PKG_CONFIG --variable=targets gdk-3.0`
(targets instead of target)

in main.c I need to undef WAVE_USE_XID

in twinwave.c I have to eclipse lines 138-180.
in twinwave.c I have to comment out lines 432 and 470 (the sprintf statements under the ifdef MAC_INTEGRATION)

As you can see the fonts look beautiful now, but Icons are still missing and I get the followoing message on cli:
CoreSVG has logged an error. Set environment variabe "CORESVG_VERBOSE" to learn more.

after setting this env variable I see this:
```
C/c command was given the wrong number of points.
Unhandled operator '.'
Unhandled operator '4'
Unhandled operator '7'
Unhandled operator '0'
Unhandled operator '7'
Unhandled operator '-'
Unhandled operator '3'
Unhandled operator '7'
Unhandled operator '.'
Unhandled operator '3'
Unhandled operator '0'
Unhandled operator '4'
Unhandled operator '7'
Unhandled operator '5'
Unhandled operator '6'
Unhandled operator '.'
Unhandled operator '7'
Unhandled operator '3'
Unhandled operator '8'
Unhandled operator '3'
Unhandled operator '-'
Unhandled operator '3'
Unhandled operator '8'
Unhandled operator '.'
```

Any advice?

markusdd, 2021-02-27:
Ok, figure out the icon issue:
adwaita-icon-theme must be installed via homebrew.

that just leaves the 

```
GTKWAVE | MESSAGE: gtk_window_add_accel_group: assertion 'GTK_IS_WINDOW (window)' failed
```

error message. Also this causes gtkwave to always open in the background. I also have the impression the graphics acceleration is a bit choppy. But it looks nice now:

[image]

gtkwave, 2021-02-28:
That looks *really* good. I suppose I'll have to try installing GTK3 on the Mac VM so I can get this working out of the box for people. I quite literally got it working back in 2011 or so and haven't messed with it for the package bundles. The big problem is that for GTK2 that I had to patch a couple of broken things in GTK that the OSX maintainers either refused to fix or ignored. It was too long ago for me to remember what was the case.

-rw-rw-r--. 1 bybell bybell 1000 Feb 4 21:37 gtk_diff_against_modulesets_2.patch
-rw-rw-r--. 1 bybell bybell 1519 Feb 4 21:37 gtk_diff_against_modulesets_3.patch
-rw-rw-r--. 1 bybell bybell 4840 Feb 4 21:37 gtk_diff_against_modulesets.patch

It didn't have to deal with graphics acceleration but flushing of buffers not occuring when it should or often enough so the display updates were extremely choppy.
Screendumps were broken as well because GDK didn't handle the RGB ordering right.
I don't know if either of those were fixed in GTK3 or if the problem exists in that codebase.

-Tony

markusdd, 2021-02-28:
hey tony, would it be of any help to you if I created a fork/branch and check in my changes I had to make?
Of course they are not the final fix but they can give you a guide on where to look and not go through the whole debugging again.

hugolundin, 2021-03-02:
@markusdd Nice work! I have tried building your repository, but cannot get it to work due to `./gtk23compat.h:12:10: fatal error: 'gdk/gdkwayland.h' file not found`. Any suggestions? Alternatively, would you be able to build the `.app`? Thank you!

markusdd, 2021-03-02:
@hugolundin please follow the branch of this pull request. https://github.com/gtkwave/gtkwave/pull/44

the configure setup and also the code needs some adaptation in some places.

from this branch of mine you should be able to build, but for the app we need to fix the main.c and twinwave.c properly.
I have eclipsed the failing sections just for now, because they still rely on X11 which is not really an option anymore on Mac as XQuartz is basically abandones.

The build failure you see is because of a configure error.

hugolundin, 2021-03-02:
> @hugolundin please follow the branch of this pull request. #44

Yep, I used that one and followed the build instructions you have added in one of the last commits. Does not work for me. 

> The build failure you see is because of a configure error.

Any idea what the configure error might be? I looked at `MAC_INTEGRATION` if that might be it, with no luck.

markusdd, 2021-03-02:
please provide your configure output

hugolundin, 2021-03-02:
> ah, i know why. Ill update the README.
> you are missing `brew install gtk-mac-integration`

Haha, right, I have actually installed that one too! No luck :(

markusdd, 2021-03-02:
please provide your configure output. gtk-mac-integration is not picked up if you have this problem.-

hugolundin, 2021-03-02:
Here is the output of my configure: [configure.txt](https://github.com/gtkwave/gtkwave/files/6069093/configure.txt)

markusdd, 2021-03-02:
ok as I thought. gtk-mac-integration is not being picked up.

Are you really on the gtk3-mac branch of my fork?

Please provide your configure and configure.ac file

hugolundin, 2021-03-02:
Oh wow, I'm so sorry. I checked out the correct branch, but in order to make sure that I started from a clean slate before asking you questions, I deleted the repository. Then I didn't checkout the branch again... Now it works. Thanks!

markusdd, 2021-03-02:
perfect.
keep in mind this is a hacked build, twinwave will not function. But is it 1000x better than the blurry gtk2 version.

We'll keep working here to get to a new shine gtk3 versin of the .app

hugolundin, 2021-03-03:
I noticed that the performance of this version is much worse, so I tried running it in Instruments. Could it be similar to the issue they had with Gimp (https://gitlab.gnome.org/GNOME/gimp/-/issues/5989)? 

[image]

markusdd, 2021-03-03:
Can you be more specific what exactly you mean with the performance? Loading waveform? Rendering? Zooming?

I have noticed that the GUI feels not 100% smooth especially when zooming, that is true, but it always felt well usable.

The issue could of course be related.

hugolundin, 2021-03-03:
The pictures shows how simply clicking in the waveform viewer causes a lot of CPU activity. 

Dragging the wave cursor and resizing the window stutters a lot, and it takes almost a second before buttons react when hovering over them. Not sure how to describe it, but overall, doing things in the program is really slow graphically. 

The behavior is the same in my Intel Mac Mini and on an M1 MacBook Air.

markusdd, 2021-03-03:
ok yes this I can confirm.

I mostly noticed it when zooming in and out of the waveform.

markusdd, 2021-03-03:
I have raised an issue https://gitlab.gnome.org/GNOME/gtk/-/issues/3714

gtkwave, 2021-03-03:
Thanks for opening that issue on them.

Note that gtk3 on Linux is quite smooth. If you have a touchscreen machine, you can smoothly fling waves around in time like you would expect to do on an iPad. It's pretty neat seeing waves scrolling like that.

I don't know how responsive the gtk-osx developers are. Please also check if File/Grab To File works correctly for the screen grab function in gtk3. It was broken on gtk2 and the developers never answered me when I asked them to fix it. It has to do with incorrect RGB/RGBA/ARGB/whatever byte ordering or something like that.

-Tony

markusdd, 2021-03-03:
interesting.
We have a shot at them having a look as GIMP seems to suffer from a similar problem.
But it could also be an issue on the end of Quartz and hence Apple.
Let's see what emerges.

Still: in my PR I have adjusted the ifdefs to MAC_INTEGRATION to what was causing issues in twinwave.
So you have an idea on how to fix that properly? Would it then be using the linux implementation without x-socket?

markusdd, 2021-03-03:
grab to file fails with this message on the CLI:

Error opening imagegrab file '/Users/mkrause/projects/sv_base_lib/test_lib/svb_clk_div_tb/Untitled.png' for writing.
Why: could not execute gdk_pixbuf_get_from_window().
Error opening imagegrab file '/Users/mkrause/projects/sv_base_lib/test_lib/svb_clk_div_tb/Untitled.png' for writing.
Why: could not execute gdk_pixbuf_get_from_window().

gtkwave, 2021-03-03:
Thanks. It looks like gtk-osx doesn't implement that on gtk3? I'll have to dig into the gdk sourcecode.

-Tony

markusdd, 2021-03-03:
It does exist https://developer.gnome.org/gdk3/stable/gdk3-Pixbufs.html

but maybe it has an issue, because of this one message we are still getting:

`GTKWAVE | MESSAGE: gtk_window_add_accel_group: assertion 'GTK_IS_WINDOW (window)' failed`

gtkwave, 2021-03-04:
It's probably a toolkit race condition in OSX.

At the very beginning of alt_window_top()

GtkWidget *alt_menu_top(GtkWidget *window)
{
if(!window) { fprintf(stderr, "window is NULL\n"); return(NULL); }

You should be able to trace where it's coming from. Alternatively, which might be better, in main.c uncomment this:
/*
#define WAVE_CRASH_ON_GTK_WARNING
*/

and give me the caller trace so we can see what called it so we can filter calls vs NULL pointers appropriately.

-Tony

markusdd, 2021-03-04:
I will check on that. Meanwhile there was a comment on the GTK performance issue:
[image]

markusdd, 2021-03-04:
I included the define you were recommending, not any more useful output:

```
~/gtkwave-gtk3-build/bin/gtkwave   dump.fst

GTKWave Analyzer v3.3.108 (w)1999-2020 BSI

FSTLOAD | Processing 76 facs.
FSTLOAD | Built 75 signals and 1 alias.
FSTLOAD | Building facility hierarchy tree.
FSTLOAD | Sorting facility hierarchy tree.
GTKWAVE | GLIB_OLD_LOG_API: 1
GTKWAVE | MESSAGE: gtk_window_add_accel_group: assertion 'GTK_IS_WINDOW (window)' failed
GTKWAVE | PRIORITY: 4
GTKWAVE | GLIB_DOMAIN: Gtk
Error opening imagegrab file '/Users/mkrause/projects/sv_base_lib/test_lib/svb_clk_div_tb/Untitled.png' for writing.
Why: could not execute gdk_pixbuf_get_from_window().
```

hugolundin, 2021-03-05:
@gtkwave Is there a way to disable client side decorations, to try that hypothesis? I found `WAVE_ALLOW_GTK3_HEADER_BAR` but it does not seem to apply for macOS (?).

gtkwave, 2021-03-06:
I don't know why I disabled in in the compat defines for gtk3: I wrote that set of code in late 2017 when I was doing the gtk3 port. It might simply be that I don't have gtk3 installed on a Mac to test it so I disabled it on the Mac platform just in case there were problems.

It also might be because of the Mac-style top of screen menus vs the menu that pops up from the menu button. Someone can try it to see if there are problems.

-Tony

markusdd, 2021-03-08:
changing this define does absolutely nothing on my end.

gtkwave, 2021-03-08:
Warnings typically take down GTK with an assert. Based on the program output I saw, for some reason that NULL pointer for window was a message rather than an assert.

See the code in main.c ifdef bracketed with WAVE_GTK3_GLIB_WARNING_SUPPRESS.
It should call abort() when it prints a warning.

The two important defines are

WAVE_GTK3_GLIB_WARNING_SUPPRESS
WAVE_CRASH_ON_GTK_WARNING

g_log_set_always_fatal(G_LOG_LEVEL_CRITICAL|G_LOG_LEVEL_WARNING);

might need to be changed to include G_LOG_LEVEL_MESSAGE as well, etc.

-Tony

markusdd, 2021-03-11:
OK I will play with the defines. But further profiling reveals this:

Looks suspect to me.
[image]

gtkwave, 2021-03-11:
Thanks for the info. Yes, it sounds like something is fundamentally wrong inside the toolkit's rendering/refresh/update/etc routines.

One thing new with GTK3 is that for screen updates they added the concept of a frame clock. I understand they expect someone will be making videogames and such with it, but software that isn't entertainment-based doesn't need real-time screen updates or the extra CPU/GPU load associated with it. I don't know how optimized the rendering back end is for OSX vs Linux and if some compositor is running over and over, but CPU usage when nothing is going on with Linux on my machine is at 0%. 

https://developer.gnome.org/gtk3/stable/chap-drawing-model.html
https://developer.gnome.org/gdk3/stable/GdkFrameClock.html

-Tony

markusdd, 2021-03-11:
Over in the GTK Thread the guy who optimized it for GTK4 mentioned it might actually be realted to cairo. I also posted a more detailed trace there. The time really seems to get lost down there:
https://gitlab.gnome.org/GNOME/gtk/-/issues/3714#note_1055703

gtkwave, 2021-03-11:
The guy talked about the expander. Out of curiosity for debugging that, try this:

544:/home/bybell/gtkwave/code/gtkwave3-gtk3> gtkwave --rcvar="hide_sst yes" examples/des.gtkw

The important part is the rcvar argument.

If it speeds up, then see if it drags when you enable
Search->Signal Search Tree.

-Tony

 
 

https://gitlab.gnome.org/GNOME/gtk/-/issues/3714#note_1055703

dehesselle, 2021-03-12:
Hi,
if you are using cairo, then you are suffering from the same performance issues we have with Inkscape on macOS. The culprit is deep within the lower layers of how cairo interfaces with Quartz. From time to time we've had people do some digging, but sadly no one stuck around long enough to solve it. I don't know if it is of help to you but here's a link to [our issue](https://gitlab.com/inkscape/inkscape/-/issues/1614).

markusdd, 2021-03-12:
Ok, I fear we have dug up something bigger here...

markusdd, 2021-03-12:
> --rcvar="hide_sst yes"

ok so this is interesting. It seems to speed up a little bit, but when activating the search list by menu it comes up as a seperate window and does not seem to impact performance.

Still it is not as fluent as the GTK2 version is or GTK3 on linux.

So partly there seems to be an impact.
What's curious is that GDK_RENDERING=image as suggested in the GTK thread does absolutely nothing, the performance trace looks the same.

markusdd, 2021-03-12:
> Warnings typically take down GTK with an assert. Based on the program output I saw, for some reason that NULL pointer for window was a message rather than an assert. See the code in main.c ifdef [...]

Ok so this is the output:

```
~/gtkwave-gtk3-build/bin/gtkwave dump.fst

GTKWave Analyzer v3.3.108 (w)1999-2020 BSI

FSTLOAD | Processing 76 facs.
FSTLOAD | Built 75 signals and 1 alias.
FSTLOAD | Building facility hierarchy tree.
FSTLOAD | Sorting facility hierarchy tree.

(gtkwave:17482): GLib-GObject-WARNING **: 19:47:49.586: invalid cast from 'GtkMenuBar' to 'GtkWindow'

(gtkwave:17482): Gtk-CRITICAL **: 19:47:49.586: gtk_window_add_accel_group: assertion 'GTK_IS_WINDOW (window)' failed

```

markusdd, 2021-03-12:
Now I also enabled the crash on message, but no stacktrace is printed:

```
GTKWave Analyzer v3.3.108 (w)1999-2020 BSI

FSTLOAD | Processing 76 facs.
FSTLOAD | Built 75 signals and 1 alias.
FSTLOAD | Building facility hierarchy tree.
FSTLOAD | Sorting facility hierarchy tree.

(gtkwave:18766): GLib-GObject-WARNING **: 19:54:57.743: invalid cast from 'GtkMenuBar' to 'GtkWindow'

(gtkwave:18766): Gtk-CRITICAL **: 19:54:57.743: gtk_window_add_accel_group: assertion 'GTK_IS_WINDOW (window)' failed
zsh: trace trap  GDK_RENDERING=image GTK_THEME=Adwaita:dark ~/gtkwave-gtk3-build/bin/gtkwave
```

gtkwave, 2021-03-12:
I think you can just put an abort() *after* the call to gtk_window_add_accel_group() in menu.c to see if that message appears. In Linux I get something like this:

(gdb) where
#0 0x00007ffff1e8f3d7 in raise () at /lib64/libc.so.6
#1 0x00007ffff1e90ac8 in abort () at /lib64/libc.so.6
#2 0x00000000004598b1 in alt_menu_top (window=<optimized out>) at menu.c:8908
#3 0x000000000044b8d5 in main_2 (opt_vcd=0, argc=2, argv=0x7fffffffdc98) at main.c:2118
#4 0x00007ffff1e7b555 in __libc_start_main () at /lib64/libc.so.6
#5 0x000000000040eea5 in _start ()
(gdb) print window
No symbol "window" in current context.
(gdb) print GLOBALS->mainwindow
$1 = 0xce4520
(gdb) quit

You might have to fflush(STDOUT) but I doubt it as those messages should go to stderr in order not to be invasive.

-Tony

markusdd, 2021-03-12:
are you directly dropping into gdb? That might explain it, under macOS this is called differently.

gtkwave, 2021-03-12:
Yes. I don't know how the debug chain works on MacOS. I figured it was the same.

-Tony

markusdd, 2021-03-12:
I was just being stupid...of course I need to run it with lldb....please find the backtrace below
[image]

gtkwave, 2021-03-12:
Thanks. That uses GLOBALS->mainwindow which is a window.

GLOBALS->mainwindow = gtk_window_new(GLOBALS->disable_window_manager ? GTK_WINDOW_POPUP : GTK_WINDOW_TOPLEVEL);

I don't know what is generating that message unless it's something weird in that version of GTK. I'll have to see if my Fedora VM kicks out any of those messages.g

566:/home/bybell/gtkwave/code/gtkwave3-gtk3/src> fgrep alt_menu_top *.c
main.c: menubar = alt_menu_top(GLOBALS->mainwindow);
main.c: menubar = alt_menu_top(GLOBALS->mainwindow);
menu.c:GtkWidget *alt_menu_top(GtkWidget *window)
menu.c: GLOBALS->main_popup_menu = menu = alt_menu_top(GLOBALS->mainwindow);

-Tony

markusdd, 2021-03-12:
could it be related to the way gtk-mac-integration interacts with the code?
(because of the menu bar affixed to the top bar etc.?)

gtkwave, 2021-03-12:
It's possible. If the function didn't execute properly, the hotkeys for menu items should not function. Do the key combos for various functions shown in the menu work? (They still might display but not do anything.)

-Tony

markusdd, 2021-03-13:
The hotkeys do work.

e.g. new tab (cmd+T) or insert blank (cmd+B)

gtkwave, 2021-03-13:
I don't know why it's giving an info message/warning then. It doesn't make any sense at the moment.

-Tony

markusdd, 2021-03-13:
What I have noticed is that it woll not open in the foreground. I get the splash screen and then it immidiately pops behind all other windows, which is very strange. GTK2 Version does not do that.
I will try to step through and see what these globals look like at that point.

markusdd, 2021-03-13:
ok that is not cool. LLDB is currently not displaying any variables while in a debug session.
GDB cannot be built yet for Apple Silicon it seems.

markusdd, 2021-03-13:
Ok, figured it out, turns out it requires -O0 -ggdb as CFLAGS but now I can get full debugging access.
Any variables/scenarios we should look at?

gtkwave, 2021-03-14:
The "window" in the function should be equal to GLOBALS->mainwindow which certainly is a GTK_WINDOW.
I know a GtkWidget* is being passed in, but that's why those uppercase casting functions exist.

Unless there is a bug in the toolkit, or memory corruption, etc., this should be ok.
gtk_window_add_accel_group(GTK_WINDOW(window), global_accel);

https://developer.gnome.org/gtk3/stable/GtkWindow.html#gtk-window-add-accel-group

Is this M1 silicon you're running on or native x86? 

-Tony

markusdd, 2021-03-14:
[image]

This is M1 silicon.

According to the debug session window is definetly defined.

markusdd, 2021-03-14:
Ok, the messages get printed somewhere else. It happens here:
[image]

markusdd, 2021-03-14:
could it be this? The AppType is listed as GtkOsxApplication, maybe this causes some of the casting issues?

gtkwave, 2021-03-14:
Thanks...so it's Mac-specific which is why I haven't seen it. I doubt it's related to M1 now unless there is some memory corruption issue, but thanks for checking.

I don't know what goes on under the covers in gtkosx_application_set_menu_bar(), but it sounds like its internal function has changed between GTK2 and GTK3. I've never seen those warnings when I've run in MacOS for GTK2 before.

For the "is not a window" thing, I'd have to dig through the gtk-mac-integration source.

The way to isolate to that line is to 
fflush(stdout); fprintf(stderr, "XXX\n");
function_call();
fflush(stdout); fprintf(stderr, "XXX\n");

To see if it's that line. 

http://gtk-osx.sourceforge.net/gtk-mac-integration/GtkosxApplication.html#gtkosx-application-set-menu-bar

void
gtkosx_application_set_menu_bar (GtkosxApplication *self,
 GtkMenuShell *menu_shell);

It specifically says GtkMenuShell, not a window. A GtkMenuBar in the hierarchical class order can be a GtkMenuShell, so I'm lost why they're casting it to a GtkWindow somewhere in their code as that is nowhere in the class hierarchy:

https://developer.gnome.org/gtk3/stable/GtkMenuBar.html

Object Hierarchy

 GObject
 ╰── GInitiallyUnowned
 ╰── GtkWidget
 ╰── GtkContainer
 ╰── GtkMenuShell
 ╰── GtkMenuBar

-Tony

[8 more replies]