gtkwave/gtkwave #46: "Open Scope" does not scrolls SST

ripopov, 2021-03-13:
"Open Scope" action in signal context menu highlights scope in SST, but it does not automatically scroll SST widget to make scope visible. This makes this action pretty much useless if you have hundreds of open scopes in SST, which is often the case when you do full-chip debugs.

Is it possible to scroll automatically to make scope visible?

gtkwave, 2021-03-14:
It absolutely does do that for me. I've used it on every single Zen design to date so I appreciate the complexity and scale of full-chip debug. :-)

What is the source (regular of -gtk3 branch), version, OS, etc. Something sounds weird here.

-Tony

ripopov, 2021-03-14:
Ubuntu 20.04, I've checked regular branch from git today. Let me check -gtk3 branch.

ripopov, 2021-03-14:
Tested:
SLES 12 gtk2 - does not work
Ubuntu 20.04 gtk2 - does not work
Ubuntu 20.04 gtk3 - works

So it looks like it works only with gtk3.  However what I've noticed is that gtk3 version is slow on remote server.

gtkwave, 2021-03-14:
Thanks. OK...actually you can get it to work with gtk2.

Compile the -gtk3 source but do not use the --enable-gtk3 ./configure flag.
However the slow on remote server might be an artifact of the -gtk3 source as it uses Cairo, etc. instead of x11 primitives.

<< I decided to dig deeper... >>

However, I think I have the gtk2 fix (around line 370 in treesearch_gtk2.c in the non -gtk3 sources):

 gtk_ctree_node_moveto(ctree, nodehist[depth-1],
 0, // depth /*column*/,
 0.5 /*row_align*/,
 0.5 /*col_align*/);

They fixed the function to do nothing if column was out of range. Originally back in 2006 it worked! It looks like a 15 year old bug triggered by a toolkit bug fix at some point.

Let me know if this works for you. I'll check this in.

-Tony

ripopov, 2021-03-14:
Thank you. Fix works!

ripopov, 2021-03-14:
Fixed : https://github.com/gtkwave/gtkwave/commit/69e2ca006e69c655d3de33e399a8c980d2129df6

closed 2021-03-14 by ripopov