gtkwave/gtkwave #48: Feature : automatically synchronize signal list in TwinWave

ripopov, 2021-03-13:
I mainly use TwinWave to find differences between two waveforms.  In this scenario you need to have same signals in both windows.

Current workaround : Create save file and open it in both windows. This is quite inconvenient when you need to add / remove signals.

gtkwave, 2021-03-14:
That is a very good idea and I'll think about it. Currently I'm far too busy with my day job.

-Tony

gtkwave, 2021-03-14:
BTW, as a quick note--I just remembered this...I don't know if you know this but you can drag signals from one window and drop them to another, and yes, it works when you run it in ++ mode. If you have a screen full of adjacent signals not meriting a full synchronization, highlight select them and simply drag them into the other viewer window.

-Tony

ripopov, 2021-03-14:
Fantastic! Did not know about this. Is it possible to drag more than 1 signal at once?

ripopov, 2021-03-14:
>Fantastic! Did not know about this. Is it possible to drag more than 1 signal at once?
It is. I just need to hold Shift when dragging.

gtkwave, 2021-03-14:
Right, shift or control (contiguous vs individual select/deselect) depending on how you were selecting/deselecting >1 signal.

-Tony