gtkwave/gtkwave #50: Indentation of the groups

Nic30, 2021-03-20:
I do have nested groups in "Signals" view:
[image]

Although the groups are nested they are displayed as they were of same level. If I click on them only a specific group highlights correctly so I guess that the groups are created correctly.

I expected something like this:
[image]

or 

[image]

Is it possible to make groups somehow look like a tree in GTKWave?

gtkwave, 2021-03-20:
I'll look into it. Thanks for the suggestion.

-Tony

rishubn, 2021-07-29:
I agree, this would be very helpful, particularly with GHW file and VHDL record types

Nic30, 2021-07-30:
@gtkwave  how long does it take from pull request to deb package?
Is there some "how to contribute" or I should just use common sense?

xXIty, 2022-08-23:
Hello, has there been any update on this? @gtkwave

gtkwave, 2022-08-24:
I've been far too busy with my day job so I haven't been able to work on this.
It's possible something similar will wind up in the gtk4 version as there has been talk of redoing how signals are handled as a tree emulated from a flat signal list, changed into an actual tree representation.  Please keep the feature request open as this could have traction for the future.

Thanks,
-Tony
Message ID: ***@***.***>

ColsonZhang, 2025-04-16:
So does the current gtk4 version supprt the function ?