gtkwave/gtkwave #51: Feature request: append/insert from SST with hierarchy

Nic30, 2021-03-20:
As far as I know there are several methods how to add signals to the main view.

1. right click on item in SST and "Recurse Import"
2. select thing in SST and then use list of the signals and append/insert buttons.

It would be nice to have the possibility to import the thing from SST while keeping hierarchy.
I would expect that the same sub tree will be built in main view using groups if I import signals with hierarchy.

It seems to me as the must have feature and I am emulating it with pre-generated .gtkw file.

My problem:

I usually work with AMBA ACE and similar interfaces composed of several channels which do have about 50+ signals in total.
Naturally interfaces of this size are on multiple places in the design and while debugging it is useful to trace the data on the channels of the interface (channel in this context is some group of signals). The amount of signals is so large that it is not practical to keep them all in the view. If they are grouped it is easy to just collapse non interesting channels of the interface, if there are not groups one end up with cyclic removing and re-adding the non interesting signals which is time consuming as they are usually deep in hierarchy.

I do have grouped signals in input trace and I see them grouped in SST, however I do not know if it is possible to just select some node from SST and add it into main signal view while keeping the hierarchy so groups of signals can be collapsed.

tbybell, 2024-06-21:
I'll think about this.

The main issue is that even though the name says "Recurse Import" and it appears to be a recursive tree walk, it's not really recursive in how it's currently implemented.  This is from 3.3 LTS, but gtkwave4 is similar:

`for(i=GLOBALS->fetchlow;i<=GLOBALS->fetchhigh;i++)`

It's just that all the signals in that conceptual visualization of a tree exist (in one form) consecutively in a 1-D array so instead of walking a tree, we know where the entire collection starts and ends and just iteratively import those signals.  It's an easy out the current implementation took.  A tree walk of course could bracket a group start, emit all the signals recursively, then emit a group end.  You can still do it in a linear fashion but you need to observe when the hierarchy lengthens or contracts/changes.

For those who are modifying C code, in dealing with loaders, whatever is done needs to be done twice, or is done with equivalent results so the signals that need to be imported from a dump file are marked before the dump file is traversed.  (We do this for efficiency so we only have to go through the dump file once.)  First pass: Signals are marked via set_s_selected() during the walk and the dumpfile loader is called.  Second pass: Then, when those signals are resident in memory, we import them into the GUI.  That's why there are two loops in the existing code.  The first loop is merely to establish for the loader which signals need to be imported.

Collapsible groups as currently implemented are a bit of a hack in that the traces in the Signals pane adjacent to the Waves one are still stored as simple linked list with a parent pointer added, so I don't know if this is worth working fixing now.  I could come up with a solution in LTS that could be ported to the current gtkwave4 implementation, but then it would have to get torn out and be "done right" later if groups are reworked to be more sane as I thought I remembered reading about a while back.

It's certainly a good suggestion from the original poster.  The current project I'm working on uses structs all over the place, so I can see its use.

rfuest, 2024-06-22:
> I could come up with a solution in LTS that could be ported to the current gtkwave4 implementation, but then it would have to get torn out and be "done right" later if groups are reworked to be more sane as I thought I remembered reading about a while back.

Yes, I wrote about that some time ago. The plan is to replace the flat trace list with a tree structure. But this will be added as part of the GTK 4 port, because it would be too much work to retrofit this into the existing UI code.