gtkwave/gtkwave #52: Feature: Copy signal name?

ripopov, 2021-03-22:
Is it possible to copy hierarchical signal or scope name (either from SST or Signals view) ? 
Use case: copy a signal name into Signal Search Regexep Window to find signals in same scope using regexp.

gtkwave, 2021-03-22:
Currently, no. Would you like to describe the sequence of operations you are thinking you would do if such functionality existed?

For example, the top and bottom panes of the SST would have a "copy name" (first, all, etc.) function into the usual text cut buffer for whatever is selected.

If you enable both View->Show Mouseover (and also Mouseover Copies To Clipboard) you'll see how signal names should be available either with the middle mouse button and/or ctrl-V, whatever is appropriate on a given system so you could paste it into an xterm, etc., which means it could be pasted into that regext window as well. I'm guessing you want something similar to that.

Thanks,
-Tony

ripopov, 2021-03-22:
Interesting. Yes, combination of  "Show Mouseover" + "Mouseover Copies To Clipboard" works for me.  Quite unexpected implementation though, usually Ctrl+C or other keys are used for copying into clipboard.

gtkwave, 2021-03-22:
It's a bit of a hack where I could point at a signal in the viewer then ctrl-V it into an RTL editor window. I've had to deal with machine-generated RTL with horrible naming conventions such as double underscores, very long names, etc.

I'll look at adding the to SST panes when I find the time. It's a good idea.

-Tony