gtkwave/gtkwave #53: Update stem file generation...

jahagirdar, 2021-04-01:
This repo https://github.com/gtkwave/gtkwave/compare/master...dh73:master seems to have some fixes for stem file generation. Would it be possible to pull these changes in gtkwave.

gtkwave, 2021-04-01:
> Counting objects: 20, done.
> Delta compression using up to 4 threads.
> Compressing objects: 100% (11/11), done.
> Writing objects: 100% (11/11), 2.10 KiB | 0 bytes/s, done.
> Total 11 (delta 9), reused 0 (delta 0)
> remote: Resolving deltas: 100% (9/9), completed with 9 local objects.
> To https://github.com/gtkwave/gtkwave.git
>  f31ae10..532bfb6 master -> master
>
>
>
>
> This repo master...dh73:master seems to have some fixes for stem file generation. Would it be possible to pull these changes in gtkwave.

Done, thanks.

-Tony

612:/home/bybell/git/gtkwave> git commit -a
[master 532bfb6] updated xml2stems.cc
 4 files changed, 734 insertions(+), 686 deletions(-)
 rewrite gtkwave3-gtk3/contrib/xml2stems/xml2stems.cc (65%)
 rewrite gtkwave3/contrib/xml2stems/xml2stems.cc (65%)
613:/home/bybell/git/gtkwave> git push origin master
Username for 'https://github.com': gtkwave

umarcor, 2021-04-01:
@gtkwave, should this be closed? :wink:

gtkwave, 2021-04-01:
Yes. Thanks for the reminder--it's been one of those days.

-Tony

closed 2021-04-01 by gtkwave

jahagirdar, 2021-04-03:
Sorry this still looks a bit broken... I can try to fix it. Is the stemsfile format defined someplace?

I see the following fprintf's in the code:

```
 fprintf(fo, "++ module %s file %s parent %s lines %d - %d\n", mName.top().c_str(), mnam, topname, lineno, lineno);
fprintf(fo, "++ module %s file %s lines %d - %d\n", nam, mnam, lineno, lineno); /* don't need line number it truly ends at */
fprintf(fo, "++ module TOP file %s lines %d - %d\n", "(VerilatorTop)", 1, 1);
fprintf(fo, "++ comp %s type %s parent TOP\n", nam, nam);
fprintf(fo, "++ comp %s type %s parent %s\n", nam, defnam, mId.top().c_str());
fprintf(fo, "++ udp %s file %s lines %d - %d\n", nam, mnam, lineno, lineno); /* don't need line number it truly ends at */
```

I am looking for details of 

1.  what are the valid values for the type argument to comp.
2. What do the two lineno arguments indicate?
3. Is it sufficient to dump the filename+ modulename/instancename or do we also need to dump the reg/wire declaration.

gtkwave, 2021-04-03:
The whole purpose of the file is to be able to allow rtlbrowse to regenerate the hierarchy tree. It doesn't care about wires or regs.

The lines range is supposed to have the start and end lines of the module, but I don't remember if Verilator dumps that info.

The original program that created stems was "vermin" whose last inclusion in the source was 3.3.92:
https://sourceforge.net/projects/gtkwave/files/gtkwave-3.3.92/gtkwave-3.3.92.tar.gz/download

If you look at verilog.g, that should answer all your questions. Mainly, I didn't want to have to support dragging along a back level version of PCCTS along with a Verilog parser. Verilator was the easiest way to achieve this with open source tools. VRQ might have worked as well. If I were going for a commercial project, I'd license Verific and have it dump out something similar to what Verdi's Verilog compiler does. We use Verific at work for various internal tools with excellent results.

-Tony

jahagirdar, 2021-04-04:
I reverse engineered the stem file format from vermin. My observation
- Verilator gives the line number where the module is declared but does not provide the line where the end module is defined.
- RTLBrowser works fine if only linestart is given but line end is skipped.
- xml2stems.cc does not use a proper XML parser, It "knows" the position of required attribute in a tag, and gets the string at that position. As as when new attributes are introduced, the position may shift resulting in xml2stems not working.
- I have written an XSL transformation stylesheet which will be more robust. The downside to this is it requires the user to download and install an xsl processor from https://sourceforge.net/projects/saxon/
- I have added the stylesheet to the contrib folder and created a pull request.

gtkwave, 2021-04-04:
I'll look at accepting your pull request later. I have to updates Makefile.am if you didn't, or it will get lost when a "make dist" is performed.

What specific attributes are not being located properly? I take it that a newer version of Verilator breaks things? Please be specific what you've observed.

Thanks,
-Tony

jahagirdar, 2021-04-05:
For the attached XML file:

Issues with xml2stem output are,
- The module name is incorrect
- The Var list is missing
- Only the top module is listed, module a and bm are not.

Reading the code what I understood is, It tries to implement a rudimentary xml parser by looking for open tags e.g. `<i` and then looking for quote characters, and picking up the 'N'th quoted string.

e.g. The module name is picked up from the statement
```
 <instance fl="e6" loc="e,6,4,6,6" name="b1" defName="bm" origName="b1">
```
xml2stems.c puts a pointer at each `"`  and tries to get the module name from the 4th pointer, It ends up getting the loc string ""e,6,4,6,6" instead. 

For the attached file, xml2stems generates

[Va.xml.gz](https://github.com/gtkwave/gtkwave/files/6257819/Va.xml.gz)

```
 xml2stems obj_dir/Va.xml

++ module e,1,8,1,11 file top.v lines 1 - 1
++ comp e,6,4,6,6 type b1 parent e,1,8,1,11
++ comp e,7,4,7,6 type b2 parent e,1,8,1,11

````

Where as what is generated from the xsl parser is
```
 java -jar ~/saxonhe/saxon-he-10.3.jar -xsl:../stems.xsl obj_dir/Va.xml

++ comp b1  type bm  parent top
++ comp b2  type bm  parent top
++ comp a0  type a  parent bm
++ comp a1  type a  parent bm
++ comp a2  type a  parent bm
++ comp a3  type a  parent bm
++ module top file top.v lines 1
++ module bm file b.v lines 1
++ module a file a.v lines 1
++ var a  module top
++ var b  module top
++ var x  module top
++ var a  module bm
++ var b  module bm
++ var x  module bm
++ var b  module a
++ var c  module a
```

gtkwave, 2021-04-05:
Thanks. I'll look at a newer version of Verilator. I suppose Steve updated his writer.

I'll still integrate your script somewhere in the distro as it's a more robust solution.

-Tony

gtkwave, 2021-04-06:
Thanks. I experimented with the new Verilator and updated the xml2stems parsing to handle arbitrary tags and locations.
There is more going on though:

1) generate statements, specifically named/iterated hierarchies for them underneath the current one.
2) Verilator starts renaming modules sometimes and you have to look at <cell> definitions to see what their real name is. It doesn't appear origName works all the time.

I have a non-trivial set of Verilog from work I can use as a large testcase on this. As it stands though, #1 is a dealbreaker for proper function--and has been forever--as I have to update rtlbrowse to handle this. I'll mess with it later in the week. I probably never noticed because long ago, back when I originally wrote rtlbrowse, I never used generate statements.

-Tony

jahagirdar, 2021-04-07:
# 1 is not a problem for me because most of my code is generated from an HLS language. I had noticed # 2 with my test code, origName worked in my case... In case there are exceptions I will be interested in looking into them and updating the xsl file.

What is the API used by RTL Browse to fetch the current time and signal values from GTKWave? 
Can this be converted to a socket service? (There is one GTKWave fork in github which partially does this)
I am interested in writing an HTML+JS version of RTL browse to support 
1. Native debugging for HLS languages.
2. hyperlink variable to their declaration/assign statements etc..
If this sounds Interesting, I will open a second ticket to discuss and followup on this...

umarcor, 2021-04-11:
@jahagirdar see https://github.com/gtkwave/gtkwave/issues/9#issuecomment-626222344