gtkwave/gtkwave #55: Feature request: Filtering by signal direction

Blebowski, 2021-04-08:
Hello, I have a small idea for improvement. Would it be possible to add filtering of signals within hierarchy
by direction?
In "signals" listbox of current hierarchy, there would be one row of small buttons (below or above of "Filter")
with descriptions like "IN" , "OUT", "INOUT", "INT" (like internal). These would act like "combo-boxes", when
"toggled", they would allow displaying these signals. When "un-toggled" they would ignored signals of respective
directions.
I find this feature really handy in commercial simulators.
AFAIK, information about signal direction is available thanks to `enum tree_kind`. Is it possible to do this?

gtkwave, 2021-04-09:
In the search box: prefix your search string (if any) with these:

+i+ for input
+o+ for output

OK, here it is from the tooltip that pops up if you hover over the filter box. Maybe your window manager hides those:

 gtk_tooltips_set_tip_2(GLOBALS->filter_entry,
 "Add a POSIX filter. "
 "'.*' matches any number of characters,"
 " '.' matches any character. Hit Return to apply."
 " The filter may be preceded with the port direction if it exists such as ++ (show only non-port), +I+, +O+, +IO+, etc."
 " Use -- to exclude all non-ports (i.e., show only all ports), -I- to exclude all input ports, etc."
 );

...I didn't want to burn screen space for buttons long ago, so I made it a part of the search string.

-Tony

Blebowski, 2021-04-10:
Hi,
thanks for comment, I see it now. Do you think it would be possible to convert this feature to the GUI/combo-box
solution? I would definitively appreciate it since it allows:
- quicker work (toggling button is shorter than writing +IO+ pattern) = more user friendly
- combining display/hide of multiple object types at once
- allowing possible future extensions of different object types (at the moment, I am working on extending GHDLs GHW
  dump, to dump state of PSL assertions, so "assertion" would be next object type).
- more intuitive (combo-box with "in" on the button is very intuitive, while reading the tooltip not so much).
- window size is not an issue these days (I can imagine that was one of the reasons for "not burning screen space",
   years ago with low display resolutions...)
What do you think?

gtkwave, 2021-04-10:
I'll consider it in the future. Currently, I'm pretty busy with my day job.

-Tony

Blebowski, 2021-04-13:
Of course :) Thanks for reply.