gtkwave/gtkwave #59: How/when is the version of GTKWave bumped

umarcor, 2021-04-28:
There was some question about GHW support in the GHDL chat room. When I checked the latest version of GTKWave, I found that I bumped it to v3.3.108 in MSYS2 on January 16: https://github.com/msys2/MINGW-packages/commits/master/mingw-w64-gtkwave. However, according to https://github.com/gtkwave/gtkwave/blob/master/gtkwave3-gtk3/ChangeLog#L1807, that version was released on 10apr21.

Tony, does it mean the the "HEAD" tarball is a rolling one and it is updated until you bump to the next version? What am I missing?

gtkwave, 2021-04-28:
3.3.108 was released on Dec 30th.
See bottom of page: http://gtkwave.sourceforge.net/
https://sourceforge.net/projects/gtkwave/files/

I'm way overdue for an update to 3.3.109. Every time I'm ready to do it, new bugs come up! :-)

In fact I forgot to update the version in revision control to 3.3.109.

As far as what tarball github creates, I have no idea. I think the ones the Linux distros pick up all come off SourceForge, such as an update to the gtkwave.sourceforge.net triggers a re-release. I think at least that is what Fedora does.

-Tony

umarcor, 2021-04-28:
Thanks a lot for clarifying! It makes sense now!

> As far as what tarball github creates, I have no idea. I think the ones the Linux distros pick up all come off SourceForge, such as an update to the gtkwave.sourceforge.net triggers a re-release. I think at least that is what Fedora does.

I do use tarballs from sourceforge too: `https://gtkwave.sourceforge.io/${_realname}-gtk3-${pkgver}.tar.gz`.
In order to use tarballs/archives from GitHub, we would need you to push tags (or version branches) to this repository. Currently, the only "named" reference is "master". I'm not requesting you to do that, though. I'm good with using the tarballs from sourceforge.

closed 2021-04-28 by umarcor