gtkwave/gtkwave #61: GUI refresh issues on Windows 10 (MSYS2)

taras-rosa, 2021-05-01:
Hello everyone,

I'm facing some problems with the waveform view on Windows 10: Zoom commands, marker movement and other GUI actions don't respond immediately if I have only one tab opened. It is needed to click several times on the same button and/or move cursor back and forth from the "Waves" to "SST" region to make it refresh. 
But the problem goes away if I open the second tab (even with the same file). In such case the waveform views in both files work as expected (as you can see the animation attached).
[image]
I built the latest version from this repository in MSYS2 (f56ae548d8be08efba72e8c6f813dcdddd2e460a) by "standalone_pkg.sh" and also tried the package from MSYS2 pacman repository (mingw-w64-x86_64-gtkwave 3.3.108-1) - both of them behave the same.

gtkwave, 2021-05-02:
Out of curiosity, does it do it with the most recent build found here:
https://sourceforge.net/projects/gtkwave/files/
...these are using my own libraries.

It sounds suspiciously to me like there is a gtk library problem.

-Tony

 
 

I built the latest version from this repository in MSYS2 (f56ae54) by "standalone_pkg.sh" and also tried the package from MSYS2 pacman repository (mingw-w64-x86_64-gtkwave 3.3.108-1) - both of them behave the same.

umarcor, 2021-05-02:
I can confirm the behaviour. @taras-rosa, when you install GTKWave through MSYS2, you get two executables: gtkwave and gtkwave-gtk2. If you use the second one instead, the problem should be gone, as it seems to be something specific about GTK3.

I have this patch ready for changing the default to the GTK2 version: https://github.com/umarcor/MINGW-packages/commits/gtkwave. I was holding it because I expect Tony to release 3.3.109 "soon". Also, I had the hope to find the underlaying and so annoying root of the problem.

taras-rosa, 2021-05-02:
Yes, GTK2 version works good. I use it for now.
Ok, so looking forward for the GTK3 fix.
Thanks for your replies.

gtkwave, 2021-05-02:
https://sourceforge.net/projects/gtkwave/files/
http://gtkwave.sourceforge.net/

...3.3.109 should be up by the time you read this. Sorry about the delay. I usually don't go 4 months between releases. This one was a bit long.

-Tony

umarcor, 2021-05-09:
I updated the GTKWave package on MSYS2, and it was upstreamed alread. Run `pacman -Syu` for getting 3.3.109, which defaults to GTK2.