gtkwave/gtkwave #63: Unable to read GHW wave file from a design that uses a generic package

patrickrst, 2021-05-13:
I compiled with GHDL a VHDL design that uses a generic package. Here is the error I get when trying to read the waveform with GTKWave:

```
$ gtkwave wave.ghw
GTKWave Analyzer v3.3.98 (w)1999-2019 BSI

ghw_read_hie: unhandled kind 196
Aborted
```

Same error with GTKWave v3.3.109 compiled from master branch.

gtkwave, 2021-05-14:
That error is being thrown by the GHW library so Tristan would need to be asked about it.

-Tony

patrickrst, 2021-05-14:
I was not able to reproduce the output. Now, I get the same GHDL bug as ghdl/ghdl#656

Anyhow, here's my (not working) example: https://gist.github.com/patrickrst/585fcea3eb5e2df48ae1e3cc3449eb1b

I will see if I can find what I did differently.