gtkwave/gtkwave #70: FST API documentation

pieter3d, 2021-08-19:
Hello - Is there any documentation on the functions in fstapi.h? I can kinda figure some things out from their use in `fstminer.c` but a lot is pretty non-obvious, and the header file has very little comments. The PDF discusses the compression techniques but not the API.

gtkwave, 2021-08-20:
For the API, the best place to look for a writer is in vcd2fst.c. It should be fairly obvious how the VCD primitives more-or-less map 1:1 to the FST function calls.Sourcecode for Icarus Verilog demonstrates it well.Sourcecode for OpenCVC also has FST writer support.

The reader is a bit harder and you can look at fst.c in gtkwave if you want to work on a reader.  It's harder because it needs to use callback functions to get data out of the file, but otherwise is fairly straightforward.You mentioned fstminer so it appears you're interested in reading from a file.  The fst.c file in gtkwave is probably the best place to look.
open FST file and get a reader contextread info like number of signals, etc. through various query functions
iterate through signal names, build scopes, etc.  fstminer does this with flattened signal names.set a signals of interest mask using functions with names with Mask in them like fstReaderSetFacProcessMaskAll.fstReaderIterBlocks() is the callback to get the read data out of the file.  fstminer demonstrates this easily.close the context whenever you're done.
I can go into more detail but those are the basics.

-Tony

pieter3d, 2021-08-20:
Thanks for the pointers, I'll dig in some more.

pieter3d, 2021-08-22:
Okay, two questions:
- What is the point of the `fstReaderPushScope` and `fstReaderPopScope` functions? Things seem to work without them.
- I tried setting a limited time range with `fstReaderSetLimitTimeRange`, but calling `fstReaderIterBlocks` still results in value callbacks for the entire time range in the FST file. I did notice that if the start_time is set to anything other than 0, nothing at all gets dumped (no callbacks). Is there just some larger granularity that I am missing?  The FST file I was playing with is only 330kB.

gtkwave, 2021-08-22:
The push scope and pop scope functions are for helping out tools like gtkwave create their own internal data structures with respect to generation of hierarchy trees, etc.  They're not generally needed.  I could have just as well done my own push/pop stuff in gtkwave, but I figured if I needed it, other tool writers might.
For the time limit range, the file needs to be sufficiently long enough.  330kB won't do it.  Processing of the FST is block based and this merely trims it to the nearest block.  Blocks are quite large to aid in compressibility as it reduces overhead that needs to exist in every block, and also groups bigger blocks of like data with respect to what is going on in sim.  If you have larger files in the MBs range, it'll limit the range.
-Tony

pieter3d, 2021-08-22:
I see. But if setting the start value to something inside the block causes the whole block to be dropped, that means I need to know where the block boundaries are. How could I know that?

In any case it would be nice for the API to skip callbacks outside the desired time range, even if it internally has to parse the data.

gtkwave, 2021-08-23:
The whole block isn't dropped.  If we have a first block that is from 0ns-9.999ns, then a second from 10ns-19.999ns and you specify 15ns = start, then we should start at 10ns.  The block isn't dropped.

Callbacks could be skipped, but you have an X problem of sorts...
You want to start at 15ns.The previous value change on a signal is 13ns.  You didn't see it if the callbacks were filtered.

The solution is really a wrapper function around the callback:Collect all value changes.At the end of the start time, emit the most recent value for the signals you're dumping.Continue as normal.
gtkwave fakes it by making the data < 15ns in that first block not visible.  I don't believe I do any special trimming of data.  That is, I got it to work but never optimized it for memory usage.

-Tony

pieter3d, 2021-08-23:
Hmm - Take a look at the attached FST.
[waves.fst.gz](https://github.com/gtkwave/gtkwave/files/7028455/waves.fst.gz)
If I set start time to 0, I can dump all values. If I set it to 1, I get no values, in the following code:
```c++
  uint64_t start_time = 1; // lots of values when set to 0.
  fstReaderSetLimitTimeRange(r, 0, 400'000);
  fstReaderSetFacProcessMask(r, h_some_signal);
  fstReaderIterBlocks(
      reader_ctx_ptr,
      +[](void *user_callback_data_pointer, uint64_t time, fstHandle facidx,
          const unsigned char *value) {
        printf("val: %s @ %ld\n", value, time);
      },
      nullptr, nullptr);

```

gtkwave, 2021-08-23:
Try the latest.
-Tony

686:/home/bybell/git/gtkwave> git push origin masterCounting objects: 42, done.
Delta compression using up to 4 threads.
Compressing objects: 100% (23/23), done.
Writing objects: 100% (23/23), 3.54 KiB | 0 bytes/s, done.
Total 23 (delta 22), reused 0 (delta 0)
remote: Resolving deltas: 100% (22/22), completed with 19 local objects.
To https://github.com/gtkwave/gtkwave
   1465738..ee52b42  master -> master

pieter3d, 2021-08-23:
Okay, seems to work now. thanks!

pieter3d, 2021-08-28:
Is there a way to selectively iterate the hierarchy and scopes? It would nice to not have to go through the entire design, instead go more on demand, only read the list of signals in some scope as needed. That avoids the costly indexing of the whole design tree at start.

gtkwave, 2021-08-28:
No, there's no way to do that.  Even for FSDB, at best I think you can iterate through either the full hierarchy per-tree (as dumped, usually as logic in one and MDAs in another) or that of scopes but there's no real shortcut.  I might be wrong, but I thought that's how it worked for them.  I suppose they have a private API that lets them zoom to a specific chunk of hierarchy. 

fsdb_obj->ffrSetTreeCBFunc(__MyTreeCB, (void *) cb);
fsdb_obj->ffrReadScopeVarTree();

How big are the designs you're looking at?  I've been dealing with ones that have tens of millions of signals, which can underscore the potentially costly startup aspect.  Designs of a couple of hundred thousand signals usually aren't noticeable as far as startup cost goes.

The main limitation is that the hierarchy and scopes are dumped in a linear fashion in the same order as received by the writer, so related signals in the same scope could conceivably be far away.  That is, there is no in-memory buffering and reordering to prevent signal declarations from being interleaved with descending into lower levels of hierarchy, popping back up, and dumping more signals at the original level of hierarchy.

-Tony

pieter3d, 2021-08-28:
Okay, thanks for the context. There's not actually a performance problem in the toy designs I've been working with, just that I was wondering if it could be an issue at some point. Sounds like for most applications it isn't a real concern

pieter3d, 2021-08-30:
How about disk thrashing for a bunch of calls to `fstReaderIterBlocks`? Each call does an fseek(SEEK_SET), so basically the whole  file gets reprocessed every time?  `strace` on gtkwave shows `lseek` calls every time a signal is added. 
Just curious about potential processing lag in large files, especially if they are on spinning disks instead of SSDs.

gtkwave, 2021-08-30:
Who cares about spinning disk.  Think NFS.
Groups of signals are bunched up for adding.  If you add one at a time, you will have to retraverse the entire database once for each signal.  If you add 20 signals at once, you go through the file once for those 20 signals.  That's the recommended way of doing it: specify all the signals you want up front.  For example, when you process a gtkwave save file, that should be done all in one pass.  If you convert to VCD, it's all done in one pass.

Anyway, it doesn't matter though because modern OSs have buffer cache for their filesystems so we're largely running out of memory.

-Tony

gtkwave, 2021-08-30:
BTW, side note but as far as I can tell when I've benchmarked it in side-to-side comparisons, FST is comparable to FSDB so I don't really see any issues.  FSDB oddly enough gets hurt very bad when reading an entire database (such as when using fsdb2vcd).  I've seen cases with large numbers of signals where fst2vcd is 6x or so faster.
It's weird that nobody else has really come up with a fast and efficient open source competitor to FSDB.  It astounds me especially since I came up with FST first in 2009, so people have had plenty of time to have their thinking stimulated and invent a better database format.

pieter3d, 2021-08-30:
I suppose it's a pretty niche market, not a ton of open source folks working with big digital time domain data, and probably Synopsys doesn't really have incentive to spend resources on FSDB since it's pretty entrenched. No EDA vendors would directly benefit from pushing open interoperable open formats sadly. FST is a decent de-facto open standard, though it could perhaps benefit from a formal spec.

gtkwave, 2021-08-30:
I think you have a valid point there...I was at IBM when I started with this and have been at AMD almost 10 years so I have access to huge models such as all the Zen core models given that I'm an RTL unit lead.  It's part of the reason gtkwave also supports FSDB: I don't want to have to use nWave/Verdi at work so I wrote my own loader in gtkwave for it as AMD has the FSDB reader libraries and headers available in their installs.  So needless to say, I got to benchmark and tweak over the years and know what works.

There's stuff I could obviously do better for an FST follow-on, but the day job is killing me at the moment.  A formal spec would be a benefit, but probably with a cleaned up follow-on, mainly because there is some backwards compatibility cruft in there that shouldn't optimally be in there but is, and putting it in a spec seems burdensome for implementors (not that I would ever expect any).

The biggest pain is that for writing the file, there's an in memory phase where value changes are buffered, then when the buffer is full, there's a "context flush" where they're reformatted and written to disk with some form of data deduplication and/or compression involved.  A spec could describe the file format, but most writers would be clueless how to get from what's coming out of a simulator (or VCD file) and what needs to go to disk.  As far as I know, Novas was the 1st entity to invent the buffering+reformatting concept in the late 1990s, followed shortly by IBM in the early 2000s with their AET2 format, which was designed by a Hungarian mathematician.  A spec would probably turn out to be a 100 page monograph when all was said and done.

tomverbeure, 2022-01-07:
Using this issue since it's a bit of an FST API documentation dumping ground.

`fstapi.h` has the following function:

```c
int fstReaderIterBlocks2(void *ctx,
    void (*value_change_callback)(void *user_callback_data_pointer, uint64_t time, fstHandle facidx, const unsigned char *value),
    void (*value_change_callback_varlen)(void *user_callback_data_pointer, uint64_t time, fstHandle facidx, const unsigned char *value, uint32_t len),
```
I'm trying to figure out when I can expect a call to `value_change_callback_varlen`. 

`value_change_callback` already returns a null-terminated string (and thus supports unrestricted length values), so it's unclear to me how `_varlen` would be of any benefit.

Tom

gtkwave, 2022-01-07:
I don't remember exactly, but I think that zero bytes are fair game for the varlen callback payload, which is why there is a length field included.  See the usage of fstUtilityEscToBin in vcd2fst.c.

-Tony

ecstrema, 2025-02-02:
I believe that with the growing demand for open source HDL languages and tools, a formal specification for what seems to be the only viable alternative to VCD should be thought of.

I think FST would gain a lot from having a definite standard. Many libraries and apps already support it, and for a lot of these, it's directly gtkwave's C code that is used.

pieter3d, 2025-02-02:
I would love to work on this actually, is there a working group? Chips Alliance might be able to support (I'll ask). I have a background in video and data compression so as you can imagine I already have lots of ideas for requirements and techniques for such a format.

wsnyder, 2025-02-02:
With both my involvement in Verilator and CHIPS Alliance, assuming that the GTKwave authors are accepting, I'd like to completely encourage this effort!

Some ideas:
1. It seems like the proposal is to start with FST - I think that's the right direction!
2. First separate what is current FST as the baseline new-standard version. This would include separating from GTK wave and into a new github repository etc.  Presumably the reading/writing library would move there (avoid duplicating in GTKwave).  Tools like Verilator can then point to that new standard repo vs GTKwave.
3. Then consider enhancements, and implement.  I'd really like to see e.g. parallelism as in #286 as there's a massive simulation speedup possible there.
4. Avoid "hard versions" that are infrequent (like SystemVerilog does), instead allow continual improvement pulls with new features implemented in a backward-compatible way.  That's the power of open source definitions and contributions.

tbybell, 2025-02-03:
I definitely think it's a good idea where the FST library gets split out into its own repo.  In a sane world, the database defines the tool as the tool's internals can be optimized around various aspects such as on-demand loading of signals rather than that being grafted in as an afterthought.  Starting from a VCD loader that slurps the entire file into memory is an OK starting point for a tool or for bringing up a tool, but it's not going to be good for large, commercial designs.  I saw a similar evolution going on with IBM's internal tools almost three decades (!) ago.

With respect to a formal spec, such a thing really would only be suitable for an FST-inspired follow-on format that drives a new follow-on implementation, mainly because there are some some clumsy aspects to the file format that were added to make the file format upwardly compatible when a new method was figured out for how to do something.  I didn't want to have to deal with users who couldn't read old sets of waves.  Those "new" features were added around 2011 and the year is now what?  Documenting what amounts to incremental evolution cruft wouldn't be very clean unless the purpose was documentation for historical reasons.

For the aggregation of multiple FST files, a wrapper over the current reader probably would get the job done similar to FSDB with its .vf files pointing to a series of FSDB files.  I haven't looked at how to do it with FST yet though.

rfuest, 2025-02-03:
I had planned to move `libfst` into it's own repo for quite a while now to make it easier for other projects to depend on it. I'll try to find some time to move the current code into the new https://github.com/gtkwave/libfst repo. We'll need to figure out what to do about the build system (GTKWave use meson and Verilator uses CMake) or if we primarily want to install `libfst` as a shared library.

I agree with Tony that a formal specification would only make sense for a new format. If there is enough interest I could create a `fst2-wg` repo to discuss this.

pieter3d, 2025-02-03:
There's an email thread going now with CHIPS Alliance, looking to make this effort an official working group. I can add you folks to it (need an email address), and I will also post any mailing lists or repo's created here.

tbybell, 2025-02-03:
You can grab my email at the bottom of https://gtkwave.sourceforge.net/ thanks.

wsnyder, 2025-02-03:
Don't worry about Verilator as far as building, it just copies the writer sources from a repo checkout (currently gtkwave and in the future libfst or whatever) and writes its own cmake/make rules.

ecstrema, 2025-02-04:
Having libfst in its own repo will already be a nice improvement, but I also agree that a full specification only makes sense for a new format (or fst2 if sufficiently related). 

To me the most important parts of a format are (unordered):

- [ ] relatively good compression (comparable with fst)
- [ ] High compression speed (needs realtime encoding)
- [ ] Streaming support. A file should be able to be open in a simulator while being appended too. This is currently the case for fst AFAIK. 
- [ ] Fast random time access. FST stores data in chunks of time intervals, which allows jumping relatively fast to any specific timestamp. However, it uses large blocks to aid for compression, so this is only useful for large simulations. 
- [ ] Fast random signal access. As opposed to other formats, such as video/image/audio where all pixels/samples are used, the consumer of a waveform only chooses to view a small portion of the file.
- [ ] sufficient metadata section.

The metadata section has a lot of room to improve over other formats. These usually only provide the signal name and number of bits and/or basic root type, where a proper metadata section would include:

- computed type (this can be a simple informative string)
- in the case of enums, a mapping of values to their name

Since type information is defined once at the start, I don't think it will make a noticeable difference in file size, especially for large simulations.

Now the fun part. 

A waveform viewer is a two dimensional viewer: for each signal, we need to be able to map a value to a time. 

That means that essentially, a waveform is an image, where the columns are the signals and the rows are the values at every point in time. Advancing simulation simply means adding a row at the bottom. 

You know with a poor internet connection, when an image loads from the top, and the bottom part slowly appears as it is streamed. 

[image]

Well this already looks like a format which could be a great source of inspiration. 

Great. But JPEG is [mostly lossy](https://en.m.wikipedia.org/wiki/Lossless_JPEG). 

PNG offers a great "custom" channel for encoding images with a small number of colors. This is in line with the different values a signal can take. It also encodes efficiently similar and contiguous data, which is great for signals changing together. 

To improve compression, and ensure streaming support, video codecs could also be used: simply encode signals in a grid rather than putting them in individual columns. Since the number of signals is known when the simulation starts, the frame size can be fixed to a correct size directly, and data streamed directly in subsequent frames. 

Using typical codecs (especially video) means that hardware support would make things tremendously faster to encode and decode. 

Finally, using image or video codecs gives us a baseline for benchmarks. It means we can try to encode any simulation in a few lossless image and video formats and see what the encoding time, decoding time and final size look like. 

Since they are not optimized for waveforms (obviously) this will not be a fair comparison for image formats, so the reverse should also be done: encode an image in an fst file.

I am very curious to see if advanced lossless compression techniques used in image and video codecs adapt well to waveforms.

This should give a great number of ideas regarding encoding and decoding speed, along with some compression numbers. 

Cheers

Timmmm, 2025-02-04:
Image an video codecs aren't really applicable here. They take advantage of the fact that spatially adjacent pixels are not independent, but that isn't really the case for waves. FST already encodes signals in blocks (over time) and allows aliasing waves within a block (for signals that are identical within the block, which is common).

I think your requirements list is quite good though. I would add:

* Ability to store delta cycles, and order value changes within an instant.
* Support for transactions.
* Rich types - you mentioned enums but there are also structs and arrays.

I think this issue probably isn't the best place to discuss it anyway.

ecstrema, 2025-02-04:
> They take advantage of the fact that spatially adjacent pixels are not independent, but that isn't really the case for waves

Well we decide how we place our waves. Simulators can very well place related signals together.

> Support for transactions.

What kind of transaction do you mean?

> Rich types - you mentioned enums but there are also structs and arrays.

structs and arrays are currently supported in the sense of grouped signals right now, I wonder if more is necessary. Providing the full type could be interesting, but that opens another full can of work, whereas enums only match a selected value to its name and does not act on groups of signals.

> I think this issue probably isn't the best place to discuss it anyway.

Sorry, you are very right, though I couldn't resist in sending first thoughts.

rfuest, 2025-02-04:
I've just uploaded the libfst code to the new repo. It still needs some work, but it's a start.

@tbybell: There's a comment about Judy in the source, which suggests that the replacement is probably faster. Did you do any further testing on this? If possible I would like to remove Judy support to simplify the license to always be MIT without any optional LGPL dependencies.

tbybell, 2025-02-04:
I'd say remove it.  Less is more sometimes.

Timmmm, 2025-02-04:
> What kind of transaction do you mean?

Commercial simulators let you record transactions in the waves, which have a start/end time, a name, and some metadata. The most common use is for bus protocols, but you can use them for anything. I use them to record RISC-V instruction execution, so you can see the decoded instructions in the waves. Here's an example in Questa (I don't have any non-transaction waves here but you can add them above/below the transactions; it's just another row in the waves).

[image]

They can have metadata, sub-transactions (Extract, Issue), styling, etc. Very useful.

wsnyder, 2025-02-04:
Consider how to trace objects/classes that are dynamically created and destroyed during execution.   Current FST is all finalized at time zero.

pieter3d, 2025-02-04:
It would also be nice to support time gaps, for when simulators start/stop/resume wave dumping. And it would allow for easy-ish concatenation of two files. Related: Files/streams that are truncated should still be usable up to the truncated portion.

The CHIPS Alliance folks asked for a slide or two that introduce the concept, benefits, project scope, timeline etc in order to approve and create the workstream.

tbybell, 2025-02-04:
Blackout regions are already supported.

tbybell, 2025-02-04:
Note that for concatenation and such, that is part of the motivation for a block based format.  I was going to write some form of fsttools at one point, but never got around to it.  Files can be read in progress as they are being written, but at least one block needs to be emitted and finalized in the dumpfile.

pieter3d, 2025-02-05:
[Here's a quick stab](https://docs.google.com/presentation/d/1DPxEnFCLY2rNfUrllBde9QHfP5iaqk-KfkiIx7bLD5U/edit#slide=id.p) at a working group proposal  for CHIPS Alliance. Google Sheet, world-comment-able.

ecstrema, 2025-02-18:
one thing that came to my mind while reading the specs for fst, is that every signal is generic, and can hold any data, which is good, but a lot (I don't have numbers for this) of signals in a simulation are counter and clocks. It might be worth investigating if some pre-compression through counter and clock detection might be worth it.

Also, a lot of signals are defined directly in accordance to other signals, beign either the sum of two or more other signals, or a bit selection of the two, or various other simple arithmetic operations. These also might be candidates for optimization. It would be much more compact to encode: B is the LSB of A than to save the full waveform for A and B.

Careful care needs to be taken with this however, because arbitrary operations adds security considerations, and a limit to the amount of operations would be needed.

tbybell, 2025-02-18:
Precompression was used in that capacity for the original LXT.  You'll see it referred to in source or docs as "clock compression".  It's not going to buy us anything here given how the signals in memory for FST are already separated, so the compression layer on top of emitting signals (e.g., LZ4) during block processing will more efficiently do this work for us.  Any gains, if present, would be marginal and would erode writer performance.

I don't understand your comment on security considerations unless you mean somebody could craft buffer overflow attacks by crafting logic eqns.  Someone could do just the same on the internal data of the FST file and a researcher did do that a little over a year ago:
```
-rw-rw-r--.  1 bybell bybell 224376 Jan 15  2024 GTKWave Vulnerability Report.zip
-rw-rw-r--.  1 bybell bybell  42221 Dec 17  2023 Talos Advisory Report.zip
```
No format will ever truly be secure, so I don't consider security outside of the good programming hygiene to be a primary design consideration.  Some tool can and will find holes.

Anyway, back to the idea: a version of that method is used by ZTDB for emulators.  All you store in the dump are primary inputs, storage elements, and forces, then the combinatorial parts of a logic cone can be reconstructed or "expanded" in the viewer as needed.  This is something that could be done outside the waveform file, because it's really an application of waveform data vs design data to do some amount of resimulation.

ecstrema, 2025-02-18:
@tbybell So if I understand well your comment, you mean that you think that clock and counter compression can be handled more efficiently directly with the LZ4 filter, but that a simplified method for describing symbols as dependent on the others could be worth investigating?

tbybell, 2025-02-18:
"It would be much more compact to encode: B is the LSB of A than to save the full waveform for A and B."

Note that VZT dynamically does a version of this searching for candidate bits, and that was an outgrowth of what IBM did with AET2.  VCD already has the concept of aliases, but this works dynamically.  A dynamic method at the bit level can provide for fantastic compression when blocks have contain a decent number (e.g., 2k or more) of timestamps/clock edges, but unfortunately, it just takes too much processor effort to encode and decode.  It someone was writing a waveform *archiver*, it would be the way to go if one wanted to maximize compression.  FST provides a limited version of this at the declared signal level, not at the bit level with the compromise being for greater speed.

```
-rw-rw-r--.  1 bybell bybell 826094121 Feb 18 16:51 icu_tb.vcd
-rw-rw-r--.  1 bybell bybell  18368841 Feb 18 16:54 icu_tb.lxt
-rw-rw-r--.  1 bybell bybell   3927840 Feb 18 16:51 icu_tb.fst
-rw-rw-r--.  1 bybell bybell    469144 Feb 18 16:51 icu_tb.fst.xz
-rw-rw-r--.  1 bybell bybell    204033 Feb 18 16:52 icu_tb.vzt
```

The invocation for VZT for those interested:
`vcd2vzt -d9 -m256 -z2 -r icu_tb.vcd icu_tb.vzt
`

In practice, if one wants to save on disk, one is better off emitting to a loosely packed quick to write version like FST then run XZ/LZMA over it as a postprocessor, either at the file or block level.

wsnyder, 2025-02-18:
I agree having the dumper compute signal relationships is unlikely a win. 

FWIW, note for Verilator, it often computes a bit select, or inversion, or other (typically but not always) simple unary expression of one signal A from another B only for purposes of a dump, made cycle-by-cycle right when the dump A/B is made.  It doesn't even compute A if dumping is off. That is, if there was a way to express simple relationship A=B[3] in the format, it would be faster, as both the computation and dump of A is removed.

I would suggest limiting to simple unary operators.  I suspect select, invert and add alone cover probably 90% of the possible gains - I can look at some models to get data if interested.

tbybell, 2025-02-18:
Wilson, sure, if you want to, go ahead.  The funny thing is I never thought of implementing the invert for dynamic aliases, but that one totally makes sense, especially for netlist sim filled with buffers and inverters.  Its usefulness depends on the internal format of signal data in the dumper, so it might not be relevant with some implementations, but it's a good datapoint to have.

pieter3d, 2025-02-19:
Added [a slide](https://docs.google.com/presentation/d/1DPxEnFCLY2rNfUrllBde9QHfP5iaqk-KfkiIx7bLD5U/edit#slide=id.g3314c95446d_0_0) for project participants. If you'd like to be involved at the start, please add a comment (or reply here) with your preferred contact name/address, I'll add you.

ecstrema, 2025-02-19:
To be clear, I don't think post-processing of signals to see which ones are dependent on which is the way to go. I think it should be done at a much earlier point, before the simulation even started, allowing the simulator to completely disregard the calculation of these values until they are used. 

However, this is simulator land, and if they don't want to implement dependent-description, then any simulator can still dump all values. 

To get back to clock-specific optimization, I am now mostly convinced it is not worth it as LZ4 already does a great job at it (although not perfect, I'll send the benchmarks I did yesterday with FST when I'm back on my desk). I haven't tried with counters yet, but I expect it to be about the same.

ecstrema, 2025-02-20:
I did some testing: here are the results. 

TLDR: compression works well for the files, but opening the files means decompressing, and that just moves the burden to the application. I think that an optimized first compression pass can do better. I'd be happy to be proved wrong.

<details>
<summary>Methodology</summary>

```bash
ghdl -a --std=08 --workdir=ghdl $file
ghdl -e --std=08 --workdir=ghdl tb
ghdl -r --std=08 --workdir=ghdl tb --fst=ghdl/wave.fst
```
</details>

## Clocks

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use std.env.finish;

architecture test of tb is
  signal clk : std_logic := '0';
begin

  clk <= not clk after 1 ns / 2;

  process
  begin
    wait for 100 ms; -- This is the changing duration
    finish;
  end process;

end;
```

Results:
```

100ms: 4.56kb
10ms: 700bytes
1ms: 335bytes

We can see that compression is working here, else we'd have several Gigabytes of data (the equivalent vcd is 3.89GB). But it is far from perfect. With proper clock handling, these should all be 335 bytes or less. 

## Counters

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use std.env.finish;

architecture test of tb is
  signal clk : std_logic := '0';
  signal counter : unsigned(4 downto 0) := to_unsigned(0, 5);
begin

  clk <= not clk after 1 ns / 2;
  process(clk)
  begin
    if rising_edge(clk) then
        counter <= counter + 1;
    end if;
  end process;

  process
  begin
    wait for 1 ms;
    finish;
  end process;

end;
```

1ms: 450 bytes
10ms: 1.04 KB
100ms: 7.64 KB

Once again, compression works, but we can do way better.

## Concatenation

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use std.env.finish;

architecture test of tb is
  signal clk : std_logic := '0';
  signal clk_clk : std_logic_vector(1 downto 0);
begin

  clk <= not clk after 1 ns / 2;

  clk_clk <= clk & clk;

  process
  begin
    wait for 10 ms;
    finish;
  end process;

end;
```

1ms: 404 bytes
10ms: 1.04 KB
100ms: 7.91KB

This one is an example of something that is trivially computable at read time.

_However_, the important thing for those two examples is, I believe, not compression, but decompression. Trying to use the 100ms version of these files in surfer takes forever to open.

With a special clock type and counter type, for example, you don't have to uncompress any of these (this would give you back multiple GB of memory for the internal representation). You can trivially know what is the state of the signal at any instant.

ecstrema, 2025-02-20:
This discussion might want to be continued elsewhere. Would the mailing list be OK? Else, the new FST repo may be a good place if this is going to be FST2.

pieter3d, 2025-02-20:
Does the name OSDB (open signal database) seem more appropriate? That matches the slide deck too.
Please also add yourself to it if you want to join the working group with CHIPS Alliance

ecstrema, 2025-02-20:
You mean add yourself to the slide deck? I think you just added me. Or is there something else?

OSDB seems good to me, but if we are going to reuse a large part of FST, a new name might not be worth it?

Anyway, this is all debatable and can be chosen later. Thanks for building the slide deck though!

tbybell, 2025-02-20:
"Clock compression" works, otherwise it would not have been put in the original LXT.  Just get a vcd that is nothing more than a free running clock and run vcd2lxt on it with and without -clockpack.  Is it more effective than letting something like LZ4 chew on the data?  Sure.  An important design point not to be forgotten is that writer speed is very important.  The more heuristics one adds, the slower writing gets, and a lot of that slowdown comes from violating locality in the TLBs, etc. even if the code from an algorithm complexity point of view looks trivial.  

Note that you won't instantly get back your data in the viewer unless the viewer is retooled to handle compressed waves as an internal representation that is a first class citizen.  This gets to be a serious pain when searching for values at a given marker time.  It's doable, but it's a lot more involved because not only do you have time/value data, you have recurrence equations randomly buried among them.  This happens often with modern designs because free running clocks burn power, so some gated clocks have very sparse activity that appears almost random.

Sometimes the producer (such as the dumper for tester data) knows up front what is a free running clock.  Back in 2015, synthetic clocks were added to FST for one such user:
```
/*
 * decompress [m b xs xe valstring]... format string into trace
 */
static void expand_synvec(int txidx, const char *s)
```

You will make many of the same observations others did in the past, and that is good as it shows you're moving along the same path.

rfuest, 2025-02-20:
> This discussion might want to be continued elsewhere. Would the mailing list be OK? Else, the new FST repo may be a good place if this is going to be FST2.

I've enabled the `Discussions` tab in the libfst repo (https://github.com/gtkwave/libfst/discussions), if you want to use that for FST2 discussions.

tbybell, 2025-02-20:
Yep, that's the place for it.  This thread has completely gone off topic.