gtkwave/gtkwave #81: TCL Commands for View menus doesn't work

carlosedp, 2021-09-30:
I'm writing a TCL script to set some defaults but I noticed that using the commands below to set some View options doesn't do anything:

```
gtkwave::/View/Show_Filled_High_Values 1
gtkwave::/View/Show_Wave_Highlight 1
gtkwave::/View/Show_Mouseover 1
```

I tried all of them with and without the "1" argument with same problem.

I'm on GTKWave 3.3.107 installed thru Homebrew on MacOS Big Sur

gtkwave, 2021-09-30:
I have to look more but I think it's because of a race condition where those are toggle boxes so they're set up by the TCL script then later are reversed out through some later config.  I noticed they all work from the console if --wish is used as a command line argument.
I'm sure that's it because the following at the very beginning fixes it:

gtkwave::nop

...that forces us to go through the GTK main loop at least once and by then everything is set up.

-Tony

carlosedp, 2021-09-30:
I've temporarily set my gtkwaverc to have the params until it's possible to set it via TCL.

gtkwave, 2021-09-30:
You should be able to set it via TCL provided you put that nop in there before any of them.  It's a workaround that shouldn't affect things any though I'll look to see about a true fix.

-Tony

carlosedp, 2021-09-30:
Didn't work... used something like:

```tcl
set tmpdir $::env(PATH)
puts $tmpdir

set nsigs [ gtkwave::getNumFacs ]
set sigs [list]

# Beautification - View settings
gtkwave::nop
gtkwave::/Edit/Set_Trace_Max_Hier 0
gtkwave::/View/Show_Filled_High_Values 1
gtkwave::/View/Show_Wave_Highlight 1
gtkwave::/View/Show_Mouseover 1

# Add TOPLevel clock and reset signals
set top [list CPU.clock CPU.reset ]
gtkwave::/Edit/Insert_Comment "Clock & Reset"
```

Also tried without the "1" argument to the menus and tried putting the nop at the absolute top as first statement of the script.

gtkwave, 2021-09-30:
OK, there might be some kind of even deeper race condition as it worked for me on Centos7.  Thanks for the update.

-Tony

carlosedp, 2021-10-01:
BTW, this is the script I'm playing with: https://gist.github.com/carlosedp/97188d9b0749c9faf99f8a79df5a3db6

gtkwave, 2021-10-03:
I finally got to look at this in some depth.

OK, I see what's going on...for this what I did was I expected the GUI to toggle the checkmark and I simply read it and passed it on:        GLOBALS->highlight_wavewindow = gtk_check_menu_item_get_active(GTK_CHECK_MENU_ITEM(menu_wlist[WV_MENU_SHW]));

No GUI operation from TCL means no toggling of checkmark state.
I'll look at a fix for this.  Of course for TCL, we don't want to toggle, but want to set values directly.  I'll have to trap all the toggle menu items and set them manually with a 1/0 on/off, etc.

-Tony

gtkwave, 2021-10-24:
OK, I finally got to look at this.
Use the menu item name without an argument to toggle its current value, otherwise use a 1 or a 0 argument to turn it on or off.
-Tony

713:/home/bybell/git/gtkwave/gtkwave3-gtk3> git push origin master
Counting objects: 15, done.
Delta compression using up to 4 threads.
Compressing objects: 100% (8/8), done.
Writing objects: 100% (8/8), 3.77 KiB | 0 bytes/s, done.
Total 8 (delta 6), reused 0 (delta 0)
remote: Resolving deltas: 100% (6/6), completed with 6 local objects.
To https://github.com/gtkwave/gtkwave
   f1c8ff6..54f1d26  master -> master