gtkwave/gtkwave #91: Signedness of time types in LXT files

rfuest, 2021-10-29:
To test my refactored waveform display code I had to create some test files and noticed an issue.

`lxttime_t` in `lxt_write.h` is defined to be an unsigned integer, which is also mentioned in the LXT format chapter in the GTKWave docs. But because `TimeType` is signed all values with the high bit set will be wrapped to negative times. GTKWave doesn't crash when opening such a file, but `GLOBALS->max_time` becomes negative and the waveform display will be empty.

This is probably not a common problem for real files, but adding a check for this case would make the import more robust.

gtkwave, 2021-10-31:
That was a good observation.
I'll look into this...there's also a bit more than just that 63b signed integer overflow crossing such that all loaders likely are affected.  The good thing is making testcases are easy for finding where/when things go wrong.

For real-world stuff, I'm not too concerned with sim crossing into values that large because it could be argued, what about 64b, or 65b, or 66b?  But yes, it's not good for something to break silently.

Back in the old days of GTK 1.2, I vaguely remember scrollbars or other GTK internals running on gfloats, and the 24b mantissa caused issues.  I wouldn't be surprised if somewhere around the 55b or 56b time value mark that some portions of the GUI start breaking down with respect to reliability of function.  For example, click on a scrollbar to advance on "page" and it does nothing at very large time values, at certain zoom values on top of that, etc.

-Tony

rfuest, 2021-10-31:
The limited resolution of doubles for time values is exactly what lead me to this discovery. My refactored widget code used doubles for time value in some places and this caused problems with large times. I'm currently trying to fix this by using as much 64 bit integer math as possible.

gtkwave, 2021-10-31:
After a certain point, 53 bits might be a hard limit (i.e., 52b + the implied leading '1' on mantissa for IEEE doubles).I don't see it as a real restriction if we cap sim times to fit in that range.  For example:

"GTKWAVE | Time stamp values greater than (2**53)-1 are unsupported by the GUI, exiting."
The largest sim runs I have done debug on for my day job that have been practical/usable (from a how long you want to wait perspective) have been in the 10M cycle range (about 24 hours of sim in VCS), and even if using real psec values instead of cycle count = time, the numbers still don't overflow outside of a useful range that the GUI can deal with.  Those who need the accuracy and are running analog stuff can use a proper analog viewer that isn't geared for digital sim.  Huge numbers are outside the range of most usage cases except a handful of special ones that require large amounts of precision.  This isn't the tool for those cases.

-Tony

rfuest, 2021-11-03:
OK, limiting the time values to fit in the mantissa of a double is a good simple solution.

I did briefly try to use more integer math inside the widget and I guess that it would have worked, but it made the code much more complicated. Keeping the code more readable and maintainable should be more important than supporting a niche use case.

Should the limited time range become an issue in the future one option would be to switch to a 128 bit float for widget internal calculations. That should work without major changes to the rest of the code, but I'm not sure if they are supported on all platforms.

gtkwave, 2021-11-03:
I'll look at limiting time time values in the near future as that seems to be the cleanest solution.

Note that 128 bit floats probably are supported on everything gcc and Clang supports these days.  I've never tried it though.  Legacy 80-bit floats exist on x86, I think through using "long double" with some compilers but I'm not sure.
514:/tmp/e> ./t
long double size is: 16
-Tony