gtkwave/gtkwave #95: How to update MSYS2 package ?

NicoPy, 2021-11-07:
Hi,

Are ` mingw-w64-x86_64-gtkwave-gtk2-ci-1-any.pkg.tar.zst ` and ` mingw-w64-x86_64-gtkwave-gtk3-ci-1-any.pkg.tar.zst` MSYS2 packages ?

I tried to use them to update my MSYS2 installation but got errors :

With ` mingw-w64-x86_64-gtkwave-gtk2-ci-1-any.pkg.tar.zst `  package :
```bash
Nicolas@Nicolas-PC MINGW64 ~
$ wget -O mingw-w64-x86_64-gtkwave-gtk2-ci-1-any.pkg.tar.zst https://github.com/gtkwave/gtkwave/releases/download/nightly/mingw-w64-x86_64-gtkwave-gtk2-ci-1-any.pkg.tar.zst
--2021-11-07 18:42:30--  https://github.com/gtkwave/gtkwave/releases/download/nightly/mingw-w64-x86_64-gtkwave-gtk2-ci-1-any.pkg.tar.zst
Resolving github.com (github.com)... 140.82.121.4
Connecting to github.com (github.com)|140.82.121.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github-releases.githubusercontent.com/222023386/e763746d-8bf8-45d2-8fee-1061fe23f531?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20211107%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20211107T174231Z&X-Amz-Expires=300&X-Amz-Signature=6ced4e9fed54fa089617013ec784dd1fef0dc23290ec3e9b3a10fafd7756fa16&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=222023386&response-content-disposition=attachment%3B%20filename%3Dmingw-w64-x86_64-gtkwave-gtk2-ci-1-any.pkg.tar.zst&response-content-type=application%2Foctet-stream [following]
--2021-11-07 18:42:30--  https://github-releases.githubusercontent.com/222023386/e763746d-8bf8-45d2-8fee-1061fe23f531?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20211107%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20211107T174231Z&X-Amz-Expires=300&X-Amz-Signature=6ced4e9fed54fa089617013ec784dd1fef0dc23290ec3e9b3a10fafd7756fa16&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=222023386&response-content-disposition=attachment%3B%20filename%3Dmingw-w64-x86_64-gtkwave-gtk2-ci-1-any.pkg.tar.zst&response-content-type=application%2Foctet-stream
Resolving github-releases.githubusercontent.com (github-releases.githubusercontent.com)... 2606:50c0:8003::154, 2606:50c0:8001::154, 2606:50c0:8000::154, ...
Connecting to github-releases.githubusercontent.com (github-releases.githubusercontent.com)|2606:50c0:8003::154|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2476157 (2.4M) [application/octet-stream]
Saving to: ‘mingw-w64-x86_64-gtkwave-gtk2-ci-1-any.pkg.tar.zst’

mingw-w64-x86_64-gtkwave-gtk2-ci-1-any.pkg.tar.zst                  100%[==================================================================================================================================================================>]   2.36M  11.1MB/s    in 0.2s

2021-11-07 18:42:31 (11.1 MB/s) - ‘mingw-w64-x86_64-gtkwave-gtk2-ci-1-any.pkg.tar.zst’ saved [2476157/2476157]

Nicolas@Nicolas-PC MINGW64 ~
$ pacman -U mingw-w64-x86_64-gtkwave-gtk2-ci-1-any.pkg.tar.zst
loading packages...
resolving dependencies...
looking for conflicting packages...

Packages (1) mingw-w64-x86_64-gtkwave-gtk2-ci-1

Total Installed Size:  4.90 MiB

:: Proceed with installation? [Y/n] y
(1/1) checking keys in keyring                                                                                                                                     [####################################################################################################] 100%
(1/1) checking package integrity                                                                                                                                   [####################################################################################################] 100%
(1/1) loading package files                                                                                                                                        [####################################################################################################] 100%
(1/1) checking for file conflicts                                                                                                                                  [####################################################################################################] 100%
error: failed to commit transaction (conflicting files)
mingw-w64-x86_64-gtkwave-gtk2: /mingw64/bin/evcd2vcd.exe exists in filesystem (owned by mingw-w64-x86_64-gtkwave)
mingw-w64-x86_64-gtkwave-gtk2: /mingw64/bin/fst2vcd.exe exists in filesystem (owned by mingw-w64-x86_64-gtkwave)
mingw-w64-x86_64-gtkwave-gtk2: /mingw64/bin/fstminer.exe exists in filesystem (owned by mingw-w64-x86_64-gtkwave)
mingw-w64-x86_64-gtkwave-gtk2: /mingw64/bin/gtkwave.exe exists in filesystem (owned by mingw-w64-x86_64-gtkwave)
mingw-w64-x86_64-gtkwave-gtk2: /mingw64/bin/lxt2miner.exe exists in filesystem (owned by mingw-w64-x86_64-gtkwave)
mingw-w64-x86_64-gtkwave-gtk2: /mingw64/bin/lxt2vcd.exe exists in filesystem (owned by mingw-w64-x86_64-gtkwave)
mingw-w64-x86_64-gtkwave-gtk2: /mingw64/bin/rtlbrowse.exe exists in filesystem (owned by mingw-w64-x86_64-gtkwave)
mingw-w64-x86_64-gtkwave-gtk2: /mingw64/bin/shmidcat.exe exists in filesystem (owned by mingw-w64-x86_64-gtkwave)
mingw-w64-x86_64-gtkwave-gtk2: /mingw64/bin/twinwave.exe exists in filesystem (owned by mingw-w64-x86_64-gtkwave)
mingw-w64-x86_64-gtkwave-gtk2: /mingw64/bin/vcd2fst.exe exists in filesystem (owned by mingw-w64-x86_64-gtkwave)
mingw-w64-x86_64-gtkwave-gtk2: /mingw64/bin/vcd2lxt.exe exists in filesystem (owned by mingw-w64-x86_64-gtkwave)
mingw-w64-x86_64-gtkwave-gtk2: /mingw64/bin/vcd2lxt2.exe exists in filesystem (owned by mingw-w64-x86_64-gtkwave)
mingw-w64-x86_64-gtkwave-gtk2: /mingw64/bin/vcd2vzt.exe exists in filesystem (owned by mingw-w64-x86_64-gtkwave)
mingw-w64-x86_64-gtkwave-gtk2: /mingw64/bin/vzt2vcd.exe exists in filesystem (owned by mingw-w64-x86_64-gtkwave)
mingw-w64-x86_64-gtkwave-gtk2: /mingw64/bin/vztminer.exe exists in filesystem (owned by mingw-w64-x86_64-gtkwave)
mingw-w64-x86_64-gtkwave-gtk2: /mingw64/bin/xml2stems.exe exists in filesystem (owned by mingw-w64-x86_64-gtkwave)
mingw-w64-x86_64-gtkwave-gtk2: /mingw64/share/applications/gtkwave.desktop exists in filesystem (owned by mingw-w64-x86_64-gtkwave)
mingw-w64-x86_64-gtkwave-gtk2: /mingw64/share/icons/gnome/16x16/mimetypes/gnome-mime-application-vnd.gtkwave-ae2.png exists in filesystem (owned by mingw-w64-x86_64-gtkwave)
mingw-w64-x86_64-gtkwave-gtk2: /mingw64/share/icons/gnome/16x16/mimetypes/gnome-mime-application-vnd.gtkwave-aet.png exists in filesystem (owned by mingw-w64-x86_64-gtkwave)
mingw-w64-x86_64-gtkwave-gtk2: /mingw64/share/icons/gnome/16x16/mimetypes/gnome-mime-application-vnd.gtkwave-evcd.png exists in filesystem (owned by mingw-w64-x86_64-gtkwave)
mingw-w64-x86_64-gtkwave-gtk2: /mingw64/share/icons/gnome/16x16/mimetypes/gnome-mime-application-vnd.gtkwave-fst.png exists in filesystem (owned by mingw-w64-x86_64-gtkwave)
mingw-w64-x86_64-gtkwave-gtk2: /mingw64/share/icons/gnome/16x16/mimetypes/gnome-mime-application-vnd.gtkwave-ghw.png exists in filesystem (owned by mingw-w64-x86_64-gtkwave)
mingw-w64-x86_64-gtkwave-gtk2: /mingw64/share/icons/gnome/16x16/mimetypes/gnome-mime-application-vnd.gtkwave-gtkw.png exists in filesystem (owned by mingw-w64-x86_64-gtkwave)
mingw-w64-x86_64-gtkwave-gtk2: /mingw64/share/icons/gnome/16x16/mimetypes/gnome-mime-application-vnd.gtkwave-lx2.png exists in filesystem (owned by mingw-w64-x86_64-gtkwave)
mingw-w64-x86_64-gtkwave-gtk2: /mingw64/share/icons/gnome/16x16/mimetypes/gnome-mime-application-vnd.gtkwave-lxt.png exists in filesystem (owned by mingw-w64-x86_64-gtkwave)
mingw-w64-x86_64-gtkwave-gtk2: /mingw64/share/icons/gnome/16x16/mimetypes/gnome-mime-application-vnd.gtkwave-lxt2.png exists in filesystem (owned by mingw-w64-x86_64-gtkwave)
mingw-w64-x86_64-gtkwave-gtk2: /mingw64/share/icons/gnome/16x16/mimetypes/gnome-mime-application-vnd.gtkwave-vcd.png exists in filesystem (owned by mingw-w64-x86_64-gtkwave)
mingw-w64-x86_64-gtkwave-gtk2: /mingw64/share/icons/gnome/16x16/mimetypes/gnome-mime-application-vnd.gtkwave-vzt.png exists in filesystem (owned by mingw-w64-x86_64-gtkwave)
[... 37 lines cut ...]
mingw-w64-x86_64-gtkwave-gtk2: /mingw64/share/man/man1/twinwave.1.gz exists in filesystem (owned by mingw-w64-x86_64-gtkwave)
mingw-w64-x86_64-gtkwave-gtk2: /mingw64/share/man/man1/vcd2fst.1.gz exists in filesystem (owned by mingw-w64-x86_64-gtkwave)
mingw-w64-x86_64-gtkwave-gtk2: /mingw64/share/man/man1/vcd2lxt.1.gz exists in filesystem (owned by mingw-w64-x86_64-gtkwave)
[... cut ...]
mingw-w64-x86_64-gtkwave-gtk2: /mingw64/share/mime/packages/x-gtkwave-extension-lxt2.xml exists in filesystem (owned by mingw-w64-x86_64-gtkwave)
mingw-w64-x86_64-gtkwave-gtk2: /mingw64/share/mime/packages/x-gtkwave-extension-vcd.xml exists in filesystem (owned by mingw-w64-x86_64-gtkwave)
mingw-w64-x86_64-gtkwave-gtk2: /mingw64/share/mime/packages/x-gtkwave-extension-vzt.xml exists in filesystem (owned by mingw-w64-x86_64-gtkwave)
Errors occurred, no packages were upgraded.

``` 

Same problem with ` mingw-w64-x86_64-gtkwave-gtk3-ci-1-any.pkg.tar.zst` package

umarcor, 2021-11-07:
It seems that you installed `mingw-w64-x86_64-gtkwave` (from MSYS2 repos) before trying to install nightly packages from this repo. Uninstall the other one first.

NicoPy, 2021-11-07:
> 
> 
> It seems that you installed `mingw-w64-x86_64-gtkwave` (from MSYS2 repos) before trying to install nightly packages from this repo.

Yes, that's what I did (I did the same for GHDL without any problem). I'll remove it and try again.
I'll let you know how it goes.

umarcor, 2021-11-08:
> > It seems that you installed `mingw-w64-x86_64-gtkwave` (from MSYS2 repos) before trying to install nightly packages from this repo.
> 
> Yes, that's what I did (I did the same for GHDL without any problem).

I think that's because we use the `provides` field in nightly GHDL recipes (see https://github.com/ghdl/ghdl/blob/fbd853c88b4b1c3008b1b39882ac2b99b6e59460/scripts/msys2-llvm/PKGBUILD#L11), but we don't use it in nightly GTKWave recipes (see https://github.com/gtkwave/gtkwave/blob/master/MSYS2/gtk2/PKGBUILD).

NicoPy, 2021-11-08:
> I'll let you know how it goes.

Well... there's no more 64-bit nightly package. So, I'll wait for  them to come back ;)

umarcor, 2021-11-08:
Oops... It seems that precisely the update step of the last CI run failed: https://github.com/gtkwave/gtkwave/actions. Unfortunately, I cannot restart CI in this repo. Let's wait until the next push :wink:

rfuest, 2021-11-09:
> Let's wait until the next push wink

Another option is to directly download the artifacts from the build page, because normally only the creation of the new pre-release fails. For example, navigate to https://github.com/gtkwave/gtkwave/actions/runs/1435703695 and download the 125 MB file with all builds in the "Artifacts" section.

There was a scheduled build in the meantime, which did succeed, so the nightly builds are back in the pre-release. But maybe this helps in the future.

umarcor, 2021-11-09:
@rfuest that's correct! I completely forgot about it. Thanks!

NicoPy, 2021-11-09:
I've just tried both Gtk2 and Gtk3 versions (64-bits).
It works !
Thanks for your help.

Which version do you recommend ? Gtk2 or Gtk3 ?

gtkwave, 2021-11-10:
Moving forward, the -gtk3 branch will be the only one getting updated, so if there aren't perf issues on you machine, I'd say go with that.
I'll continue to fold in GHW loader updates and such to the original gtk2 version to maintain file compatibility, but I don't plan on doing dual feature update like I have for the past 4 years or so since GTK4 will be an issue at some point.
-Tony

closed 2021-11-10 by NicoPy