gtkwave/gtkwave #98: BitsToReal not working for 32-bit signal

YingXiangLi10, 2021-11-13:
BitsToReal seems to work fine for 32-bit signal according to the help text.

I have a 32-bit reg type signal, when i select "BitsToReal" option in data format, it only shows the binary of the signal.

gtkwave, 2021-11-13:
What version, OS, etc?  I just tried it and it seemed to work for me.  Also, if you have a VCD you can share, that would help too.
-Tony

YingXiangLi10, 2021-11-17:
Sorry for not replying for a long time.

YingXiangLi10, 2021-11-17:
Sorry for not replying for so long. The problem is solved when I use the latest version. The problem was happened in the version 3 years ago, back to  3.3.9x, which can be install with apt-get. 

Thanks for this wonderful wave viewer, it's fast and super helpful.

closed 2021-11-17 by YingXiangLi10