gtkwave/gtkwave #100: dconf-WARNING failed to commit changes to dconf

jiangtianxue, 2021-12-06:
when gtkwave my vcd file I enconter this warning, does it matter?

(gtkwave:1459337): dconf-WARNING **:  failed to commit changes to dconf:failed to commit changes to dconf: Error spawning command line ?dbus-launch --autolaunch=20200914172502291423539425701795 --binary-syntax --close-stderr?: Child process exited with code 1

jiangtianxue, 2021-12-06:
[image]

gtkwave, 2021-12-06:
It sounds like a system-level problem with dconf/dbus has issues.  Something inside glib function calls for gconf or something similar.
It's impossible to debug without more information.

-Tony

Rohlik, 2022-09-12:
Not related to gtkwave probably, but I got similar error:

```
test@localhost:~$ DISPLAY=:0 gnome-shell-extension-tool -e disable-gestures@mattbell.com.au
'disable-gestures@mattbell.com.au' is now enabled.

(process:6369): dconf-WARNING **: 20:02:50.557: failed to commit changes to dconf: Error spawning command line “dbus-launch --autolaunch=b1d87f638352486683a1bb07a6b45c7a --binary-syntax --close-stderr”: Child process exited with code 1
```