gtkwave/gtkwave #103: [+] and [-] in populateBuffer

rfuest, 2021-12-07:
`populateBuffer` contains some code to handle `[+]` and `[-]` prefixes but I couldn't find any instances where traces with these prefixes are created. Is this feature still used?

gtkwave, 2021-12-07:
I believe it's for BlueSpec-specific usage:
GitHub - B-Lang-org/bdw: BSC Development Workstation (BDW)
./testing/bsc.bdw/commands/1v.NovasRC.typed.dump.expected:userCompositeMem /main/top/iv/data3 [0] [1] [+] [Vector#(2, Bool)]./testing/bsc.bdw/commands/1v.NovasRC.typed.dump.expected:userCompositeMem /main/top/iv/data3 [0] [0] [+] [Vector#(2, Bool)]

-Tony

rfuest, 2021-12-08:
Thanks for the link. So this is just a visual indicator that these traces can be expanded and requires an alias in the savefile: https://github.com/B-Lang-org/bdw/blob/main/testing/bsc.bdw/commands/1v.GtkWaveScript.bits.dump.expected#L863-L869
I'll add it to my growing list of tests.

closed 2021-12-08 by rfuest