gtkwave/gtkwave #105: ghw files signals tree

NicoPy, 2021-12-19:
I'm not sure this is the right place to ask since the subject is related to GHDL GHW plugin.
My question is : Why are vectors not located at the same level in the signals tree ?
In the following example :
[image]
`clk`, `data1` and `data2` signals are all declared at the same place in the VHDL file.
Why are `data1`and `data2` not located under `clk` in the signal list ?
[image]
I find it's difficult to find the signals I want to view in the tree with this representation.

gtkwave, 2021-12-19:
It's a side-effect of how GHW dumps signals.  Every mullti-bit signal is bitblasted.  I didn't write the ghw.c loader, but the appropriate coalesce routines would need to be there.I don't have any GHW files handy at the moment so I don't remember if there is an extra level of hierarchy for those multi-bit signals.  (For the above, I'm using the word "signal" loosely, and not with a strictly VHDL-ish meaning.)

-Tony

NicoPy, 2021-12-20:
Here are the files I used (included the ghw file) to create this issue : [Reg.zip](https://github.com/gtkwave/gtkwave/files/7744809/Reg.zip)

gtkwave, 2021-12-20:
Thanks.  I'll look to see if there's any way to pull back signal generation for vectors into the same level of hierarchy.
-Tony

NicoPy, 2021-12-20:
Thanks for your interest in this issue.