gtkwave/gtkwave #106: Request: no lines for blanks in black&white mode

jotego, 2021-12-21:
The black & white mode is very convenient because of its high contrast. But as soon as I plot a few signals, it gets confusing. Using blanks to space signals out doesn't help because blanks are represented as a constant low signal, so it just adds more clutter.

Would it be possible **to make blanks blank**? This also happens in colour mode but because of the dark blue used is not as bad.

[image]

rfuest, 2021-12-21:
I'm currently working on rewriting the waveform widget and could include this in my PR. With the my changes it would be easy to use the print colors for the black and white mode:
[image]
Maybe the grid lines could be made a bit darker to improve visibility.

jotego, 2021-12-21:
I see you're drawing the blank rows as light gray lines. That's nice too.

gtkwave, 2021-12-21:
Use View->Show Grid toggled to off.
-Tony

jotego, 2021-12-22:
It was so simple it embarrases me :-)
Thank you

closed 2021-12-22 by jotego

rfuest, 2021-12-23:
@gtkwave What is you opinion on changing the black and white mode to use some grayscale colors? Or do you think that both, a real black and white and a grayscale mode should exist? The current output of the black and white mode has some issues and hides some information, e.g. `U`, `W`, `X` and `-` values are all rendered as a black box without a label.

gtkwave, 2021-12-24:
Grayscale is fine.  Its sole intent was for much higher contrast than the usual color traces.
-Tony