gtkwave/gtkwave #107: [enhancement] default zoom level on startup

goekce, 2021-12-22:
I recently started using gtkwave and I am starting to like it.

Gtkwave starts with the default zoom level at femtoseconds. In my opinion most of the designs the people work with do not work at the fs level.

We could implement a default zoom level for the configuration file like `fs`, `ns`, `us`, where the tool would start at the respective zoom level. This setting would be also very useful for the savefile. Alternatively, as a basic workaround `do_initial_zoom_fit` could be activated as default (currently I use this. Unfortunately it took me about three hours to find the useful setting `do_initial_zoom_fit` which saves me from pressing `Ctrl+0` every time. ).

What do you think about these ideas?

I also found [a similar request from 2016](https://sourceforge.net/p/gtkwave/mailman/message/35195198/).

goekce, 2021-12-22:
Update:
I also created a wavefile in ghdl using the option [`--time-resolution=ns`](https://ghdl.github.io/ghdl/using/InvokingGHDL.html?highlight=time%20resolution#cmdoption-ghdl-time-resolution). Gtkwave still opens the wavefile at `fs` level.

gtkwave, 2021-12-22:
That's something specific to the GHW loader as Verilog doesn't see this problem:
in ghw.c:
 GLOBALS->time_scale = 1;
 GLOBALS->time_dimension = 'f';

It doesn't matter what you specify in sim.  GHW apparently writes all waves out at the fs resolution.

-Tony

goekce, 2021-12-23:
Oh, good to know that Verilog doesn't see this problem.

AFAIK only the mcode version of ghdl supports time resolution so maybe that is the reason why GHW files use fs as default. It seems to me that this cannot be fixed in the short-run.

Does @tgingold have a comment on this?

tgingold, 2021-12-23:
I think the ghw file now contains the resolution.  So ghw.c has to read it to set the time scale.

gtkwave, 2021-12-23:
I'd need to know what's in libghw to do that or if that's a new(er) feature missing from the reader library source in gtkwave.

-Tony