gtkwave/gtkwave #108: [enhancement] setting current working dir for relative `[dumpfile]` path

goekce, 2021-12-22:
In my working flow I start gtkwave using my window manager. Then I open a savefile which loads a *relative* dumpfile like this:
```
[dumpfile] "wave.ghw"
...
```
On my Linux system this leads to the following error:
```
...
GTKWAVE | Could not initialize 'wave.ghw' found in '/home/... ...proj/wave.gtkw', exiting.
```
Probably gtkwave starts in my home folder and cannot convert `"wave.ghw"` to an absolute path. If I start gtkwave manually in my project folder, then everything seems to be fine.

It would be great if gtkwave would search for `wave.ghw` in the same folder as the savefile. Alternatively the savefile could support changing the current working directory.

What do you think about idea?

gtkwave, 2021-12-22:
If these are save files you're generating by hand, try putting a dummy path like /tmp/ in front of them.  Relative loading vs the save file should work--at least I thought it did when I tested it (given that gtkwave writes out canonicalized filenames with full path).

-Tony

goekce, 2021-12-23:
I cannot fully understand what you suggest dear Tony :(

I tried the following which did not work:
```
[dumpfile] "/tmp/wave.ghw"
...
```
outputs:
```
...
GTKWAVE | Could not initialize '/tmp/wave.ghw' found in '/home/... .../wave.gtkw', exiting.
```

> Relative loading vs the save file should work--at least I thought it did when I tested it (given that gtkwave writes out canonicalized filenames with full path).

You are right, if I save the savefiles using gtkwave, everything is ok. The reason is that gtkwave uses absolute paths instead of relative which does not fit the bill for me.