gtkwave/gtkwave #109: RFE: Method for an external process to move the cursor/add marker/run TCL commands on an already running GTKWave process

tomverbeure, 2021-12-27:
I have a work-in-progress tool that reads in an FST file, extracts the program counter and memory transactions, and then acts as a gdb server. This allows me to debug a simulated CPU with GDB after the fact: set breakpoints, inspect memory locations etc.

What I'd like to add is to link the gdb server to GTKWave, so that the server can move the waveform viewer to the time location where, say, a breakpoint was triggered. In practice, I think this means that GTKWave would have to create a server socket through which it can accept TCL commands.

Is there already some kind of mechanism in GTKWave that would allow me to do something like this?

Thanks!
Tom

gtkwave, 2021-12-27:
There's no explicit mechanism to have a server socket accept Tcl commands.  I would suppose it's possible with a Tcl script running that listened to a socket and sent what was sent to it through something like "eval" or whatever.  I don't know if it would require usage of the repscript (-R/-P) functionality.  My guess is it might.

-Tony

tomverbeure, 2021-12-27:
I wasn't aware of the existence of the repscript option. It'd obviously be a dirty hack, but it might just be the way to get to where I want.

Thanks!
Tom

closed 2021-12-27 by tomverbeure