gtkwave/gtkwave #112: VCD Incompatibility with IVerilog

ZipCPU, 2022-01-18:
Icarus Verilog produces scope names containing a $ as their first character.

When GTKWave processes a VCD file, it [labels all tokens that do not begin with a $ as strings](https://github.com/gtkwave/gtkwave/blob/c1fb33419129144eb1b145c23c74dcbf18503731/gtkwave3-gtk3/src/vcd.c#L404-L418).  Then, when GTKWave processes these scope commands, it [assumes that any non-string token must be an $end or other token--rather than a scope name that it doesn't understand](https://github.com/gtkwave/gtkwave/blob/c1fb33419129144eb1b145c23c74dcbf18503731/gtkwave3-gtk3/src/vcd.c#L1186).  As a result, it doesn't allocate any scope for this scope with an unrecognized name.

Later, when GTKWave processes the corresponding upscope VCD Token (for the scope it didn't recognize), it [pops a scope off of its stack--immediately becoming out of sync with the IVerilog generated VCD](https://github.com/gtkwave/gtkwave/blob/c1fb33419129144eb1b145c23c74dcbf18503731/gtkwave3-gtk3/src/vcd.c#L1214-L1244).  The result is all kinds of other tokens that should be in the hierarchy but ... are not.

I'd submit a solution here (shouldn't be too hard to create) except ... I can't seem to get GTKWave to build on my machine for whatever reason.  (First it was a lack of a configure script, then when I managed to edit the configure.ac enough to generate a configure script I didn't have gtk+-2.0 installed--even though I have gtk+-3.0 installed, etc.)

As a temporary fix, I can run `sed -e 's/scope begin $unm/scope begin unm/' < dump.vcd | sed -e 's/scope begin $ivl/scope begin ivl/' > newdump.vcd` to get around this bug.  If I do this, the hierarchy is back in place as it should be.

Dan

gtkwave, 2022-01-18:
I take it this is this how it deals with unnamed blocks of hierarchy that contain signal declarations?
If iverilog would output like VCS does, this would work.  Here's an example:

$scope begin unnamed$$_0 $end

...unfortunately, I don't have the time to fix this right now.

-Tony

rfuest, 2022-01-18:
> I'd submit a solution here (shouldn't be too hard to create) except ... I can't seem to get GTKWave to build on my machine for whatever reason. (First it was a lack of a configure script, then when [...]

Are you using a Linux system or another OS? The documentation hasn't been updated since the autotools generated files were removed from the GIT repository. These steps should be everything that is required to build the GIT version:
```
git clone https://github.com/gtkwave/gtkwave.git
cd gtkwave3-gtk3
./autogen.sh
./configure --enable-gtk3
make
```

ZipCPU, 2022-01-18:
I'm running Ubuntu 20.04.3.  When following your instructions, I get the following relevant lines from configure:

> checking for GTK... no
> configure: error: Package requirements (gtk+-2.0 >= 2.24.9) were not met:
>
> No package 'gtk+-2.0' found
>
> Consider adjusting the PKG_CONFIG_PATH environment variable if you
> installed software in a non-standard prefix.
...

I do not have GTK 2.0 installed.  I have GTK3.0 installed.  This seems to be the problem.

Dan

rfuest, 2022-01-18:
> I do not have GTK 2.0 installed. I have GTK3.0 installed. This seems to be the problem.

That's strange, the configure script shouldn't check for GTK2 if you pass the `--enable-gtk3` argument.

ZipCPU, 2022-01-18:
Hmm ... okay, didn't know about that argument.  Let me try it.

ZipCPU, 2022-01-18:
Okay, with that argument GTKWave now builds on my machine.  Let me see about posting a pull request with an update you can use.

ZipCPU, 2022-01-18:
Here are [my proposed changes](https://github.com/ZipCPU/GTKWave/commit/8811684884ad7fb6180dbf48e241bc7b3487b0e6).  I made a [second commit to get the non-GTK3 version](https://github.com/ZipCPU/GTKWave/commit/effe5c0645baa75d6abd431cacfcd10e3bc839a6).  I don't have permission to create a pull request, so we'll have to let those be my recommendations.

rfuest, 2022-01-19:
> I don't have permission to create a pull request, so we'll have to let those be my recommendations.

You don't need any special permissions to create pull requests. But you need to create a fork using the GitHub web interface or the GitHub CLI tool. If you push your changes to a new repository GitHub doesn't know that the repository is related to the upstream repository.

ZipCPU, 2022-01-19:
Well that would explain my problem then.

gtkwave, 2022-01-22:
> Counting objects: 45, done.
> Delta compression using up to 4 threads.
> Compressing objects: 100% (24/24), done.
> Writing objects: 100% (24/24), 2.71 KiB | 0 bytes/s, done.
> Total 24 (delta 23), reused 0 (delta 0)
> remote: Resolving deltas: 100% (23/23), completed with 20 local objects.
> To https://github.com/gtkwave/gtkwave
>    c1fb334..9fe9d7e  master -> master

I integrated your proposed changes.
-Tony

609:/home/bybell/git/gtkwave> git push origin masterUsername for 'https://github.com': gtkwave

ZipCPU, 2022-01-22:
Awesome!

Thank you!

Dan

closed 2022-01-22 by ZipCPU