gtkwave/gtkwave #116: gtkwave gui not updating changes made when clicking on icons or inserting signals

ivanm451, 2022-01-30:
Hi all, 

This is my first time opening an issue on git so pardon me if I miss some things. I compiled icarus Verilog for the M1 Mac and also brew installed gtkwave. I was running a simple Verilog script and was hoping to see the waveforms using gtkwave but when I open the .vcd file and try to access the hierarchy tree it doesn't open unless I click the SST tab, close the side menu and then reopen. The same thing for insert signals to be viewed, and even for viewing waveforms and moving across the horizontal axis; it all requires clicking on SST tab to close side menu and clicking on it again to reopen the side menu. I saw a previous ticket being raised for a similar issue on windows but was curious if there is any support for this on Mac. Btw I'm running macOS Monterey. Also, I get this output on the terminal when I click on anything in the gui: 

`
(gtkwave:31443): Gdk-WARNING **: 15:15:21.819: Source with depth 1 unsupported

`

At start up I also get the following messages when I start gtkwave: 

`(gtkwave:31443): GLib-GObject-WARNING **: 15:14:59.427: invalid cast from 'GtkMenuBar' to 'GtkWindow'

(gtkwave:31443): Gtk-CRITICAL **: 15:14:59.428: gtk_window_add_accel_group: assertion 'GTK_IS_WINDOW (window)' failed
`
I appreciate any and all help!

gtkwave, 2022-02-01:
Unfortunately, I don't have a modern Mac setup so I can't look into this.  However, I'll have to look to see if recent versions of gtk have broken things because this might not be a platform-specific issue.  So far I haven't seen any problems on the most recent Fedora.
What version of gtk are you using?

-Tony

ivanm451, 2022-02-01:
> Unfortunately, I don't have a modern Mac setup so I can't look into this.  However, I'll have to look to see if recent versions of gtk have broken things because this might not be a [...]

I'm using v3.3.104. Is this the latest?

gtkwave, 2022-02-01:
I'm interested in the version of the GTK toolkit, not gtkwave.
-Tony

ivanm451, 2022-02-01:
Oh my bad. The gtk version is 2.24.33

gtkwave, 2022-02-01:
I just tried running 2.24.33 (using LD_LIBRARY_PATH tricks verified by ldd) on Linux/X11 and didn't see any problems.  It sounds either rendering back end or platform-specfic.

-Tony

DanielO, 2024-02-06:
This is mac specific because in the GTK source (quartz/gdkdrawable-quartz.c):
```
  /* Handle drawable and pixmap sources. */
  if (src_depth == 1)
    {
      /* FIXME: src depth 1 is not supported yet */
      g_warning ("Source with depth 1 unsupported");
    }
  else if (dest_depth != 0 && src_depth == dest_depth)
    {
```

If I run gtkwave under lldb with G_DEBUG=fatal-warnings then I get this trace:
```
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BREAKPOINT (code=EXC_I386_BPT, subcode=0x0)
  * frame #0: 0x000000010140f2ff libglib-2.0.0.dylib`_g_log_abort + 22
    frame #1: 0x000000010140ee34 libglib-2.0.0.dylib`g_logv + 979
    frame #2: 0x000000010140e9f7 libglib-2.0.0.dylib`g_log + 120
    frame #3: 0x000000010074db66 libgdk-quartz-2.0.0.dylib`gdk_draw_drawable + 447
    frame #4: 0x0000000100751d44 libgdk-quartz-2.0.0.dylib`_gdk_gc_add_drawable_clip + 394
    frame #5: 0x000000010076534d libgdk-quartz-2.0.0.dylib`start_draw_helper + 309
    frame #6: 0x000000010076c711 libgdk-quartz-2.0.0.dylib`gdk_window_draw_drawable + 91
    frame #7: 0x000000010074db66 libgdk-quartz-2.0.0.dylib`gdk_draw_drawable + 447
    frame #8: 0x0000000100cdca1c libgtk-quartz-2.0.0.dylib`draw_cell_pixmap + 236
    frame #9: 0x0000000100cda51a libgtk-quartz-2.0.0.dylib`draw_row + 6170
    frame #10: 0x0000000100cc6cd5 libgtk-quartz-2.0.0.dylib`draw_rows + 278
    frame #11: 0x0000000100ccb640 libgtk-quartz-2.0.0.dylib`gtk_clist_expose + 204
    frame #12: 0x0000000100baa941 libgtk-quartz-2.0.0.dylib`_gtk_marshal_BOOLEAN__BOXED + 100
    frame #13: 0x0000000100a14404 libgobject-2.0.0.dylib`g_closure_invoke + 201
    frame #14: 0x0000000100a2a34c libgobject-2.0.0.dylib`signal_emit_unlocked_R + 1943
    frame #15: 0x0000000100a28e11 libgobject-2.0.0.dylib`signal_emit_valist_unlocked + 1770
    frame #16: 0x0000000100a28706 libgobject-2.0.0.dylib`g_signal_emit_valist + 52
    frame #17: 0x0000000100a29630 libgobject-2.0.0.dylib`g_signal_emit + 120
    frame #18: 0x0000000100ca2606 libgtk-quartz-2.0.0.dylib`gtk_widget_event_internal + 559
    frame #19: 0x0000000100ba8a14 libgtk-quartz-2.0.0.dylib`gtk_main_do_event + 1083
    frame #20: 0x000000010076573b libgdk-quartz-2.0.0.dylib`_gdk_window_process_updates_recurse + 538
    frame #21: 0x0000000100765667 libgdk-quartz-2.0.0.dylib`_gdk_window_process_updates_recurse + 326
    frame #22: 0x0000000100765667 libgdk-quartz-2.0.0.dylib`_gdk_window_process_updates_recurse + 326
    frame #23: 0x000000010077010e libgdk-quartz-2.0.0.dylib`-[GdkQuartzView drawRect:] + 296
    frame #24: 0x00007ff8045b5f00 AppKit`_NSViewDrawRect + 121
    frame #25: 0x00007ff805055ab4 AppKit`-[NSView _recursive:displayRectIgnoringOpacity:inContext:stopAtLayerBackedViews:] + 1848
    frame #26: 0x00007ff8045b562a AppKit`-[NSView(NSLayerKitGlue) _drawViewBackingLayer:inContext:drawingHandler:] + 747
    frame #27: 0x00007ff804b74a2b AppKit`-[NSViewBackingLayer drawInContext:] + 64
    frame #28: 0x00007ff804da449b AppKit`-[NSViewBackingStore update:] + 509
    frame #29: 0x00007ff804b7398f AppKit`-[NSViewBackingLayerContents update:] + 43
    frame #30: 0x00007ff804b7466b AppKit`-[NSViewBackingLayer display] + 1716
    frame #31: 0x00007ff80934ddf5 QuartzCore`CA::Layer::display_if_needed(CA::Transaction*) + 867
    frame #32: 0x00007ff8094e6a48 QuartzCore`CA::Context::commit_transaction(CA::Transaction*, double, double*) + 758
    frame #33: 0x00007ff80932efa1 QuartzCore`CA::Transaction::commit() + 725
    frame #34: 0x00007ff8045c683d AppKit`__62+[CATransaction(NSCATransaction) NS_setFlushesWithDisplayLink]_block_invoke + 285
    frame #35: 0x00007ff80509c75d AppKit`___NSRunLoopObserverCreateWithHandler_block_invoke + 41
    frame #36: 0x00007ff800e5d576 CoreFoundation`__CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 23
    frame #37: 0x00007ff800e5d49a CoreFoundation`__CFRunLoopDoObservers + 493
    frame #38: 0x00007ff800e5ca1c CoreFoundation`__CFRunLoopRun + 850
    frame #39: 0x00007ff800e5c102 CoreFoundation`CFRunLoopRunSpecific + 557
    frame #40: 0x00007ff80b7349d9 HIToolbox`RunCurrentEventLoopInMode + 292
    frame #41: 0x00007ff80b734616 HIToolbox`ReceiveNextEventCommon + 201
    frame #42: 0x00007ff80b734531 HIToolbox`_BlockUntilNextEventMatchingListInModeWithFilter + 66
    frame #43: 0x00007ff80446a7b9 AppKit`_DPSNextEvent + 880
    frame #44: 0x00007ff804d62f64 AppKit`-[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1304
    frame #45: 0x000000010077656b libgdk-quartz-2.0.0.dylib`poll_func + 176
    frame #46: 0x000000010140858f libglib-2.0.0.dylib`g_main_context_iterate_unlocked + 299
    frame #47: 0x0000000101408451 libglib-2.0.0.dylib`g_main_context_pending + 48
    frame #48: 0x0000000100ba8502 libgtk-quartz-2.0.0.dylib`gtk_events_pending + 30
    frame #49: 0x0000000100013dae gtkwave`gtk_events_pending_gtk_main_iteration + 15
    frame #50: 0x0000000100036a3e gtkwave`main_2 + 12077
    frame #51: 0x0000000100033b0d gtkwave`main + 16
    frame #52: 0x00007ff8009f7386 dyld`start + 1942
```

tbybell, 2024-02-06:
GTK2 for OSX has issues that its maintainers would not fix all the way back in 2011 after I identified issues and fixes.  I got annoyed with it and used my own patches (which also are what I patched the libraries in the gtkwave.app with on sourceforge):

```
746:/home/bybell/git/gtkwave4/gtkwave/contrib/bundle_for_osx> ls -la *patch
-rw-rw-r--. 1 bybell bybell 1000 Oct  6 12:48 gtk_diff_against_modulesets_2.patch
-rw-rw-r--. 1 bybell bybell 1519 Oct  6 12:48 gtk_diff_against_modulesets_3.patch
-rw-rw-r--. 1 bybell bybell 4840 Oct  6 12:48 gtk_diff_against_modulesets.patch
```

Later versions of gtk are possibly broken in the same exact areas.  The problems I encountered were in _gdk_quartz_drawable_flush not behaving as it should.

-Tony

DanielO, 2024-02-06:
That is rather depressing, I wonder if I can interest the Macports GTK maintainers in the patches at least :)

I ended up fixing the "Unsupported depth" message with the patch in here https://gitlab.gnome.org/GNOME/gtk/-/issues/580 (and also posted to try and get the maintainers to commit)

tbybell, 2024-02-06:
It might be a better situation now.  2011 as tagged in the top line of those diffs was a very long time ago.  All the OSX binaries that I have posted on SF are built off an old Snow Leopard image running on a VM from back in the days when you could mail order the OSX install CD for something like $29.  I've never owned a Mac.  Given how app bundles are self contained, I didn't care back then as nobody else was really compiling their own binaries.  Those three sets of patches are what I needed for the jh-build.  The other broken functionality it fixed was for screenshot, which was another unsupported color depth (or byte ordering--can't remember which) issue.

SphericalCowww, 2024-07-22:
Hello @DanielO, do you know if we can manually apply the patch?
I am working with Macport on MacOS Monterey with the M1 chip, and installed via:
`sudo port -v -N install gtk2 +quartz`
`sudo port install gtkwave`
The `gtk2 +quartz +x11` doesn't seem to be compatible with each other if it matters.

DanielO, 2024-07-22:
I ended up making a private MacPort with the custom patches to get it working.

It is stale now though, upstream has updated but I haven't rebuilt and tried it in a while.