gtkwave/gtkwave #117: High CPU utilization when nothing is happening

Vascom, 2022-02-02:
I use version 3.3.111 from official Fedora 35 repo.
Just open vcd file and look at some signals. GTKWave use ~40% of one CPU core (Ryzen 7).

It is normal?
What can I do to help solve this issue?

gtkwave, 2022-02-02:
PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND      
 2822 bybell    20   0 3431056   2.5g  35676 S   0.0 16.4   0:40.82 gtkwave                      
...that's what I see on my CentOS 7 machine.  I don't have a Fedora machine handy.  Is there a difference between X11 and Wayland?  I can't immediately think of any reason for CPU usage that high when idle.

-Tony

Vascom, 2022-02-02:
Seems it is really Wayland. Also I use KDE.
Just after start without any data loaded:
Wayland:
```
USER         PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
vascom   1375347 24.5  0.1 910848 58468 ?        Sl   23:11   0:10 /usr/bin/gtkwave
```

X11:
```
USER         PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
vascom   1376436  1.2  0.3 2524112 103012 ?      Sl   23:14   0:00 /usr/bin/gtkwave
```

gtkwave, 2022-02-02:
OK, thanks for looking into it.  That's really helpful.

It's probably going to be a set of code like this:

#ifdef GDK_WINDOWING_WAYLAND
if(GDK_IS_WAYLAND_DISPLAY(gdk_display_get_default())) gtk_widget_queue_draw(GLOBALS->wavearea);
#endif

I don't know if there is an extra one in there somewhere accidentally causing it to be called every frame clock on Wayland.  I had to add these GDK_WINDOWING_WAYLAND ifdefs in about a dozen places because Wayland wasn't getting triggered properly to redraw some screen areas.
-Tony

rfuest, 2022-02-02:
I did remove a lot of these explicit redraw calls in my efforts to rewrite the waveform display. And I didn't noticed this issue the last time I tested the new code with Wayland. But progress is very slow at the moment and I don't know when I will be able to open a PR with my changes.

gtkwave, 2022-02-02:
I'll take a look.  If I see the same thing on my Fedora machine, I'll just comment them out one by one and figure out what could be causing it through the process of elimination.  My guess is it's something attached to the frame clock so the perf drain would be continuous.

-Tony

gtkwave, 2022-02-03:
It's definitely the one in draw_marker() in wavewindow.c.  If I comment it out though, the screen updates for the marker when I use a touchscreen are very choppy or non-existent which looks to be fixed if I move the code to the very bottom of wavearea_drag_update_event().  For the original poster, try it out, I think that's it.

-Tony

gtkwave, 2022-02-03:
Counting objects: 11, done.
Delta compression using up to 4 threads.
Compressing objects: 100% (6/6), done.
Writing objects: 100% (6/6), 626 bytes | 0 bytes/s, done.
Total 6 (delta 5), reused 0 (delta 0)
remote: Resolving deltas: 100% (5/5), completed with 5 local objects.
To https://github.com/gtkwave/gtkwave
   9a87e48..48c6409  master -> master

-Tony

Vascom, 2022-02-04:
Next release will be soon?
How I can apply this patch to 3.3.111 to test it?

gtkwave, 2022-02-04:
I don't know when the next release will be.  It would be 3.4.0.  I haven't tried using it on a daily basis so I haven't promoted it yet to a release.  For patching older versions:

At the very bottom of draw_marker(), comment out these lines as shown:/*
#ifdef GDK_WINDOWING_WAYLAND
if(GDK_IS_WAYLAND_DISPLAY(gdk_display_get_default())) gtk_widget_queue_draw(GLOBALS->wavearea);
#endif
*/

At the very bottom of wavearea_drag_update_event() add these lines as shown:#ifdef GDK_WINDOWING_WAYLAND
if(GDK_IS_WAYLAND_DISPLAY(gdk_display_get_default())) gtk_widget_queue_draw(GLOBALS->wavearea);
#endif

-Tony

Vascom, 2022-02-09:
Thanks. It work.
```
diff -uNdr gtkwave-gtk3-3.3.111_old/src/wavewindow.c gtkwave-gtk3-3.3.111_new/src/wavewindow.c
--- gtkwave-gtk3-3.3.111_old/src/wavewindow.c	2021-09-06 23:39:11.000000000 +0300
+++ gtkwave-gtk3-3.3.111_new/src/wavewindow.c	2022-02-09 12:31:45.419573122 +0300
@@ -680,9 +680,9 @@
 cairo_destroy (cr);
 #endif
 
-#ifdef GDK_WINDOWING_WAYLAND
-if(GDK_IS_WAYLAND_DISPLAY(gdk_display_get_default())) gtk_widget_queue_draw(GLOBALS->wavearea);
-#endif
+// #ifdef GDK_WINDOWING_WAYLAND
+// if(GDK_IS_WAYLAND_DISPLAY(gdk_display_get_default())) gtk_widget_queue_draw(GLOBALS->wavearea);
+// #endif
 }
 ```

pghmcfc, 2022-10-04:
Looks like this fix didn't make it into 3.3.112.

gtkwave, 2022-10-04:
Thanks...yep, missed that one--sorry for the version churn, Paul.  Serves me right for running on my old X11 machine.  3.3.113 for gtk3 should be up on SF's files area if it's not there already.

From here on out, 3.3 series should be bugfixes relative to 3.3.111 only.  All the new GUI stuff will eventually be either in a 3.4 or a 4.0 release.

Vascom, 2023-11-15:
Same problem return in 3.3.117 (or earlier). High CPU load while just view waveforms.
Any new suggestions?

MikaelStrom, 2025-04-11:
I see this behaviour in v3.3.121 under Arch linux. I use i3 without compositor running under VMware Fusion 13.6.3, Intel iMac.

tbybell, 2025-04-11:
I tried 3.3.121 on RHEL with X11.  0.3% CPU utilization for me.  No change versus what I'm looking at will soon be a release candidate for 3.3.122, either.

rfuest, 2025-04-12:
I'm also running Arch and didn't encounter this issue with the Arch build of GTKWave 3.3.121.

@MikaelStrom: Could you run GTKWave by using `GTK_DEBUG=interactive gtkwave` and post a screenshot of the "General" tab in the GTK Inspector that should automatically pop up along side the GTKWave main window? The infos about the display and OpenGL might contain something useful to replicate the issue.

It would also be helpful if you could try to build and run the development version, to see if the issue is fixed there. You can just run it from the source directory without installing it:
```
git clone https://github.com/gtkwave/gtkwave.git
cd gtkwave
meson setup build
cd build
meson compile
src/gtkwave ../examples/des.gtkw
```

MikaelStrom, 2025-04-15:
@rfuest:
[image]
[image]
I built bb978d9d and so far no CPU usage issues, but it is not always it happens so I try this for a while and report back.