gtkwave/gtkwave #122: FST API DOCUMENTATION ADVANCED

NourFr10, 2022-04-26:
Hi Anthony, 
could you please tell us what these FST functions and variables does : 

`uint64_t fstReaderGetValueChangeSectionCount(void *ctx);
char *          fstReaderGetValueFromHandleAtTime(void *ctx, uint64_t tim, fstHandle facidx, char *buf);
uint32_t fstReaderGetNumberDumpActivityChanges(void *ctx)
int fstReaderGetFseekFailed(void *ctx)
uint64_t fstReaderGetValueChangeSectionCount(void *ctx)
uint64_t *blackout_times;
unsigned char *blackout_activity;`

And finally is there any function that return the handle of signal name like fstReaderGetSignalHandle(char *signal); 
Thanks

Timmmm, 2022-11-22:
I'm not an expert but I have been reading the code (trying to write an actual spec for the FST format). Here are a few:

`uint64_t fstReaderGetValueChangeSectionCount(void *ctx);`

Get the number of Value Change sections in the file (`FST_BL_VCDATA`/`FST_BL_VCDATA_DYN_ALIAS`/`FST_BL_VCDATA_DYN_ALIAS2`). The file is divided in time into sections (AKA blocks). This has two benefits:

1. You can start reading the file before it is finished writing.
2. It makes seeking faster since each block starts with the values of every variable at the block start time.

`char * fstReaderGetValueFromHandleAtTime(void *ctx, uint64_t tim, fstHandle facidx, char *buf);`

Not sure about this yet.

`uint32_t fstReaderGetNumberDumpActivityChanges(void *ctx)`

Get the number of times that `$dumpoff` and `$dumpon` were run during the simulation.

`int fstReaderGetFseekFailed(void *ctx)`

Return 1 if `fseek()` failed when reading the file. 0 otherwise. The reader code seeks all over the place and IIRC doesn't verify the total file length so this likely indicates a truncated file or possibly corrupted data (I'd say a truncated file is way more likely though).

`int fstUtilityEscToBin(unsigned char *d, unsigned char *s, int len)`

Unescapes a C string. I.e. `"foo\\n"` is converted to `"foo\n"`. It doesn't check the destination buffer has enough space. I'm sure there are no buffer overflows here. /s :-)

`uint64_t *blackout_times;`

This is an array of the times that `$dumpon` and `$dumpoff` were called (of length `num_blackouts`). Each entry is a time delta from the previous one. I guess the first one is a time delta from `$dumpvars` but I haven't checked that.

`unsigned char *blackout_activity;`

An array of the same length as `blackout_times` that says whether the event was `$dumpon` (1) or `$dumpoff` (0). This may be redundant since I guess it will always contain [0, 1, 0, 1, 0, 1...] but maybe not.

Hope that helps! I may be wrong about some of those - it's just based on reading the code and I've barely used GtkWave.

NourFr10, 2022-11-22:
Hi I appreciate your effort for explanation thank you so much bro.

NourFr10, 2022-11-22:
You mentioned that you were trying to write the actual "spec" for the FST format. Honestly, I didn't find any spec (well written and documented kind of paper) 7 months ago when I was working on the FST reader/writer.

gtkwave, 2022-11-23:
char * fstReaderGetValueFromHandleAtTime(void *ctx, uint64_t tim, fstHandle facidx, char *buf);

Not sure about this yet.

...it's used for tools like rtlbrowse that immediately want to get the value for a signal at a given time without having to cache all the value changes manually.  I haven't used it outside of rtlbrowse.  Anyway in the .c code where it talks about RVAT (return value at time), that's what it is.
-Tony

Timmmm, 2022-11-23:
Ha yeah I did notice the strange use of the date field to add extra fields, and the 5 (I think?) different compression schemes used! I would also suggest:

* Use TLV (Tag Length Value) in more places. Some are just Tag Value, and some have the tag/lengths at the end which is weird.
* Require little endian for all values. Nobody uses big endian.
* Have a magic number at the start of the file (this can include the major version, e.g. `FST0`, `FST1`, etc.)

There are definitely some head scratchers in the format but overall it's *way* better than VCD, and Verilator supports it so it's actually useful and I think worthy of a proper spec.

I'll open a new issue when I've actually finished it.

gtkwave, 2022-11-24:
Big endian mainly was used as a machine-independent XDR (see RFC 4506) mainly because it's far more human readable in hex dumps back when I was debugging the format during development.  Most of the heavy lifting that burns machine cycles involves reading of variable length integers which by their nature are stored in little endian given how they're encoded using repeated logical right shifting until a value is zero (see fstWriterVarint()).  I'm an old 68000/PPC programmer even though my day job is x86 logic design, so I'm way too familiar with big endian.  Perfwise, there's no real perf loss where those integers are used.

Lengths at the end, wherever it was done, I believe was done because you'd have some chunk of memory you know is x number of bytes (that x came back from some loader or parsing phase), so the lengths are at a fixed position from the end which is treated as an anchor position.
Note that knowing the file format itself isn't enough to write out an FST file successfully, though it's enough to read it.  To write it requires some pretty weird handstands in moving data around as well as data reformatting, which is what's covered in the one appendix in the gtkwave user manual.  Internally, those representations in memory are quite similar to the old MVLSIM AET format used at IBM during the late 1990s.
It took about 11 years of trial and error to go from VCD (1998) to FST (2009):

1) LXT (2001) more or less based on MVLSIM AET concepts using linked lists that point backward in time with time values represented implicitly by position in the file vs a position vs time lookup table.
2) VCD recoder (2005) which is where in-memory reformatting was discovered.  This was driven to squeeze out better capacity to deal with the 3GB 32-bit user space limitations encountered by various users.
3) FST (2009) using a block based approach writing MVLSIM-flavored data into memory then reformatting it similar to the VCD recoder.  I believe FSDB does something similar.

Weirdness often came from missteps along the way.  LXT2 (2003) and VZT (2004) were evolutionary dead ends but provided useful concept learning.
Note that CVC also supports FST.  I spent a couple of months somewhere between 2009 and 2011 doing performance tuning of various sections of writer code with their one programmer who is now at Synopsys.  Later on I did add attribute features to support some requests from the VHDL compiler nvc.

-Tony

Timmmm, 2022-11-24:
That's some great info, thanks!