gtkwave/gtkwave #124: Any way to get Highlighted traces within gtkwave ?

NourFr10, 2022-05-12:
Hi Anthony, 

Is there a way to get the Highlighted traces flat names within the gtkwave with a tcl scripts ?

the gtkwave's tcl callbacks allow just to highlight a trace based on its flat name or its index, but  not the inverse

Thanks.

gtkwave, 2022-05-13:
Please describe to me exactly what you're trying to do.  Are you trying to get a list of all highlighted signals?
Currently, no, there is no function to do that.  Possibly iterating through signals by examining getTraceFlagsFromIndex might work.  Highlight should be bit zero from the enum TR_HIGHLIGHT_B in analyzer.h.

-Tony

NourFr10, 2022-05-14:
[image]
I am trying to get the names (for example: top.des.round1.b1x[1:6]) of the highlighted traces within the Signal list as the image bellow illustrate.

NourFr10, 2022-05-14:
Hi, the idea of

>  Possibly iterating through signals by examining getTraceFlagsFromIndex

 worked for me thank you Anthony.