gtkwave/gtkwave #126: Saving View settings

francis2tm, 2022-05-24:
Hello,
First of all, amazing work with gtkwave, thank you!
I was wondering if it's possible to add a feature that enables saving the current view settings (like show waves highlight, show filled high values, show base symbols etc...). It's rather cumbersome to have to enable all the view settings every time one closes and reopens gtkwave.

NourFr10, 2022-05-25:
Hi, as far as i can understand your use case, you can click <control>S  and then name your save file(exp: save.gtkw), after that close gtkwave and run it in command line like this: gtkwave dump.vcd save.gtkw

francis2tm, 2022-05-26:
I was already doing that. the .gtkw saves everything except the View settings...

francis2tm, 2022-05-26:
My OS is linux, and the gtkversion I'm using is v3.3.86

rfuest, 2022-05-26:
You can change the default view settings using a `.gtkwaverc` file in your home directory. See the `gtkwaverc` man page or the users guide at http://gtkwave.sourceforge.net/gtkwave.pdf for more information.

francis2tm, 2022-05-27:
> You can change the default view settings using a `.gtkwaverc` file in your home directory. See the `gtkwaverc` man page or the users guide at http://gtkwave.sourceforge.net/gtkwave.pdf for more information.

Thank you. That was exactly what I was looking for. For those who come looking for the same answer, I had to create a .gtkwaverc in my home directory and then filled the configs that I wanted enabled.
Example .gtkwaverc:
```
highlight_wavewindow 1
fill_waveform 1
clipboard_mouseover 1
disable_mouseover 0
show_base_symbols 1
```

closed 2022-05-27 by francis2tm